DIGImend/digimend-kernel-drivers #497: Ugee hk1060pro Supported?

keranapy, 2020-12-23:
Hello, I was just wondering if the UGEE HK1060 pro is currently supported.

nic3-14159, 2020-12-24:
Hello! As far as I can tell, that tablet is not currently supported. Do you happen to have that tablet? If so, would you be able to post diagnostics as outlined on [this page](https://digimend.github.io/support/howto/trbl/diagnostics/)? That would be useful to anyone working on adding support for that tablet.

keranapy, 2020-12-24:
> Hello! As far as I can tell, that tablet is not currently supported. Do you happen to have that tablet? If so, would you be able to post diagnostics as outlined on [this [...]

Ok thanks for answering I will do it, I will try

closed 2020-12-31 by keranapy