DIGImend/digimend-kernel-drivers #498: XP-Pen Artist 13.3 Pro

Urfoex, 2020-12-25:
https://www.xp-pen.de/product/470.html
https://www.xp-pen.com/product/464.html

Diagnostics as requested on http://digimend.github.io/support/howto/trbl/diagnostics/:

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5741912/descriptors.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5741913/frame_buttons.txt)
[frame_dials.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5741914/frame_dials.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5741915/hid_report_descriptors.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5741916/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5741917/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5741918/pen_pressure.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5741919/pen_tilt.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5741920/probe.txt)

Using "driver" from XP-Pen
https://www.xp-pen.de/download-436.html
https://www.xp-pen.com/download-430.html
- custom mapping for tablet buttons, ring, pen buttons
- both pen buttons work
- Problems:
  - not a real driver; a program that needs to run
  - program has high CPU load
  - program runs with SUDO
  - no tilt at all

[image]
[image]
[image]
[image]

I tried different possibilities:
Without any settings (which should be libinput?):
- Plug and Play - works out of the box
- everything is kind of working
- remapping of 1st button on pen
- map-to-screen in xinput
- problems:
  - 2nd pen button not working
  - couldn't figure out on how to remap the tablet-buttons or tablet ring
  - tilt is not working in certain areas; also not completely using all values as advertised (-30 to 30 instead -64 to 64)
  - tilt: turning the pen 360° it sticks at  some positions, and snaps to others
  - pressure is just 2047 instead of 8191

With wacom driver:
- GUI Settings in Plasma/KDE:
Using wacom driver in xorg.conf in tablet settings in plasma/kde:
[image]
[image]
[image]
[image]

- Problems:
  - pressure with 65536 too high
  - no 2nd pen button
  - no tablet buttons nor tablet ring
  - tilt from -61 to 61, but still buggy in some areas
  - most UI settings in Plasma are not doing or changing anything

With evdev driver:
- seems a little better then libinput
- tablet buttons and ring work; still don't know how to remap
- 2nd pen button working, but it's same as tip; no way to separately remap
- pressure till 8191
- tilt from -61 to 61, but advertised as -127 to 127; and still buggy in some cases
- 

More information:
[Pentablet.log](https://github.com/DIGImend/digimend-kernel-drivers/files/5741922/Pentablet.log)

Feels like there is just a tiny little setting to do left somewhere and it would work nicely.
Any ideas? Recommends?
Ways to help?

esotericDisciple, 2021-01-03:
it seems that the parblo a610 plus is also just a rebranded xp-pen, but i can't get the buttons or the touchring to work with either evdev or digimend...

kurikaesu, 2021-08-22:
See if my work in https://github.com/DIGImend/digimend-kernel-drivers/pull/557 helps resolve your issues.