DIGImend/digimend-kernel-drivers #502: Tablet diagnostics collection Ugee hk 1060 pro ready!

keranapy, 2020-12-31:
Hello I have done a Compilation of tablet diagnostics of the Ugee Hk 1060pro. I hope and it helps you to make a support for that graphic tablet.
test performed on: Manjaro Linux
Tablet diagnostics collection link:https://ya.co.ve/tFo
 Manufacturer Page: http://www.ugee.net/
Tablet model: http://www.ugee.net/goods/60.html

nic3-14159, 2020-12-31:
Hello again! Could you try running the lsusb commands again? The descriptors file in the diagnostics you posted doesn't seem to contain the information that should be in it

keranapy, 2020-12-31:
I suppose it is the first time that I do those things xd, I will try again

nic3-14159, 2021-01-20:
Doh! After taking a look at the lsusb dump, I realized I should have asked for that first before saying your tablet wasn't supported. It turns out that the usb id (5543:0081) already exists in the Digimend code as a sort of generic id shared by a few tablets, so it is possible that your tablet will work without any changes to the code. I didn't see the tablet by name, so I assumed it wasn't supported. Sorry about that!

So maybe you could try installing it and report if everything works?

keranapy, 2021-01-20:
> Doh! After taking a look at the lsusb dump, I realized I should have asked for that first before saying your tablet wasn't supported. It turns out that the usb id (5543:0081) already exists in the [...]

Ok I'll try it as soon as I can, thanks

closed 2021-02-06 by keranapy