DIGImend/digimend-kernel-drivers #505: Artisul Pencil

Samalbi, 2021-01-18:
Hi, my tablet LED doesnt light up and i cant download the drivers for some reason and when it is connected to the computer it doesnt show as connected to the pc i thought it was the drivers at first but i cant even download the drivers if it isnt even connected i really need help

nic3-14159, 2021-01-18:
Hello! I'm not quite sure if Digimend supports your tablet in the first place. If you open a terminal, plug in your tablet, and type in the command `lsusb`, what is the output?

Samalbi, 2021-01-18:
> 
> 
> Hello! I'm not quite sure if Digimend supports your tablet in the first place. If you open a terminal, plug in your tablet, and type in the command `lsusb`, what is the output?

'lsusb' is not recognized as an internal or external command,
operable program or batch file.

nic3-14159, 2021-01-18:
It seems that are using Windows. Digimend provides drivers for Linux, which won't work on Windows.