DIGImend/digimend-kernel-drivers #506: Huion H1060P stuck in phone mode on Ubuntu 20.04

olivershetler, 2021-01-18:
My Huion H1060P is stuck in phone mode on Ubuntu 20.04.

When I change the active area using xsetwacom, it doesn't make any difference.

I've read that this is not a unique issue. Several kinds of tablets mistake Ubuntu for the Android OS and get stuck in phone mode.

How can I get my tablet out of phone mode?

cebe, 2021-11-16:
I have the same issue with a Huion H950P.

godlixe, 2022-04-22:
I am also having the same issue with Huion H650P, are there still no solutions?

godlixe, 2022-04-22:
Got the solution, check the support website and adjust for each tablet. https://support.huion.com/en/support/solutions/articles/44001906540-how-to-manually-set-your-tablet-to-phone-mode