DIGImend/digimend-kernel-drivers #510: Setup Huion Q11 v2 tablet on Linux

valent1draw, 2021-01-25:
Hello everyone, 
I share my success of the day, that of having succeeded in setting the keys of the Huion Q11 V2 graphics tablet on linux! 
(With the help of David Revoy, and his post 
[link](https://www.davidrevoy.com/article331/setup-huion-giano-wh1409-tablet-on-linux-mint-18-1-ubuntu-16-04 )
and a programmer friend. 

I use Linux Mint 20.1

For the first two steps, go to the post and follow the instructions.

_1/Install Digimend driver via DKMS 

2/Add a custom X11 rule_s

**What changes is the third part "User preference, buttons, settings."**

Now, let configure the tablet with xsetwacom.
For this, we can load xsetwacom command. But they are not persistent avec restarting computer.
So we need to store this collections of commands in a bash script and call this bash script when the system open.

```
mkdir scripts  
xed ~/scripts/Huion_Q11v2.sh
```

... and inside paste and customize the script under. To know the list of special "key" you can map on buttons, look at this list. Don't forget to change the screen size to your screen size (the tablet is not 100% same ratio than 1920x1080, it needs code to force good vertical and horizontal ratio so when you trace a circle on your tablet, a circle appear on the screen, and not an ellipse potato ). And comment or delete all the line you are not interested.

```
#! /bin/bash  
# Setup HUION Q11 v2, after bridged to wacom driver with Digimend Kernel module.  
# License: CC-0/Public-Domain license  
# author: deevad  

# Tablet definition  
tabletstylus="HUION Huion Tablet stylus"  
tabletpad="HUION Huion Tablet Pad pad"  

# Reset  
xsetwacom --set "$tabletstylus" ResetArea  
xsetwacom --set "$tabletstylus" RawSample 4  

# Mapping  
# get maximum size geometry with:  
# xsetwacom --get "$tabletstylus" Area  
# 0 0 55880 34925  
tabletX=55880  
tabletY=34925  
# screen size:  
screenX=1920  
screenY=1080  
# map to good screen (dual nvidia)  
# xsetwacom --set "$tabletstylus" MapToOutput "HEAD-0"  
# setup ratio :  
newtabletY=$(( $screenY * $tabletX / $screenX ))  
xsetwacom --set "$tabletstylus" Area 0 0 "$tabletX" "$newtabletY"  

# Buttons  
# =======  
xsetwacom --set "$tabletstylus" Button 2 2   
xsetwacom --set "$tabletstylus" Button 3 3  
# ---------  
# | 12 |
# | -- |
# | 11 |
# | -- |
# | 10 |
# | -- |
# |  9 |
# |=======|  
# |  8 |
# | -- |
# |  3 |
# | -- |
# |  2 |
# | -- |
# |  1 |
# |=======| 

xsetwacom --set "$tabletpad" Button 12 "key +ctrl a" # Select all
xsetwacom --set "$tabletpad" Button 11 "key +ctrl shift a" # Deselect
xsetwacom --set "$tabletpad" Button 10 "key +ctrl r" # Selection tool
xsetwacom --set "$tabletpad" Button 9 "key t" # Transform tool
xsetwacom --set "$tabletpad" Button 8 "key b"  # Brush
xsetwacom --set "$tabletpad" Button 3 "key Shift_L" # Resize brush
xsetwacom --set "$tabletpad" Button 2 "key m" # Mirror 
xsetwacom --set "$tabletpad" Button 1 "key +ctrl z" # Undo 

# Xinput option  
# =============  
# for the list:  
# xinput --list  zz

# xinput list-props 'HUION Huion Tablet Pen Mouse' 
xinput list-props 'HUION Huion Tablet Touch Strip pad' "Evdev Middle Button Emulation" 0 
# xinput set-prop 'HUION Huion Tablet Pen' "Evdev Middle Button Emulation" 0  
# alternate way to map to a single screen  
# execute "xrander" in a terminal to get the screen name ( DVI-D-0 in this example )  
# xinput set-prop 'HUION' DVI-D-0
```

tadaaa

Miratu, 2021-07-01:
Hardware: Huion Inspiroy Q11K v2
lsusb: 256c:006e HUION Huion Tablet
Vendor: Huion
Product: Inspiroy Q11K v2
Status: 
Specs: 279,4 x 174,6mm (11″×6.875″), 5080 LPI, 266 RPS , 8192 PL, tilt-sensitive, Frame controls 8 buttons, wireless.

Distro: Ubuntu 20.04 LTS
Kernel: 5.7.7-050707-generic(x86_64)

I already had Digimend 10 installed for the previous tablet ([Gaomon M10K Pro](https://github.com/DIGImend/digimend-kernel-drivers/issues/433#issue-651427185)) that my son now uses, with windows.

Currently I use a Huion Q11K V2 and I have followed the steps indicated by @valent1draw from point 2 and as he says: tadaa !!!

Everything works perfectly, although I have not been able to test the inclination of the pen. At the end I attach all the tests. I wish they were useful.

The script has been as follows. 

```
#! /bin/bash  
# Setup HUION Q11 v2, after bridged to wacom driver with Digimend Kernel module.  
# License: CC-0/Public-Domain license  
# author: deevad  

# Tablet definition  
tabletstylus="HUION Huion Tablet stylus"  
tabletpad="HUION Huion Tablet Pad pad"  

# Reset  
xsetwacom --set "$tabletstylus" ResetArea  
xsetwacom --set "$tabletstylus" RawSample 4  

# Mapping  
# get maximum size geometry with:  
# =============
# xsetwacom --get "$tabletstylus" Area  
# doesn't work, I don't know why, but it works: :
# xsetwacom --get "HUION Huion Tablet stylus" Area 
# =============
# 0 0 55880 34925  
tabletX=55880  
tabletY=34925  
# screen size:  
screenX=1920  
screenY=1080  
# map to good screen (dual nvidia)  
# xsetwacom --set "$tabletstylus" MapToOutput "HEAD-0"  
# setup ratio :  
newtabletY=$(( $screenY * $tabletX / $screenX ))  
xsetwacom --set "$tabletstylus" Area 0 0 "$tabletX" "$newtabletY"  

echo change orientation to use left hand
echo if you are right-handed, deactivate this line
xsetwacom --set "$tabletstylus" Rotate half

# Buttons  
# =======  
xsetwacom --set "$tabletstylus" Button 2 2   
xsetwacom --set "$tabletstylus" Button 3 3 
echo for right-handed use, reverse the order of the buttons  
# ---------  
# |  1 |
# | -- |
# |  2 |
# | -- |
# |  3 |
# | -- |
# |  8 |
# |=======|  
# |  9 |
# | -- |
# | 10 |
# | -- |
# | 11 |
# | -- |
# | 12 |
# |=======| 

echo for right-handed use, reverse the order of the buttons 
echo enter the commands with quotation marks to suit the user 
xsetwacom --set "$tabletpad" Button 1 "key control"
xsetwacom --set "$tabletpad" Button 2 "key shift"
xsetwacom --set "$tabletpad" Button 3 "key s"
xsetwacom --set "$tabletpad" Button 8 "key r"
xsetwacom --set "$tabletpad" Button 9 "key g"
xsetwacom --set "$tabletpad" Button 10 "key +ctrl +z -z -ctrl"
xsetwacom --set "$tabletpad" Button 11 "key +shift +d -d -shift"
xsetwacom --set "$tabletpad" Button 12 "key tab"

# Xinput option  
# =============  
# for the list:  
# xinput --list  zz

# xinput list-props 'HUION Huion Tablet Pen Mouse' 
xinput list-props 'HUION Huion Tablet Touch Strip pad' "Evdev Middle Button Emulation" 0 
# xinput set-prop 'HUION Huion Tablet Pen' "Evdev Middle Button Emulation" 0  
# alternate way to map to a single screen  
# execute "xrander" in a terminal to get the screen name ( DVI-D-0 in this example )  
# xinput set-prop 'HUION' DVI-D-0
```

I'm left-handed so it's set to hold the stylus with my left hand and the buttons on the right side. 

Thanks "DIGImenders" for your help.
Apologies for my "google-english". 

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6746668/descriptors.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6746669/frame_buttons.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6746670/hid_report_descriptors.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6746671/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6746672/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6746673/pen_pressure.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6746675/pen_tilt.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6746676/probe.txt)

valent1draw, 2021-07-04:
Hello @Miratu !

To put the tablet in left-handed mode, which is also my case, you can go into the software "tablette graphic", 
as with the previous code, all the drivers are well installed this software should detect your tablet and offer you different settings.   (Hoping that the software "Tablette graphic" exists on your linux)

If not, to translate, you can use the site https://www.deepl.com 

^^

nexar, 2022-02-11:
Since writing the above I did a reboot and now everything has sprung to life.  I've also managed to set up the buttons, although reversed as I'm right handed.

Thanks very much @MimosaBoo and @Miratu

MimosaBoo, 2022-02-14:
@nexar Sorry I didn't see your message. 
I'm glad everything is back to normal! 

The shortcut keys have not been working for a few days. Did anyone have the problem?

nexar, 2022-02-15:
@MimosaBoo, the keys on my Q11 pad are working fine all the time.  Have you upgraded your OS at all?

MimosaBoo, 2022-02-15:
@nexar All the keys worked until a short time ago.
My software is up to date.

MimosaBoo, 2022-02-18:
> Hmmm I do something slightly different. Rather than having the script
> assigning the keys running at boot up I run it as part of loading my
> graphics software. This is because my tablet is not permanently connected
> to the laptop but only when I want to use it.
>
> Is it possible that your tablet is not activated before the shell script
> runs? If you run the shell script manually do the keys work after it has
> run?

I didn't set the script reading to automatic.
Is it possible that the key ids have changed?

MimosaBoo, 2022-02-21:
$ tabletpad="HUION Huion Tablet Pad pad"
$ xsetwacom set "$tabletpad" Button 2 "key m"

When i test this code in the terminal, it does not result in an error. 
But the keys still do nothing

MimosaBoo, 2022-03-07:
I found it! 
It has a button on the side of the tablet to lock it.
I must have activated it by mistake... ( ಠ ʖ̯ ಠ)