DIGImend/digimend-kernel-drivers #511: Express keys settings in H1060P erase when rebooted

Duduz7, 2021-01-26:
Everytime I restart my computer(Ubuntu 20.04, kernel 5.10), the express keys or buttons settings which I configured before get erased, by looking the issue up on other sites and they all say that you should create a file to make a script at the booting of the machine, something like: 

```
#!/bin/sh
xsetwacom set "HID 256c:006d Pad pad" button 1 key Ctrl Z # Undo
xsetwacom set "HID 256c:006d Pad pad" button 2 key Ctrl Y # Redo
xsetwacom set "HID 256c:006d Pad pad" button 3 key Shift E # Eraser
xsetwacom set "HID 256c:006d Pad pad" button 8 key N # Pen tool
xsetwacom set "HID 256c:006d Pad pad" button 9 key Shift B # Bucket/Fill tool
xsetwacom set "HID 256c:006d Pad pad" button 10 key [ # Decrease brush size
xsetwacom set "HID 256c:006d Pad pad" button 11 key ] # Increase brush size
xsetwacom set "HID 256c:006d Pad pad" button 12 key Ctrl S # Save file
exit

```

But I don't know where to put this text file, please help
but I don't

spxak1, 2021-01-27:
This is not a digimend issue. You need to make a service to run this script at boot. Research that. Nothing to do with this project.

Duduz7, 2021-01-27:
> This is not a digimend issue. You need to make a service to run this script at boot. Research that. Nothing to do with this project.

Well... commands on a terminal are as well a "research issue" and they have to do with this project if a command is refering or using this driver, therefore a script that is related to this project and something that usually happens to plenty of people is as well as a matter of this project, too, that's why people open issues and ask questions here, thanks.