DIGImend/digimend-kernel-drivers #512: driver install errors

protospheric, 2021-02-01:
This is what I'm getting on Ubuntu 20.04:

Seems there is a problem with dkms although it's installed on my system, or at least the digimend-10 drivers use of them.

Anyone else seeing / fixed this on their system?

```
Loading new digimend-10 DKMS `files...
Building` for 5.8.0-38-generic `5.8.0-41-generic
Building` initial module for `5.8.0-38-generic
ERROR:` Cannot create report: [Errno 17] File exists: `'/var/crash/digimend-dkms.0.crash'
Error!` Bad return status for module build on kernel: 5.8.0-38-generic `(x86_64)
Consult /var/lib/dkms/digimend/10/build/make.log for more information.
dpkg: error processing package digimend-dkms (--configure):
 installed digimend-dkms package post-installation script subprocess returned error exit stat
us 10
Errors were encountered while processing:
 digimend-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

nic3-14159, 2021-02-11:
What does the file `/var/lib/dkms/digimend/10/build/make.log`contain?

LoopingRocks, 2021-02-12:
I encountered the same problem with the Linux Mint 20.1 which is also based on Ubuntu 20.04.
The file /var/lib/dkms/digimend/10/build/make.log contains

DKMS make.log for digimend-9 for kernel 5.4.0-65-generic (x86_64)
Fri 12 Feb 2021 12:19:46 PM CET
make -C /lib/modules/5.4.0-65-generic/build SUBDIRS=/var/lib/dkms/digimend/9/bui
ld modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-65-generic'
  HOSTCC  scripts/basic/fixdep
scripts/basic/fixdep.c:97:10: fatal error: sys/types.h: No such file or director
y
   97 | #include <sys/types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[3]: *** [scripts/Makefile.host:107: scripts/basic/fixdep] Error 1
make[2]: *** [Makefile:550: scripts_basic] Error 2
make[1]: *** [Makefile:723: include/config/auto.conf.cmd] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-65-generic'
make: *** [Makefile:22: modules] Error 2

nic3-14159, 2021-02-12:
@LoopingRocks It looks like you tried installing v9. I believe v9 no longer builds for newer kernels, requiring changes in the makefile. This was fixed in v10. Could you try downloading and installing v10 and see if that works?

protospheric, 2021-02-12:
Yes I’ll give it a shot

nic3-14159, 2021-02-12:
> Yes I’ll give it a shot
> […](#)

@protospheric From the log you posted originally, it does look like you installed v10, so I don't think that is the issue. I was replying to the other person, as it looked like they installed v9.

Based on the log you originally posted, make encountered an error, and it says you should consult `/var/lib/dkms/digimend/10/build/make.log`. Could you post the contents of that file after you get the error when installing digimend?

LoopingRocks, 2021-02-14:
@nic3-14159 Thanks for your answer. The v10 worked fine.
For future reference, the v10 is at https://github.com/DIGImend/digimend-kernel-drivers/releases/download/v10/digimend-dkms_10_all.deb

protospheric, 2021-02-18:
> > Yes I’ll give it a shot
> > […](#)
> 
> @protospheric From the log you posted originally, it does look like you installed v10, so I don't think that is the issue. I was replying to the other person, as it looked like they installed v9.
> 
> Based on the log you originally posted, make encountered an error, and it says you should consult `/var/lib/dkms/digimend/10/build/make.log`. Could you post the contents of that file after you get the error when installing digimend?

So - I downloaded v10 package today, and installed it successfully. Table is working in 20.04 now. Not sure why it did not work the first time around.

For the reported make issue - there was not a make.log file in the location you specified, but I found one here - var/lib/dkms/digimend/10/5.8.0-43-generic/x86_64/log/make.log. That file contained the following (but since its working now with the v10 package, seems this is a closed issue, at least from my perspective):

make -C /lib/modules/5.8.0-43-generic/build M=/var/lib/dkms/digimend/10/build mo
dules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-43-generic'
  CC [M]  /var/lib/dkms/digimend/10/build/hid-kye.o
  CC [M]  /var/lib/dkms/digimend/10/build/hid-uclogic-core.o
  CC [M]  /var/lib/dkms/digimend/10/build/hid-uclogic-rdesc.o
  CC [M]  /var/lib/dkms/digimend/10/build/hid-uclogic-params.o
  CC [M]  /var/lib/dkms/digimend/10/build/hid-polostar.o
  CC [M]  /var/lib/dkms/digimend/10/build/hid-viewsonic.o
  LD [M]  /var/lib/dkms/digimend/10/build/hid-uclogic.o
  MODPOST /var/lib/dkms/digimend/10/build/Module.symvers
  CC [M]  /var/lib/dkms/digimend/10/build/hid-kye.mod.o
  CC [M]  /var/lib/dkms/digimend/10/build/hid-polostar.mod.o
  CC [M]  /var/lib/dkms/digimend/10/build/hid-uclogic.mod.o
  CC [M]  /var/lib/dkms/digimend/10/build/hid-viewsonic.mod.o
  LD [M]  /var/lib/dkms/digimend/10/build/hid-uclogic.ko
  LD [M]  /var/lib/dkms/digimend/10/build/hid-viewsonic.ko
  LD [M]  /var/lib/dkms/digimend/10/build/hid-kye.ko
  LD [M]  /var/lib/dkms/digimend/10/build/hid-polostar.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-43-generic'

Polaczeq22, 2021-03-24:
For me (Ubuntu 20.04, installing via .deb package) it returns an error 10, requiring autoconf.h which exists in required directory

make.log:
DKMS make.log for digimend-10 for kernel 5.8.0-45-generic (x86_64)
śro, 24 mar 2021, 09:41:10 CET
make -C /lib/modules/5.8.0-45-generic/build M=/var/lib/dkms/digimend/10/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-45-generic'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[1]: *** [Makefile:746: include/config/auto.conf] Error 1
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-45-generic'
make: *** [Makefile:25: modules] Error 2

Deb package installer log:
Selecting previously unselected package digimend-dkms.
(Reading database ... 402130 files and directories currently installed.)
Preparing to unpack .../digimend-dkms_10_all.deb ...
Unpacking digimend-dkms (10) ...
Setting up digimend-dkms (10) ...
Loading new digimend-10 DKMS files...
Building for 5.8.0-45-generic 5.8.0-48-generic
Building initial module for 5.8.0-45-generic
ERROR (dkms apport): unable to determine source package for digimend-dkms
Error! Bad return status for module build on kernel: 5.8.0-45-generic (x86_64)
Consult /var/lib/dkms/digimend/10/build/make.log for more information.
dpkg: error processing package digimend-dkms (--install):
 installed digimend-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 digimend-dkms
(Reading database ... 402673 files and directories currently installed.)
Preparing to unpack .../digimend-dkms_10_all.deb ...

------------------------------
Deleting module version: 10
completely from the DKMS tree.
------------------------------
Done.
Unpacking digimend-dkms (10) over (10) ...
Setting up digimend-dkms (10) ...
Loading new digimend-10 DKMS files...
Building for 5.8.0-45-generic 5.8.0-48-generic
Building initial module for 5.8.0-45-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/digimend-dkms.0.crash'
Error! Bad return status for module build on kernel: 5.8.0-45-generic (x86_64)
Consult /var/lib/dkms/digimend/10/build/make.log for more information.
dpkg: error processing package digimend-dkms (--install):
 installed digimend-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 digimend-dkms

Polaczeq22, 2021-03-24:
Nevermind, I had leftover files from older kernel versions and thus I couldn't install driver.