DIGImend/digimend-kernel-drivers #514: Letsketch 9628 graphic tablet (Diagnostics)

Mustafa-MS, 2021-02-02:
I've bought a cheap graphic tablet ([Letsketch](http://www.letsketch.cn/a/xiangmuyufuwu/zhinenshouxieban/9628shuweiban/) 9628) rebranded in turkey as (Starcom gc-422).
It has a driver for Windows and Mac but no support for Linux!
It's not mentioned in the list of known tablets, so I followed the guideline to collect diagnostic data, and attached the logs.

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5914227/descriptors.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5914233/hid_report_descriptors.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5914239/probe.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5914363/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5914366/pen_pressure.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5914369/pen_buttons.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5914372/frame_buttons.txt)

ilmanowar, 2021-03-14:
Hello Mustafa!
I have the same device with another brand. I got it working after installing the .deb installer (which trigger some other installations) and set the Xorg conf file according to the info page. Therefore this tablet should be listed as compatible.

Although, I chose a different name and location for the conf file.
Here the command and file content:

sudo nano /etc/X11/xorg.conf

Section "InputClass"
    Identifier "Tablet"
    Driver "wacom"
    MatchDevicePath "/dev/input/event*"
    MatchUSBID "6161:4d15"
EndSection

I hope it will help you.
Massimo.

nic3-14159, 2021-03-14:
@ilmanowar As far as I can tell, there is nothing in the digimend code that supports this tablet. It's possible that it simply works with the generic hid drivers, and that the xorg conf file is all that is needed (In other words, installing the deb isn't necessary) Could you post the output of the command
`lsmod | grep hid`
when the tablet is plugged in?

Mustafa-MS, 2021-03-14:
Hello Massimo,
It worked like a charm!
Thank you, it did really help.

ilmanowar, 2021-03-14:
Hello Mustafa, I am happy it worked!

@nic3-14159 , I can assure you it is working with the Wacom driver because it was recognized as a tablet and mouse, but both devices did not enable until I installed this driver. The driver added devices too (some others failed, but it is OK). Here the output you requested:

hid_multitouch         24576  0
mac_hid                16384  0
hid_generic            16384  0
usbhid                 49152  0
i2c_hid                24576  0
hid                   122880  4 i2c_hid,usbhid,hid_multitouch,hid_generic

This output is the same as with or without the tablet. Therefore I am not sure it does help you.

Instead, look at this output (before drivers):
cat /var/log/Xorg.0.log

[344820.716] (II) config/udev: Adding input device LetSketch LetSketch (/dev/input/mouse4)
[344820.716] (II) No input driver specified, ignoring this device.
[344820.716] (II) This device may have been added with another device file.
[344820.829] (II) config/udev: Adding input device LetSketch LetSketch (/dev/input/event22)
[344820.829] (\*\*) LetSketch LetSketch: Applying InputClass "libinput tablet catchall"
[344820.829] (II) Using input driver 'libinput' for 'LetSketch LetSketch'
[344820.829] (\*\*) LetSketch LetSketch: always reports core events
[344820.829] (\*\*) Option "Device" "/dev/input/event22"
[344820.829] (\*\*) Option "_source" "server/udev"
[344820.888] (II) event22 - LetSketch LetSketch: is tagged by udev as: Tablet
[344820.888] (EE) event22 - LetSketch LetSketch: libinput bug: missing tablet capabilities: resolution.Ignoring this device.
[344820.888] (II) event22 - LetSketch LetSketch: device is a tablet
[344820.956] (II) event22 - failed to create input device '/dev/input/event22'.
[344820.956] (EE) libinput: LetSketch LetSketch: Failed to create a device for /dev/input/event22
[344820.956] (EE) PreInit returned 2 for "LetSketch LetSketch"
[344820.956] (II) UnloadModule: "libinput"

While this is the new output:

[ 31208.446] (II) config/udev: removing device LetSketch LetSketch stylus
[ 31208.455] (II) UnloadModule: "wacom"
[ 31208.455] (II) config/udev: removing device LetSketch LetSketch eraser
[ 31208.488] (II) UnloadModule: "wacom"
[ 36626.367] (II) config/udev: Adding input device LetSketch LetSketch (/dev/input/mouse3)
[ 36626.367] (II) No input driver specified, ignoring this device.
[ 36626.367] (II) This device may have been added with another device file.
[ 36626.480] (II) config/udev: Adding input device LetSketch LetSketch (/dev/input/event11)
[ 36626.480] (\*\*) LetSketch LetSketch: Applying InputClass "libinput tablet catchall"
[ 36626.480] (\*\*) LetSketch LetSketch: Applying InputClass "Tablet"
[ 36626.480] (II) Using input driver 'wacom' for 'LetSketch LetSketch'
[ 36626.480] (\*\*) LetSketch LetSketch: always reports core events
[ 36626.480] (\*\*) Option "Device" "/dev/input/event11"
[ 36626.540] (II) LetSketch LetSketch: type not specified, assuming 'stylus'.
[ 36626.540] (II) LetSketch LetSketch: other types will be automatically added.
[ 36626.540] (--) LetSketch LetSketch stylus: using pressure threshold of 851 for button 1
[ 36626.540] (\*\*) LetSketch LetSketch stylus: panscroll is 0
[ 36626.540] (\*\*) LetSketch LetSketch stylus: panscroll modified to 13
[ 36626.540] (--) LetSketch LetSketch stylus: maxX=32767 maxY=32767 maxZ=8191 resX=1016 resY=1016  tilt=enabled
[ 36626.540] (II) LetSketch LetSketch stylus: hotplugging dependent devices.
[ 36626.540] (EE) LetSketch LetSketch stylus: Invalid type 'cursor' for this device.
[ 36626.540] (EE) LetSketch LetSketch stylus: Invalid type 'touch' for this device.
[ 36626.540] (EE) LetSketch LetSketch stylus: Invalid type 'pad' for this device.
[ 36626.540] (II) LetSketch LetSketch stylus: hotplugging completed.
[ 36626.620] (\*\*) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.1/0003:6161:4D15.0018/input/input37/event11"
[ 36626.620] (II) XINPUT: Adding extended input device "LetSketch LetSketch stylus" (type: STYLUS, id 22)
[ 36626.620] (\*\*) LetSketch LetSketch stylus: (accel) keeping acceleration scheme 1
[ 36626.620] (\*\*) LetSketch LetSketch stylus: (accel) acceleration profile 0
[ 36626.621] (\*\*) LetSketch LetSketch stylus: (accel) acceleration factor: 2.000
[ 36626.621] (\*\*) LetSketch LetSketch stylus: (accel) acceleration threshold: 4
[ 36626.682] (\*\*) LetSketch LetSketch eraser: Applying InputClass "libinput tablet catchall"
[ 36626.682] (\*\*) LetSketch LetSketch eraser: Applying InputClass "Tablet"
[ 36626.682] (II) Using input driver 'wacom' for 'LetSketch LetSketch eraser'
[ 36626.682] (\*\*) LetSketch LetSketch eraser: always reports core events
[ 36626.682] (\*\*) Option "Device" "/dev/input/event11"
[ 36626.682] (\*\*) Option "Type" "eraser"
[ 36626.682] (\*\*) LetSketch LetSketch eraser: panscroll is 13
[ 36626.682] (\*\*) LetSketch LetSketch eraser: panscroll modified to 13
[ 36626.682] (--) LetSketch LetSketch eraser: maxX=32767 maxY=32767 maxZ=8191 resX=1016 resY=1016  tilt=enabled
[ 36626.716] (\*\*) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.1/0003:6161:4D15.0018/input/input37/event11"
[ 36626.716] (II) XINPUT: Adding extended input device "LetSketch LetSketch eraser" (type: ERASER, id 23)
[ 36626.717] (\*\*) LetSketch LetSketch eraser: (accel) keeping acceleration scheme 1
[ 36626.717] (\*\*) LetSketch LetSketch eraser: (accel) acceleration profile 0
[ 36626.717] (\*\*) LetSketch LetSketch eraser: (accel) acceleration factor: 2.000
[ 36626.717] (\*\*) LetSketch LetSketch eraser: (accel) acceleration threshold: 4

nic3-14159, 2021-03-14:
@ilmanowar Thanks for the response. The output of the lsmod command tells me that the tablet isn't using any of the drivers provided by DIGImend.

There might have been a misunderstanding when I referred to the generic hid driver. That driver is a kernel driver, and handles the actual data sent by the tablet over USB. DIGImend is also a kernel driver, supporting tablets that are not supported by the generic hid driver. However, the events from both of these kernel drivers eventually get passed to an X.org input driver, which is what the Wacom driver is referring to.

From the X.org output you posted, I think I know what is going on. Thanks for posting that. By default, X.org uses the libinput input driver. That is supported by the line `[344820.829] (II) Using input driver 'libinput' for 'LetSketch LetSketch'` in the old output. The README on the main GitHub page also says that libinput is used by default. When you install the deb package, the `xserver-xorg-input-wacom` package is also installed, which provides the Wacom input driver. That is what the line `[ 36626.480] (II) Using input driver 'wacom' for 'LetSketch LetSketch'` in the new output refers to.

So before the DIGImend deb package was installed, the tablet used the generic hid driver to handle the usb events, passing them to the libinput input driver. After installing the deb package, the tablet still uses the generic hid driver, but can now pass events to the Wacom input driver that happened to be installed alongside DIGImend, enabling both devices as you said.

If you uninstall DIGImend, but install `xserver-xorg-input-wacom` by itself and add your xorg.conf file as before, I suspect the tablet will still work.

ilmanowar, 2021-03-15:
@nic3-14159 What you wrote is very interesting:

I did remove **xserver-xorg-input-wacom** which triggered the removal of the **Digimend** drivers.
Then I checked the tablet before the reboot was working after the reboot it stopped.
Therefore I did reinstall it (without Digimend). And it worked (no need for a reboot).

If you believe that I could select another driver (which means no need for Wacom drivers) I should modify accordingly the xorg.conf. 

Maybe something like: 

Section "InputClass"
Identifier "Tablet"
Driver "libinput"
MatchDevicePath "/dev/input/event*"
MatchUSBID "6161:4d15"
EndSection

After removing the wacom driver and changing the xorg.conf file as above, rebooted, it does not work. More to try? Any other driver suggestion?

Therefore for now what works is: **xserver-xorg-input-wacom** + xorg.conf (with wacom as driver).

As a side effect since I did the first installation of the drivers, I have gedit that it is randomly flickering. Any ideas about that?

nic3-14159, 2021-03-15:
As far as I can tell, the only other xorg input driver that works with drawing tablets is the evdev driver. That is provided by the package `xserver-xorg-input-evdev`, and you would replace "wacom" with "evdev" in the xorg.conf file. I'm not familiar with the inner workings of the xorg input drivers, so I can't say why it doesn't work with libinput. The "X.org driver setup" section on the [support page of the DIGImend website](https://digimend.github.io/support/) has some details the wacom and evdev input drivers and some differences between them.

All that being said, I suspect a kernel driver might still need to be written in DIGImend in order for everything on the tablet to work properly. It seems like the 

As for the Gedit issue, I'm not sure about that.

mankarr1, 2021-08-31:
> Hello Massimo,
> It worked like a charm!
> Thank you, it did really help.

Could you please help  me , I too have bought the same tablet but I am unable to run on ubuntu 20.04

Mustafa-MS, 2021-11-04:
> > Hello Massimo,
> > It worked like a charm!
> > Thank you, it did really help.
> 
> Could you please help me , I too have bought the same tablet but I am unable to run on ubuntu 20.04

Check this out.
https://github.com/DIGImend/digimend-kernel-drivers/issues/528#issuecomment-817323860

omerbalkas, 2022-07-30:
mustafa reyiz ufak caplı bi acıklama yapar mısın aynı durumdan muzdaripim bi yardımcı olursan sevinirim bilgisayarım yavas oldugu için lubuntuya gectim pek de bilmiyorum acıkcası saygılar