DIGImend/digimend-kernel-drivers #518: Bosto 16HD - Could use some guidance on what to do

bonezinger, 2021-02-20:
Official product page - http://www.bostotablet.com/prod_view.aspx?TypeId=29&id=186&fid=t26:29:3
Driver page - http://www.bostotablet.com/down.aspx?FId=n14:14:8
Product page - https://aliexpress.com/item/33006442581.html

Interesting that it's listed as a WinMaxGroup Tablet Device with the `lsusb` command. Is there any other similar listed as such already implemented?

In any case, for ease of access, I forked the project just to upload the .txt files so it's easier to check the usb dumps:

https://github.com/bonezinger/digimend-kernel-drivers/blob/master/bosto16.txt
https://github.com/bonezinger/digimend-kernel-drivers/blob/master/hid_bosto16.txt
https://github.com/bonezinger/digimend-kernel-drivers/blob/master/hid_bosto_pencoords.txt
https://github.com/bonezinger/digimend-kernel-drivers/blob/master/hid_bosto_pentilt.txt
https://github.com/bonezinger/digimend-kernel-drivers/blob/master/hid_bosto_penpressura.txt
https://github.com/bonezinger/digimend-kernel-drivers/blob/master/hid_bosto_screenbuttons.txt

I am a complete beginner regarding drivers, but I have some knowledge of C and I'd like to help in this specific case. If anyone can give me some pointers on what to do or what files to look at, what might need changes for this tablet to work, I'd be very grateful!

GrandTreeWizard, 2021-02-20:
the drivers are compressed in a RAR format. I found that running them through 7zip got them going just fine tho any unzip program should work.

bonezinger, 2021-02-20:
> the drivers are compressed in a RAR format. I found that running them through 7zip got them going just fine tho any unzip program should work.

I know the official drivers are compacted with .rar, but there's none for Linux. I'm on Ubuntu 20.04, I forgot to mention that on the OP. Or did you install it with wine? Or is it possible to try and use the macOS drivers by extracting the .pkg file somewhere in the filesystem?

bonezinger, 2021-03-01:
After a bid of fiddling, I managed to figure some of the bytes in the stream of usbhid. When the pen is near the screen, the stream has 10 bytes:

00 11 22 33 44 55 66 77 88 99

- 00 - Defaults to 01. Dunno what it actually represents.
- 11 - Pen proximity and clicks. 00 = pen not near the screen, stop reading position; 10 = pen is near (reads position); 11 = pen is clicked; 30 = lower pen button pressed; 13 = this is a weird value. It reads this both for lower button + pen clicking, ignores pressure AND upper pen button. Clicking while holding the upper pen button keeps the value 13, but the pressure reads FF 1F
- 22, 33 - X position. Zero at left of the screen. Changes happen in the first byte, 22
- 44, 55 - Y position. Zero at top of the screen. Same as above
- 66, 77 - Pen pressure. Similar to XY position, the first byte is 66
- 88, 99 - Defaults to 00. Haven't managed to make it change

I could not figure out if there's anything for tilt, but at least from what I could gather from the stream, there might not be a pen tilt value. Note that on Windows, the driver also installs a utility to calibrate the pen, which acts as an offset for the XY axis.

The screen's top buttons generate a different stream of 8 bytes. The left and right buttons are probably mirrored, since, at least on the stream, there's no difference between pressing the leftmost button on the left or the right part.

00 11 22 33 44 55 66 77 - Only the pair 44 changes when any button is pressed. From left to right, it goes 01, 02, 04, 08, 10, 20, 40, 80. Pressing more than one at the same time sums the values. 00 means the button was released. The rest always reads 04 00 00 00 XX 00 00 00

pshobowale, 2021-03-05:
I am using it with Arch without a specific driver. I had the same concerns, but after some research I used evtest on /dev/input/eventX and found out that one matched with the tablet. So i just added it to my xorg config and started using it.

cat /etc/X11/xorg.conf.d/50-tablet.conf

Section "InputClass"
Identifier "Tablet"
Driver "wacom"
MatchDevicePath "/dev/input/event*"
MatchUSBID "0ed1:7821"
EndSection

bonezinger, 2021-03-16:
> I am using it with Arch without a specific driver. I had the same concerns, but after some research I used evtest on /dev/input/eventX and found out that one matched with the tablet. So i just added [...]

This was incredibly helpful, to be honest. Here's my little odyssey in helped me take:

- I first tested it by editing the `/etc/X11/xorg.conf` file, adding that to the end. Save and quit, didn't work.
- Created a folder, made the file, didn't work either.
- Manually test each of the `/dev/input/eventN` files to find out which, if any, corresponds to the tablet. At first, none of the listed files was for it.
- Google around trying to figure how `/dev/input` works, experiment evtest straight off and notice that 22 is listing a "BYX Tablet Device". It reads the pen input, but ignores the physical buttons.
- xinput command does not list anything remotely similar to that name above, or WinMax or anything else.
- I eventually came across this AU question - https://askubuntu.com/questions/1241381/graphics-tablet-shows-up-in-evtest-but-not-in-xinput
- The file 50-tablet.conf file is in `/usr/share/X11/xorg.conf.d/` for Ubuntu. Changed, save and quit, still no good out of the box, but it seems I need to restart/reboot my system first, something I still wasn't able to do. However, this got me hopeful, since it reads the input of the position and the pressure just fine.

I didn't have the chance to reboot my system yet, but I'm very hopeful this will be enough to make it work.

bonezinger, 2021-03-21:
Ok, after lots of trial and error, I managed to find the closest to ideal configs using the wacom driver.

```
Section "InputClass"
        Identifier "Bosto 16HD"
        MatchUSBID "0ed1:7821"
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
        Option "Area" "0 0 17910 10240"
        Option "Threshold" "1" #optional, should reduce the lag on pressing the pen
EndSection
```

Since the input event was going way over the actual max allowed for ABS_X, I toyed with xsetwacom parameters. Had to experiment a lot with the Area until I got that 17910 value, which is the most precise my eye could notice.

pshobowale, 2021-03-22:
I just recently tried wayland and therefore could not use Xorg. So I digged a little deeper for a udev hwdb file. Does someone know how to add the above mentioned calib?

pwd:
/lib/udev/hwdb.d

cat 61-evdev-local.hwdb:
evdev:input:b0003v0ED1p7821*

 EVDEV_ABS_00=0:17918:53
 EVDEV_ABS_01=0:10228:53

bonezinger, 2021-03-22:
> I just recently tried wayland and therefore could not use Xorg. So I digged a little deeper for a udev hwdb file. Does someone know how to add the above mentioned calib? Maybe we could contact the folks from systemd then to push it into the master branch. I could not figure out how to detect the lower button on the pen and the touch buttons on the pad itself. Has someone a solution to this yet?
> 
> pwd:
> /lib/udev/hwdb.d
> 
> cat 61-evdev-local.hwdb:
> evdev:input:b0003v0ED1p7821*
> 
> EVDEV_ABS_00=0:17918:53
> EVDEV_ABS_01=0:10228:53
> 
> Some helpful links for debugging:
> https://yulistic.gitlab.io/2017/12/linux-keymapping-with-udev-hwdb/
> https://wayland.freedesktop.org/libinput/doc/1.12.0/tablet-debugging.html
> https://manpages.debian.org/unstable/libevdev-tools/touchpad-edge-detector.1.en.html
> https://tldp.org/HOWTO/XFree-Local-multi-user-HOWTO/tweak_input_devs-realdev.html

Yeah, for now I've settled with just the screen and stylus working. Are you also using wacom drivers for yours?

Anyway, this is going to be a rather long post, as I'll be following those links and seeing what gets listed

Here's how the tablet showed up for me, following that first link
```bash
#sudo cat /proc/bus/input/devices

: Bus=0003 Vendor=0ed1 Product=7821 Version=0110
N: Name="BYX Tablet Device"
P: Phys=usb-0000:00:14.0-1/input0
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:0ED1:7821.0006/input/input25
U: Uniq=A02017120201
H: Handlers=event23 mouse4 
B: PROP=2
B: EV=1b
B: KEY=c03 3f 0 0 0 0
B: ABS=d000003
B: MSC=10
```
Notice the H, Handlers: it lists event23 (which handles mostly the input from the stylus) and mouse4. There is nothing that seems to indicate keys other than those in the pen, which I think makes sense, given that the packets listed on `usbhid-dump` for the physical buttons are 8 bytes instead of 10. Also, **none** of the events in /dev/input read any of the physical buttons at all, at least with wacom drivers. This is our main problem: the keys exist, they send data packets of 8 bytes, but that is not caught by wacom drivers or any other events. We can't map the inputs of the physical keys, at best we can map the expected wacom buttons to other things

Then, the lsusb using its bus:
```bash
#lsusb -v -s 001:006

Bus 001 Device 006: ID 0ed1:7821 WinMaxGroup Tablet Device
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0ed1 WinMaxGroup
  idProduct          0x7821 
  bcdDevice           35.76
  iManufacturer           1 BYX
  iProduct                2 Tablet Device
  iSerial                 3 A02017120201
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     242
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
```
I don't know if it's relevant the fact that wacom tablets use bus 003, and bosto uses bus 001.

I'm currently taking a look at how to make an input driver - https://www.kernel.org/doc/html/latest/input/input-programming.html , as it seems we'll really need this to get the physical keys working. I'll edit this post later if I find anything else worth mentioning

pshobowale, 2021-03-24:
Yes and no. For painting sessions I use it with my laptop and libwacom via X. As I wanna use it with my Pinephone I need it to 
support wayland and libinput, thats why I tested modifying udev with the same results as under libwacom and X. As you also figured out both methods don't recognize the padbuttons and the first stylus button. Something I just noted recently was that a device is created in /dev/usb/hiddev0 which corresponds to the data that is send from the pad, maybe this could be a good starting point for further investigation.

If you are interessted we could join forces to produce a kernel module for the missing buttons, after my exams in three weeks.

bonezinger, 2021-03-24:
It's not that the lower pen button isn't caught, it's that it only really works when the pen is also pressed, and it behaves in the same way as the upper button being pressed, the bytestream for "upper pen button" and "lower pen button + pen pressed" is the same value, you can check that on my 2nd reply.

I'm all up for trying to write something to capture the screen's button inputs, though the first problem I have to fix on my end is figuring why the hell gcc can't find the headers linux/init.h, asm/irq.h and asm/io.h, all listed in the example given in the kernel's documentation for input drivers: https://www.kernel.org/doc/html/latest/input/input-programming.html?highlight=input

Trying to hack something into the wacom drivers or this other git for older bosto tablets, https://github.com/aidyw/bosto-2g-linux-kernel-module, might be easier and faster, too.

nic3-14159, 2021-03-24:
@bonezinger Just making sure I understand what you mean with the pen buttons:

- Pen not touching tablet, no buttons pressed --> 10
- Pen touching tablet, no buttons pressed --> 11
- Pen not touching tablet, lower button pressed --> 30
- Pen not touching tablet, upper button pressed --> 13
- Pen touching tablet, lower button pressed --> 13, and ignores pressure (I assume that means pressure reads 0000?)
- Pen touching tablet, upper button pressed --> 13, and pressure reads FF1F

Is this correct?

pshobowale, 2021-03-24:
You can not use gcc https://tldp.org/LDP/lkmpg/2.6/html/x181.html

bonezinger, 2021-03-25:
> @bonezinger Just making sure I understand what you mean with the pen buttons:
>
>     * Pen not touching tablet, no buttons pressed --> 10
>
>     * Pen touching tablet, no buttons pressed --> 11
>
>     * [...]

Yes, that is how `usbhid-dump` shows the byte changing for each situation. And for the lower button pressed + pen touching the pressure indeed remains 00 00, even if you try pressing the pen harder.

@pshobowale is that still valid? Asking because that documentation is from 2007, kernel 2.6

dancingfoot, 2021-04-16:
Thanks in advance the wacom driver do work..
Regarding the Multi Touch ...have you manage to get it working?

ninahdb, 2021-12-22:
Hi, I am new to Linux and I wanted to ask you how did you made your pen work, I have the same Bosto tablet and I have tried everything but only the screen works... maybe you could point me in the right direction

pshobowale, 2021-12-23:
create a custom udevrule under:

nano /lib/udev/hwdb.d/61-evdev-local.hwdb

put this into it and restart:

evdev:input:b0003v0ED1p7821*

EVDEV_ABS_00=0:17918:53
EVDEV_ABS_01=0:10228:53

pshobowale, 2021-12-23:
It basically tells the os the resolution of the drawing area if it detects the correct id, when i remember correctly. So that it is not dropped during init. If you are using X you could also create a XOrg Config as mentioned above

pshobowale, 2021-12-23:
> Thanks in advance the wacom driver do work.. Regarding the Multi Touch ...have you manage to get it working?

Can't test it as i only have the pen-only version (16 HD). I kind of guess that it should be a rather similar problem as with the pen. Probably just a missing config, so just try evtest and look if there is some kind of multitouch event if you use two fingers.

thehamstermuffin, 2022-07-21:
> Ok, after lots of trial and error, I managed to find the closest to ideal configs using the wacom driver. The file to edit is `/usr/share/X11/xorg.conf.d/50-digimend.conf`. Might not even need digimend installed, you can add the following to 70-wacom.conf file instead, but only do it in ONE conf file.
> 
> ```
> Section "InputClass"
>         Identifier "Bosto 16HD"
>         MatchUSBID "0ed1:7821"
>         MatchDevicePath "/dev/input/event*"
>         Driver "wacom"
>         Option "Area" "0 0 17910 10240"
>         Option "Threshold" "1" #optional, should reduce the lag on pressing the pen
> EndSection
> ```
> 
> Since the input event was going way over the actual max allowed for ABS_X, I toyed with xsetwacom parameters. Had to experiment a lot with the Area until I got that 17910 value, which is the most precise my eye could notice.
> 
> EDIT: Just noticed that it didn't immediately take effect on plugging the tablet. In that case, you have to manually set those Options:
> 
> ```shell
> #Check how the device is named in xsetwacom and correct as needed. The bosto will probably end listed like mine, BYX
> #As a side note, experiment with different values instead of 17910 if your screen is not 1920x1080
> xsetwacom --list devices
> xsetwacom --set "BYX Tablet Device stylus" Threshold 1
> xsetwacom --set "BYX Tablet Device stylus" Area 0 0 17910 10240
> ```

This is what worked for me with just one change. For some reason mine appears as three devices: 
```
奂X 楄楧楴敺r stylus            id: 14  type: STYLUS    
奂X 楄楧楴敺r eraser            id: 15  type: ERASER    
奂X 楄楧楴敺r pad               id: 16  type: PAD    
```
- The area needs to be set on the stylus
- The eraser doesn't seem to do anything

But when working with several monitors it tries to map the whole screens suface to the tablet, instead of just one screen

Rommancen, 2024-11-13:
Этот текст описывает обсуждение и обмен опытом пользователей о том, как заставить планшет Bosto 16HD работать на Linux, в частности, на Ubuntu и Arch. 

Пользователь **bonezinger** столкнулся с трудностями при подключении Bosto 16HD к системе на Ubuntu 20.04, так как официальные драйверы были доступны только для Windows, и у него не было нативного драйвера для Linux. Он попытался создать рабочий драйвер, изучив данные, поступающие с устройства, и записывая байтовые потоки, отражающие положения пера, давление и нажатие кнопок. Несмотря на это, он отметил, что данные о наклоне пера не фиксируются, что затрудняет полное использование возможностей планшета.

Также bonezinger поделился файлами, в которых зафиксированы данные с разных элементов планшета, таких как положение пера и кнопки экрана. Ему помогали другие участники, например **GrandTreeWizard**, предложивший использовать распаковщики файлов, и **pshobowale**, поделившийся опытом настройки устройства на Arch с помощью драйвера Wacom и файла конфигурации xorg.conf.d.

В ходе многочисленных тестов bonezinger обнаружил, что нужно добавлять устройство в конфигурации X11 и следить за правильным подключением к USB-порту через evtest. Он пришел к выводу, что для корректной работы требуется перезагрузка системы. Тем не менее, после перезагрузки экран планшета отображался некорректно – изображение оказывалось растянутым.

Эта история отражает типичный путь Linux-энтузиаста, который упорно пробует разные подходы для настройки своего устройства без официальной поддержки! 

---

**Хэштеги:** #Linux #Ubuntu #ArchLinux #BostoTablet #НастройкаПланшета #OpenSource #Технологии

Rommancen, 2024-11-13:
Просто смотрю как это работает)