DIGImend/digimend-kernel-drivers #519: Weird issues with Huion H640P

Mek101, 2021-02-23:
OS: Linux Mint 20.1
Kernel: 5.8.0-43-generic
Install source: dkms deb, version 10

After installing the deb, I rebooted:

- First reboot: the tablet works as expected, a bit too sensible, but that's it. My installation however suffers from a problem where sometimes the nvidia gpu drivers aren't loaded correctly, which I usually solve with a reboot.
- Second reboot: the nvidia drivers aren't loaded again, and the tabled doesn't work anymore
- Third and fourth reboot same as second.

I tried to do some troubleshooting: the led on the device lights on (supposedly meaning it's correctly powered), however `lsusb` doesn't show any device.
So I inspected `dmesg`, and this logs pop up every time I try to attach the tablet
```
[ 1257.591378] usb 3-1.4: new full-speed USB device number 25 using xhci_hcd
[ 1257.671010] usb 3-1.4: device descriptor read/64, error -71
[ 1257.859423] usb 3-1.4: device descriptor read/64, error -71
[ 1258.046995] usb 3-1.4: new full-speed USB device number 26 using xhci_hcd
[ 1258.127009] usb 3-1.4: device descriptor read/64, error -71
[ 1258.315007] usb 3-1.4: device descriptor read/64, error -71
[ 1258.423083] usb 3-1-port4: attempt power cycle
[ 1259.027123] usb 3-1.4: new full-speed USB device number 27 using xhci_hcd
[ 1259.027205] usb 3-1.4: Device not responding to setup address.
[ 1259.235058] usb 3-1.4: Device not responding to setup address.
[ 1259.938978] usb 3-1.4: device not accepting address 28, error -71
[ 1259.939081] usb 3-1-port4: unable to enumerate USB device

```

Mek101, 2021-02-24:
Same issue on Manjaro 20.2.1, kernel 5.10.15 and Linux Mint with kernel 5.4

nic3-14159, 2021-02-24:
Hello! I found this post describing a potential solution to those errors. Perhaps try this? https://urukrama.wordpress.com/2009/01/27/usb-drive-not-recognised-error-71/

Mek101, 2021-02-24:
Thank you, it seems like it solved the issue: the tablet now works!
***
For anyone in the future, if the link has gone down: it's an old blog post from 2009, where the blogger solved the problem by scraping on an even older [mail list](http://www.spinics.net/lists/usb/msg02644.html). Basically you have to create `/etc/modprobe.d/options` (can you name it differently? I think so, but dunno) and put `options usbcore use_both_schemes=y` inside it, then reboot

Mek101, 2021-02-24:
Apparently only for a moment: I installed Krita via flathub, rebooted, and I'm back at the beginning

Mek101, 2021-02-24:
Closing again because, after a couple of tries (even without the `options usbcore use_both_schemes=y`), apparently I just have been sent a bad cable.

closed 2021-02-24 by Mek101