DIGImend/digimend-kernel-drivers #521: Can't bind Touch Strip actions on Huion HS611 with xsetwacom.

I-dont-need-name, 2021-02-28:
So I installed digimend kernel drivers from here (sudo make dkms_install) and Everything works fine, except for two buttons on the right (I don't really care about those) and touch strip. `Xsetwacom list` out is:
```
$ xsetwacom list
HUION Huion Tablet_HS611 stylus         id: 14  type: STYLUS    
HUION Huion Tablet_HS611 Pad pad        id: 15  type: PAD       
HUION Huion Tablet_HS611 Touch Strip pad        id: 16  type: PAD 
```
I'm trying to bind some keys to AbsWheelUp:
`xsetwacom set "HUION Huion Tablet_HS611 Touch Strip pad" AbsWheelUp key +Ctrl +Z`
But touch strip still doesn't react.
`evtest` output is:
<details>
<summary> evtest (touching from left to right) </summary>

```
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6d version 0x111
Input device name: "HUION Huion Tablet_HS611 Touch Strip"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 256 (BTN_0)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max        1
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max        1
    Event code 8 (ABS_WHEEL)
      Value      0
      Min        0
      Max        7
      Resolution       3
    Event code 40 (ABS_MISC)
      Value      0
      Min        0
      Max      255
Properties:
Testing ... (interrupt to exit)
Event: time 1614519964.992699, type 3 (EV_ABS), code 40 (ABS_MISC), value 15
Event: time 1614519964.992699, -------------- SYN_REPORT ------------
Event: time 1614519967.162703, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 1
Event: time 1614519967.162703, -------------- SYN_REPORT ------------
Event: time 1614519969.172706, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 2
Event: time 1614519969.172706, -------------- SYN_REPORT ------------
Event: time 1614519970.460703, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 3
Event: time 1614519970.460703, -------------- SYN_REPORT ------------
Event: time 1614519971.628704, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 4
Event: time 1614519971.628704, -------------- SYN_REPORT ------------
Event: time 1614519972.954704, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 5
Event: time 1614519972.954704, -------------- SYN_REPORT ------------
Event: time 1614519974.280706, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 6
Event: time 1614519974.280706, -------------- SYN_REPORT ------------
Event: time 1614519975.750709, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 7
Event: time 1614519975.750709, -------------- SYN_REPORT ------------
Event: time 1614519977.182708, type 3 (EV_ABS), code 40 (ABS_MISC), value 0
Event: time 1614519977.182708, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 0
Event: time 1614519977.182708, -------------- SYN_REPORT ------------
```
</details>

azratul, 2021-03-14:
Exactly the same problem here, but I can't get "HUION Huion Tablet_HS611 Consumer Control" and "HUION Huion Tablet_HS611 Keyboard" to work either

Brandon-Beck, 2021-08-28:
I believe this, and a few other reports are all due to libinput not liking something about the devices we create, and deciding to ignore it.
```
event10 - HUION Huion Tablet_HS611 Touch Strip: libinput bug: missing tablet capabilities: pen btn-stylus resolution. Ignoring this device.
```
The touch-strip, and keyboard (2 buttons on the right) are not themselves full-blown tablets, and thus probably don't have/need the pen, btn-stylus, or resolution capabilities alone. But it seems libinput thinks they are and thus that they need it. Since they don't provide it, it marks it as a buggy device and discards their input. 

Output of `libinput record` makes me think the driver is otherwise functioning, its just libinput isn't happy and decides to throw away the bugged device/driver's events. I'm not sure if your supposed to remove the ID_INPUT_TABLET property or set the bits for the pen, btn-stylus and add the resolution.... I'd hope the former as the latter doesn't make sense to me for a virtual keyboard and scroll device, but maybe its required for a tablet device group? (idk, not a hardware/driver dev, just guessing)

Basing guess on what I understood from https://wayland.freedesktop.org/libinput/doc/1.12.0/tablet-debugging.html

<details>
<summary><b>libinput record - HUION Huion Tablet_HS611 Keyboard.</b>
First pressed, file button (second rightmost button). Then pressed the rightmost button (idk, some window or workspace size button)</summary>
```
# libinput record
version: 1
ndevices: 1
libinput:
  version: "1.18.1"
  git: "unknown"
system:
kernel: "5.13.13-arch1-1"
dmi: "dmi:bvnLENOVO:bvrR12ET57W(1.27):bd11/24/2020:br1.27:efr1.27:svnLENOVO:pn20NJ0007US:pvrThinkPadT495:skuLENOVO_MT_20NJ_BU_Think_FM_ThinkPadT495:rvnLENOVO:rn20NJ0007US:rvrNotDefined:cvnLENOVO:ct10:cvrNone:"
devices:
- node: /dev/input/event23
  evdev:
    # Name: HUION Huion Tablet_HS611 Keyboard
    # ID: bus 0x3 vendor 0x256c product 0x6d version 0x111
    # Supported Events:
    # Event type 0 (EV_SYN)
    # Event type 1 (EV_KEY)
    #   Event code 1 (KEY_ESC)
    #   Event code 2 (KEY_1)
    #   Event code 3 (KEY_2)
    #   Event code 4 (KEY_3)
    #   Event code 5 (KEY_4)
    #   Event code 6 (KEY_5)
    #   Event code 7 (KEY_6)
    #   Event code 8 (KEY_7)
    #   Event code 9 (KEY_8)
    #   Event code 10 (KEY_9)
    #   Event code 11 (KEY_0)
    #   Event code 12 (KEY_MINUS)
    #   Event code 13 (KEY_EQUAL)
    #   Event code 14 (KEY_BACKSPACE)
    #   Event code 15 (KEY_TAB)
    #   Event code 16 (KEY_Q)
    #   Event code 17 (KEY_W)
    #   Event code 18 (KEY_E)
    #   Event code 19 (KEY_R)
    #   Event code 20 (KEY_T)
    #   Event code 21 (KEY_Y)
    #   Event code 22 (KEY_U)
    #   Event code 23 (KEY_I)
    #   Event code 24 (KEY_O)
    #   Event code 25 (KEY_P)
    #   Event code 26 (KEY_LEFTBRACE)
    #   Event code 27 (KEY_RIGHTBRACE)
    #   Event code 28 (KEY_ENTER)
    #   Event code 29 (KEY_LEFTCTRL)
    #   Event code 30 (KEY_A)
    #   Event code 31 (KEY_S)
    #   Event code 32 (KEY_D)
    #   Event code 33 (KEY_F)
    #   Event code 34 (KEY_G)
    #   Event code 35 (KEY_H)
    #   Event code 36 (KEY_J)
    #   Event code 37 (KEY_K)
    #   Event code 38 (KEY_L)
    #   Event code 39 (KEY_SEMICOLON)
[... 165 lines cut ...]
    - [  0,      0,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +0ms
  - evdev:
    - [  0, 112540,   4,   4,      30] # EV_MSC / MSC_SCAN                 30 (obfuscated)
    - [  0, 112540,   1, 125,       0] # EV_KEY / KEY_LEFTMETA              0
    - [  0, 112540,   4,   4,      30] # EV_MSC / MSC_SCAN                 30 (obfuscated)
    - [  0, 112540,   1,  15,       0] # EV_KEY / KEY_TAB                   0
    - [  0, 112540,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +112ms
  # Current time is 12:14:40
  # Current time is 12:14:49
  - evdev:
    - [  9, 832464,   4,   4,      30] # EV_MSC / MSC_SCAN                 30 (obfuscated)
    - [  9, 832464,   1, 125,       1] # EV_KEY / KEY_LEFTMETA              1
    - [  9, 832464,   4,   4,      30] # EV_MSC / MSC_SCAN                 30 (obfuscated)
    - [  9, 832464,   1,  30,       1] # EV_KEY / KEY_A                     1 (obfuscated)
    - [  9, 832464,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +9720ms
  - evdev:
    - [  9, 908013,   4,   4,      30] # EV_MSC / MSC_SCAN                 30 (obfuscated)
    - [  9, 908013,   1, 125,       0] # EV_KEY / KEY_LEFTMETA              0
    - [  9, 908013,   4,   4,      30] # EV_MSC / MSC_SCAN                 30 (obfuscated)
    - [  9, 908013,   1,  30,       0] # EV_KEY / KEY_A                     0 (obfuscated)
    - [  9, 908013,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +76ms
  # Current time is 12:14:50
```
</details>
Seemingly related #536

I-dont-need-name, 2021-08-31:
> I believe this, and a few other reports are all due to libinput not liking something about the devices we create, and deciding to ignore it.

I managed to solve this by adding ID_INPUT_TABLET_PAD=1 and then libinput doesn't seem to give any errors, but it still doesn't work.

albfan, 2022-01-26:
Hi, I just check strip events (looks there're 8 segments) and they are exposed as combinations of buttons.

last buttons from pad are 8 and 9, those are shown to in strip, and then combining 8,9,10,and 11 it represents this 8 segments:

https://user-images.githubusercontent.com/220968/151236282-2bea8d76-4973-46c4-9922-8a1b76bbcddb.mp4

that gesture is recorded in:

    sudo libinput record /dev/input/event18 huion-hs611-strip.yml

albfan, 2022-01-26:
about two upper right buttons:

```
$ sudo libinput list-devices
event19 - HUION Huion Tablet_HS611 Keyboard: libinput bug: missing tablet capabilities: xy pen btn-stylus resolution. Ignoring this device.
```

I checked with evtest and this is where those buttons are reported (as Alt+tab and Super+D) which are shortcut on windows for change window and show desktop

```
$ sudo evtest /dev/input/event19
Event: time 1643227147.441050, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e3
Event: time 1643227147.441050, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 1
Event: time 1643227147.441050, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7002b
Event: time 1643227147.441050, type 1 (EV_KEY), code 15 (KEY_TAB), value 1
Event: time 1643227147.441050, -------------- SYN_REPORT ------------
Event: time 1643227147.587013, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e3
Event: time 1643227147.587013, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 0
Event: time 1643227147.587013, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7002b
Event: time 1643227147.587013, type 1 (EV_KEY), code 15 (KEY_TAB), value 0
Event: time 1643227147.587013, -------------- SYN_REPORT ------------

Event: time 1643227149.415181, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e3
Event: time 1643227149.415181, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 1
Event: time 1643227149.415181, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70007
Event: time 1643227149.415181, type 1 (EV_KEY), code 32 (KEY_D), value 1
Event: time 1643227149.415181, -------------- SYN_REPORT ------------
Event: time 1643227149.591051, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e3
Event: time 1643227149.591051, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 0
Event: time 1643227149.591051, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70007
Event: time 1643227149.591051, type 1 (EV_KEY), code 32 (KEY_D), value 0
Event: time 1643227149.591051, -------------- SYN_REPORT ------------
```

albfan, 2022-01-27:
To compare, I get this reports of what should be a valid report from a strip (not generated with my tabler or my environment

libinput:

```
-event1   DEVICE_ADDED            Power Button                      seat0 default group1  cap:k
-event0   DEVICE_ADDED            Power Button                      seat0 default group2  cap:k
-event3   DEVICE_ADDED            Wacom Cintiq 12WX Pen             seat0 default group3  cap:T  size 265x167mm calib
-event5   DEVICE_ADDED            Wacom Cintiq 12WX Pad             seat0 default group3  cap:P buttons:10 strips:2 rings:0 mode groups:1
-event2   DEVICE_ADDED            PixArt USB Optical Mouse          seat0 default group4  cap:p left scroll-nat scroll-button
-event4   DEVICE_ADDED            HID 046a:0011                     seat0 default group5  cap:k
-event5   TABLET_PAD_STRIP        +0.000s   strip 0 position 1.00 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +0.162s   strip 0 position 0.92 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +0.242s   strip 0 position 0.83 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +0.262s   strip 0 position 0.75 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +0.302s   strip 0 position 0.67 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +0.402s   strip 0 position 0.58 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +0.482s   strip 0 position 0.50 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +0.502s   strip 0 position 0.42 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +0.602s   strip 0 position 0.33 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +0.642s   strip 0 position 0.25 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +0.722s   strip 0 position 0.17 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +0.763s   strip 0 position 0.08 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +0.823s   strip 0 position 0.00 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +1.205s   strip 0 position 0.08 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +1.225s   strip 0 position 0.17 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +1.286s   strip 0 position 0.25 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +1.306s   strip 0 position 0.33 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +1.386s   strip 0 position 0.42 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +1.426s   strip 0 position 0.50 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +1.506s   strip 0 position 0.58 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +1.527s   strip 0 position 0.67 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +1.627s   strip 0 position 0.75 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +1.647s   strip 0 position 0.83 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +1.728s   strip 0 position 0.92 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +1.808s   strip 0 position 1.00 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +1.867s   strip 0 position -1.00 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +1.988s   strip 0 position 1.00 (source finger) (mode 0)
 event5   TABLET_PAD_STRIP        +2.008s   strip 0 position -1.00 (source finger) (mode 0)
```

or evdev:

```
# EVEMU 1.3
# Kernel: 5.15.16-200.fc35.x86_64
# DMI: dmi:bvnAmericanMegatrendsInc.:bvrP3.60:bd07/31/2019:br5.14:svnToBeFilledByO.E.M.:pnToBeFilledByO.E.M.:pvrToBeFilledByO.E.M.:rvnASRock:rnB450MPro4:rvr:cvnToBeFilledByO.E.M.:ct3:cvrToBeFilledByO.E.M.:skuToBeFilledByO.E.M.:
# Input device name: "Wacom Cintiq 12WX Pad"
# Input device ID: bus 0x03 vendor 0x56a product 0xc6 version 0x100
# Supported events:
#   Event type 0 (EV_SYN)
#     Event code 0 (SYN_REPORT)
#     Event code 1 (SYN_CONFIG)
#     Event code 2 (SYN_MT_REPORT)
#     Event code 3 (SYN_DROPPED)
#     Event code 4 ((null))
#     Event code 5 ((null))
#     Event code 6 ((null))
#     Event code 7 ((null))
#     Event code 8 ((null))
#     Event code 9 ((null))
#     Event code 10 ((null))
#     Event code 11 ((null))
#     Event code 12 ((null))
#     Event code 13 ((null))
#     Event code 14 ((null))
#     Event code 15 (SYN_MAX)
#   Event type 1 (EV_KEY)
#     Event code 256 (BTN_0)
#     Event code 257 (BTN_1)
[... 85 lines cut ...]
E: 0.241008 0000 0000 0000  # ------------ SYN_REPORT (0) ---------- +40ms
E: 0.281029 0003 0003 0512  # EV_ABS / ABS_RX               512
E: 0.281029 0000 0000 0000  # ------------ SYN_REPORT (0) ---------- +40ms
E: 0.341034 0003 0003 0256  # EV_ABS / ABS_RX               256
E: 0.341034 0000 0000 0000  # ------------ SYN_REPORT (0) ---------- +60ms
E: 0.401041 0003 0003 0128  # EV_ABS / ABS_RX               128
E: 0.401041 0000 0000 0000  # ------------ SYN_REPORT (0) ---------- +60ms
E: 0.501041 0003 0003 0064  # EV_ABS / ABS_RX               64
E: 0.501041 0000 0000 0000  # ------------ SYN_REPORT (0) ---------- +100ms
E: 0.561062 0003 0003 0032  # EV_ABS / ABS_RX               32
E: 0.561062 0000 0000 0000  # ------------ SYN_REPORT (0) ---------- +60ms
E: 0.621070 0003 0003 0016  # EV_ABS / ABS_RX               16
E: 0.621070 0000 0000 0000  # ------------ SYN_REPORT (0) ---------- +60ms
E: 0.682078 0003 0003 0008  # EV_ABS / ABS_RX               8
E: 0.682078 0000 0000 0000  # ------------ SYN_REPORT (0) ---------- +61ms
[... cut ...]
E: 0.762088 0000 0000 0000  # ------------ SYN_REPORT (0) ---------- +80ms
E: 0.823095 0003 0003 0002  # EV_ABS / ABS_RX               2
E: 0.823095 0000 0000 0000  # ------------ SYN_REPORT (0) ---------- +61ms
E: 0.883090 0003 0003 0001  # EV_ABS / ABS_RX               1
E: 0.883090 0000 0000 0000  # ------------ SYN_REPORT (0) ---------- +60ms
E: 1.330169 0003 0003 0000  # EV_ABS / ABS_RX               0
E: 1.330169 0003 0028 0000  # EV_ABS / ABS_MISC             0
E: 1.330169 0000 0000 0000  # ------------ SYN_REPORT (0) ---------- +447ms
 ```