DIGImend/digimend-kernel-drivers #525: I can't use H640P in Gnome (Mint 20)

JumperLuko, 2021-03-29:
I installed the v10 driver via deb, but I may be being stupid with that.
I can't configure the table, with or without the driver, I didn't feel any difference.
The tablet even works in a simple way on krita for example.

[image]

[image]

JumperLuko, 2021-03-29:
I tested it on the desktop plasma and didn't recognize it either.
In Cinnamon I can configure pressure and pen buttons as a generic device

[image]

nic3-14159, 2021-03-29:
Does your tablet show up in the output of the command `xsetwacom list`?

JumperLuko, 2021-03-29:
```
$ xsetwacom list
HID 256c:006d Pen stylus        	id: 10	type: STYLUS    
HID 256c:006d Pad pad           	id: 11	type: PAD  
```

JumperLuko, 2021-03-29:
If I install the kde-config-tablet package (`sudo apt install kde-config-tablet`), enter the desktop plasma and open the program, I can set the 6 buttons on the tablet there.
But I can only set what is what, I cannot define functions for these buttons, because the normal KDE settings panel still doesn't show the options.

Here is an example of a [Video on Youtube](https://youtu.be/_V79I3EKCvc?t=139) where the normal KDE settings work normally, for me this region is very empty.

JumperLuko, 2021-03-29:
I tried it now on another computer with Mint 20, and the same result.

JumperLuko, 2021-03-29:
Searching and trying hard, I got the [issue 360](https://github.com/DIGImend/digimend-kernel-drivers/issues/360) layout files and something worked, you need to be able to configure the buttons now, and adjust pressure through Gnome. I believe that if you only have a good layout file for now, can you do that for me?

I believe that this should come in the installation of the drivers and ended up being forgotten.

I will be without the pen for a few hours, but as soon as I am home I will test it.

[image]

nic3-14159, 2021-03-29:
I don't think the layout files were forgotten in the install, as it seems like the DIGImend project leaves that to the libwacom project. I guess not every tablet supported by DIGImend has been added to libwacom. I suppose it's done to limit the scope of the kernel driver, and to keep the scope of different projects such as DIGImend and linuxwacom separate and well organized.

I've been experimenting with those layout files with my tablet, but I haven't been able to get it fully working. Once I figure it out I can try and see if I can get one working for your tablet.

In the meantime, you should be able to configure the tablet using `xsetwacom`
Here's one tutorial I found for how to use it:
https://joshuawoehlke.com/wacom-intuos-and-xsetwacom-on-ubuntu-18-04/

JumperLuko, 2021-03-29:
Its help a lot.
Buttons now works!

JumperLuko, 2021-03-30:
> I don't think the layout files were forgotten in the install, as it seems like the DIGImend project leaves that to the libwacom project. I guess not every tablet supported by DIGImend has been added [...]

how I capture the buttons I type, in case I want a strange button?
For example to hit me to trigger the ctrl

JumperLuko, 2021-03-30:
I get something
`$ xsetwacom --list devices`

and [this link](https://www.linux-magazine.com/Issues/2018/212/xsetwacom)

nic3-14159, 2021-03-30:
Each of the buttons has a number. Since the H640P has 6 buttons, they are numbered 1,2,3,8,9, and 10 (for technical reasons, xsetwacom always skips 4 through 7). Each one can be mapped using the command `xsetwacom set {DEVICE} Button {number} "key {keys you want to press}"`.
Replace {DEVICE} with the id of the tablet pad (the id is in the output of `xsetwacom --list devices`, which is the same as `xsetwacom list`). Alternatively, you can replace it with the name of the pad interface as it shows up in `xsetwacom list`, using quotes around the name.
Replace {number} with the number of the button you want to modify.
Replace {keys you want to press} with a list of keys to be pressed.

For example, if you wanted to map button 1 to "ctrl alt m", you could do
`xsetwacom set 11 Button 1 "key ctrl alt m"` (11 is the id of the pad when in the output of `xsetwacom list` you posted before)
or
`xsetwacom set "HID 256c:006d Pad pad" Button 1 "key ctrl alt m"`

As far as I know, there isn't really a way to capture the keystrokes as you type them, but the way you specify keys in xsetwacom is pretty straightforward

JumperLuko, 2021-03-30:
progress and problems

After playing the layout files in the system folders, the pen settings and some of the table settings appeared, work.
[image]
[image]

The problem is the buttons on the tablet, yesterday I was using the command line to set the buttons, which is not ideal but it was working, today I went to turn it on and it just doesn't work, I don't know why.
OMG
While writing this I found that when the button is turned, it doesn't work ...
That is, my mistake

nic3-14159, 2021-03-30:
> Now with layout files in the right place, I checked in KDE and Cinnamon, both have mapping of the 6 buttons just right.
> Then I realize that missing the interface is a problem for Gnome, looking for solutions ...

It seems like the missing interface for mapping the buttons could possibly be a bug in GNOME:
https://gitlab.gnome.org/GNOME/gnome-shell/-/issues/3926

JumperLuko, 2021-04-01:
> Each of the buttons has a number. Since the H640P has 6 buttons, they are numbered 1,2,3,8,9, and 10 (for technical reasons, xsetwacom always skips 4 through 7). Each one can be mapped using the command `xsetwacom set {DEVICE} Button {number} "key {keys you want to press}"`.
Work well, every time I started I had to configure the settings. I even created a script that gives way to make it work
https://github.com/JumperLuko/Set-Buttons-Wacon-Tablet-Pad

And now I'm using it, it's great, a way of not depending on the command line.
https://github.com/sezanzeb/key-mapper

with that the settings are saved when starting the system, and the settings are applied even on the login screen 😮
has presets, which is great when using multiple programs.