DIGImend/digimend-kernel-drivers #528: Digimend problem with letsketch WP9620

Bhanu-tarun, 2021-04-11:
@ilmanowar please fix this for me im new to linux

im using a wp9620 tablet rebranded as Case U graphics tablet 
i have an issue with the process of drivers, I like linux and dont want to move away from it but this driver incompatibility may force me to leave linux. i have made up my mind to use applications that are available in linux for my animation workflow but the driver incompatibility is bad. please walk me through the process.

here is the output of the command cat /var/log/Xorg.0.log

[    14.784] 
X.Org X Server 1.20.8
X Protocol Version 11, Revision 0
[    14.784] Build Operating System: Linux 4.19.0-8-amd64 x86_64 Debian
[    14.784] Current Operating System: Linux Uutax 5.7.0-kali1-amd64 #1 SMP Debian 5.7.6-1kali2 (2020-07-01) x86_64
[    14.784] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.7.0-kali1-amd64 root=UUID=1a564cb9-a708-4ae4-a354-1e775775a7ca ro quiet splash
[    14.784] Build Date: 31 March 2020  10:14:40AM
[    14.784] xorg-server 2:1.20.8-2 (https://www.debian.org/support) 
[    14.784] Current version of pixman: 0.36.0
[    14.784]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    14.784] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    14.784] (==) Log file: "/var/log/Xorg.0.log", Time: Sun Apr 11 11:20:30 2021
[    14.834] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    14.862] (==) No Layout section.  Using the first Screen section.
[    14.862] (==) No screen section available. Using defaults.
[    14.862] (**) |-->Screen "Default Screen Section" (0)
[    14.862] (**) |   |-->Monitor "<default monitor>"
[    14.863] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[    14.863] (==) Automatically adding devices
[    14.863] (==) Automatically enabling devices
[    14.863] (==) Automatically adding GPU devices
[    14.863] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    14.884] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    14.884]    Entry deleted from font path.
[    14.889] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        /usr/share/fonts/X11/100dpi/:unscaled,
        /usr/share/fonts/X11/75dpi/:unscaled,
        /usr/share/fonts/X11/Type1,
        /usr/share/fonts/X11/100dpi,
        /usr/share/fonts/X11/75dpi,
        built-ins
[    14.889] (==) ModulePath set to "/usr/lib/xorg/modules"
[    14.889] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[    14.889] (II) Loader magic: 0x55f9dadc8e20
[    14.889] (II) Module ABI versions:
[    14.889]    X.Org ANSI C Emulation: 0.4
[    14.889]    X.Org Video Driver: 24.1
[    14.889]    X.Org XInput driver : 24.1
[    14.889]    X.Org Server Extension : 10.0
[    14.890] (++) using VT number 7

[    14.891] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    14.892] (II) xfree86: Adding drm device (/dev/dri/card0)
[    14.910] (--) PCI:*(0@0:2:0) 8086:3e91:1028:0995 rev 0, Mem @ 0xf6000000/16777216, 0xe0000000/268435456, I/O @ 0x0000f000/64, BIOS @ 0x????????/131072
[    14.910] (II) LoadModule: "glx"
[    14.911] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    15.049] (II) Module glx: vendor="X.Org Foundation"
[    15.049]    compiled for 1.20.8, module version = 1.0.0
[    15.049]    ABI class: X.Org Server Extension, version 10.0
[    15.049] (==) Matched modesetting as autoconfigured driver 0
[    15.049] (==) Matched fbdev as autoconfigured driver 1
[    15.049] (==) Matched vesa as autoconfigured driver 2
[    15.049] (==) Assigned the driver to the xf86ConfigLayout
[    15.049] (II) LoadModule: "modesetting"
[    15.049] (II) Loading /usr/lib/xorg/modules/drivers/modesetting_drv.so
[    15.066] (II) Module modesetting: vendor="X.Org Foundation"
[    15.066]    compiled for 1.20.8, module version = 1.20.8
[... 410 lines cut ...]
[    36.636] (II) modeset(0): Modeline "800x600"x0.0   40.00  800 840 968 1056  600 601 605 628 +hsync +vsync (37.9 kHz e)
[    36.636] (II) modeset(0): Modeline "640x480"x0.0   31.50  640 656 720 840  480 481 484 500 -hsync -vsync (37.5 kHz e)
[    36.636] (II) modeset(0): Modeline "640x480"x0.0   25.18  640 656 752 800  480 490 492 525 -hsync -vsync (31.5 kHz e)
[    36.637] (II) modeset(0): Modeline "720x400"x0.0   28.32  720 738 846 900  400 412 414 449 -hsync +vsync (31.5 kHz e)
[    36.637] (II) modeset(0): Modeline "1280x1024"x0.0  135.00  1280 1296 1440 1688  1024 1025 1028 1066 +hsync +vsync (80.0 kHz e)
[    36.637] (II) modeset(0): Modeline "1024x768"x0.0   78.75  1024 1040 1136 1312  768 769 772 800 +hsync +vsync (60.0 kHz e)
[    36.637] (II) modeset(0): Modeline "1024x768"x0.0   65.00  1024 1048 1184 1344  768 771 777 806 -hsync -vsync (48.4 kHz e)
[    36.637] (II) modeset(0): Modeline "800x600"x0.0   49.50  800 816 896 1056  600 601 604 625 +hsync +vsync (46.9 kHz e)
[    36.637] (II) modeset(0): Modeline "1152x864"x0.0  108.00  1152 1216 1344 1600  864 865 868 900 +hsync +vsync (67.5 kHz e)
[    36.637] (II) modeset(0): Modeline "1280x1024"x0.0  108.00  1280 1328 1440 1688  1024 1025 1028 1066 +hsync +vsync (64.0 kHz e)
 
I get this ^ 

i installed digimend as this person on youtube told me to 
https://youtu.be/_V79I3EKCvc

and linux does recognise this as a tablet in the screenshot attached below.
[image]

but for some reason it doesnt take input. it says run a command when connected what should i write in the command space 
also I'm new to linux so please answer my question please.

please walk me through this.

thank you

nic3-14159, 2021-04-11:
Hello! The digimend driver does not support this tablet, but based on other issues, this tablet may work regardless. But first, what Linux distribution is this (such as Ubuntu, Arch, etc)?

Bhanu-tarun, 2021-04-11:
@nic3-14159 I use kali

Bhanu-tarun, 2021-04-11:
@nic3-14159 look at this screenshot, here it is recognised as a touchpad but it doesnt work 
[image]

Bhanu-tarun, 2021-04-11:
i also use the default desktop environment

Bhanu-tarun, 2021-04-11:
i will attach the user guide too 
[Uploading WP9620 Graphics Tablet User Guide-20200430.pdf…]()

nic3-14159, 2021-04-11:
It seems like the situation with your tablet is very similar to the one in #514, so I will try and give a step by step guide to the solution that @ilmanowar and I discussed in that issue
1. Open a terminal, and run the command `sudo apt install xserver-xorg-input-wacom`
2. Run the command `sudo nano /etc/X11/xorg.conf.d/50-tablet.conf
3. Copy the following into that file:
```
Section "InputClass"
  Identifier "Tablet"
  Driver "wacom"
  MatchDevicePath "/dev/input/event*"
  MatchUSBID "6161:4d15"
EndSection
```
4. Press Ctrl+O to save, then press Ctrl+X to exit
5. Log out and log back in and test the tablet. If it still doesn't work, try restarting your computer

These steps should be able to get the pen to move the cursor. The Digimend driver is not used by this tablet, so you could uninstall it if you want. Assuming you previously followed the same steps in that youtube video:
1. Enter the directory where you extracted the zip file with the source code (the same one where you ran `sudo make dkms_install` before)
2. Run the command `sudo make dkms_uninstall`

Bhanu-tarun, 2021-04-17:
thank you very much, it is working but it is only taking in input from half the tablet and not the rest, how can i fix it

Bhanu-tarun, 2021-04-17:
I also reinstalled Linux, because the previous Dualboot lacked storage

Bhanu-tarun, 2021-04-17:
i thought installing wine and installing the driver through wine would fix the problem, but the graphics driver wont launch.

Bhanu-tarun, 2021-04-17:
i restarted hoping it would fix the problem but nothing happened

chandan00761, 2021-07-02:
I followed the steps and got the same issue where one half of the driver is unusable.

Bhanu-tarun, 2021-07-05:
same with me half my graphics tablet doesn't work but the other does

iamashwin99, 2021-07-20:
I have the same issue with another tablet rebranded as RTSY WP9620 from LetSketch. Input is just from half of the tab

kode-python, 2021-09-01:
My problem of half screen got solved by simultaneously pressing buttons which are adjacent to the middle one for approximately 6 seconds till green light turned red. After that, it did not go again in Android mode even after disconnecting and reconnecting or restarting the computer, not even after pressing those 2 buttons again.

Bhanu-tarun, 2021-09-07:
thank you very much ill try doing it

Bhanu-tarun, 2021-09-07:
it worked thanks a lot

closed 2021-09-07 by Bhanu-tarun

jwrdegoede, 2021-10-21:
Here is some info for anyone ending up here with the WP9620N (which is different from the WP9620, the WP9620N does not use a battery).

On this tablet by default only the left half of the tablet works under Linux. 

The pictures in the Chinese manual are helpful to resolve this. You can set an aspect ratio for the device you want to use it with and depend on the aspect ratio only a part of the tablet is used in the Android interface (which is what Linux uses).

And it defaults to a portrait aspect-ratio of 9:16, if you still have the little leaflet with came with it it has a picture showing how only the left side of the tablet is used in this case.

To program a 16:9 aspect ratio do the following:

1. Orient the tablet with the USB-connector and the buttons on the left side of the tablet (which is its normal orientation AFAICT).
2. Press the buttons which are closest to the bottom together and keep them pressed. The status LED will now turn off and after approx. 3 seconds turn on again.
3. Release the buttons.

Now you can use the entire width of the tablet, but you are missing an edge at the top and bottom of the active area. To make the missing edge smaller set the tablet to a 16:10 aspect ratio:

1. Orient the tablet with the USB-connector and the buttons on the left side of the tablet (which is its normal orientation AFAICT).
2. Press the buttons which are closest to the bottom together and keep them pressed. The status LED will now turn off and after approx. 3 seconds turn on again.
3. Keep the buttons pressed, the LED will turn off again and after again approx. 3 seconds (so approx 7 seconds in total) it will turn on a second time.
4. Release the buttons.

Note you will now still miss a small part of the active area at the top and bottom.

There appears to be a custom protocol used by the Windows drivers, see:
https://bugzilla.redhat.com/show_bug.cgi?id=2005575

Fixing the missing top and bottom edges of the active area, as well as making the buttons on the tablet and the stylus buttons behave as expected by Linux userspace, will require writing a driver to use the custom Windows protocol.

I'm a bit busy atm, so I'm not sure when I will get around to writing a driver to use the custom Windows protocol, but just fixing the aspect ratio should make the tablet a lot more usable for you.

jwrdegoede, 2021-12-17:
Good news I've finished writing the kernel driver for the WP9620 tablets and submitted it upstream:
https://lore.kernel.org/linux-input/20211212212333.239644-1-hdegoede@redhat.com/

I still need to add a tablet profile to libwacom for this tablet, but the GNOME UI to assign actions to the 5 buttons on the pad of the tablet does not work, which seems to be a GNOME bug:
https://gitlab.gnome.org/GNOME/gnome-control-center/-/issues/1321

Once I can test that my libwacom profile is correct I will also submit that upstream.

rorocui, 2022-05-04:
> Good news I've finished writing the kernel driver for the WP9620 tablets and submitted it upstream: https://lore.kernel.org/linux-input/20211212212333.239644-1-hdegoede@redhat.com/
>
> I still need to [...]

hi, jwrdegoede

I have built your driver on my ubuntu machine, load it successfully, looks working fine. Do you have a chance to create a profile to libwacom as well?
Thanks.
roro

jwrdegoede, 2022-05-05:
> I have built your driver on my ubuntu machine, load it successfully, looks working fine. Do you have a chance to create a profile to libwacom as well? Thanks. roro

I've put what I have here: https://fedorapeople.org/~jwrdegoede/letsketch-wp9620.tablet

This should mostly work, just drop it in /usr/share/libwacom/ and then start a libwacom using app. Note I've pointed the svg for assigning actions to the buttons to the svg from a completely different tablet which also happens to have 5 buttons. If someone can create a proper svg for the Letsketch WP9620 that would be great.

rorocui, 2022-05-09:
> 
hi, Jwrdegoede

That is profile works very well on my LetSketch WP9612N. Thank you very much.
And there are only 3 buttons on my tablet.

roro

nengvalent, 2023-08-29:
Hi!
Sorry to jump in.
But my LetSketch WP9622N unable to connect to either to windows or android even though I have installed the driver on my windows.
Is there any steps should I follow?
[image]