DIGImend/digimend-kernel-drivers #534: Need Help with HS64 on Ubuntu 20.04

TetsuShiro, 2021-05-10:
- I have tried my best to find a way to configure my HS64 and get it to be shown on the settings> wacom tablets option but it shows:

[image]

- I haven't really tried changing the buttons configuration using whatever is stated in readme. but I hope someone will tell me how to configure it and make ubuntu 20.04 to detect my tablet and stylus ASAP.

- Here is some more info:

[image]

simos, 2021-05-10:
While your tablet does not show in the settings, it should work in applications. 

See https://blog.simos.info/how-to-setup-the-huion-430p-drawing-tablet-on-ubuntu-20-04/ on how to verify.

omnitrix9991, 2021-07-05:
[huion-HS64.tablet.tar.gz](https://github.com/DIGImend/digimend-kernel-drivers/files/6762843/huion-HS64.tablet.tar.gz)

the tablet works fully, even the GUI configurator, you need digimend v10 drivers, now copy the huion-HS64.tablet file to /usr/share/libwacom

thanks to juansegaes for making the .tablet file , just had to edit the name according to xsetwacom devices for v10 drivers
[juansegaes huion-HS64.tablet file](https://github.com/juansegaes/HS64)
maybe you can use the file as it is, if you edit files in v10 driver , as per this [commit](https://github.com/Huion-Linux/DIGImend-kernel-drivers-for-Huion/commit/66605efcffdb00232439a209acf2cd3fda714980)
[image]

sahaspoopoo, 2021-11-20:
> 

is it only for hs64