DIGImend/digimend-kernel-drivers #535: XP-Pen Artist 24 Pro

Deevad, 2021-05-12:
[image]
_A photo of my model on my desk, CC-By Deevad_

Hi, here is my thread for requesting the support of the XP-Pen Artist 24 Pro.  

## Full Diagnostics

- Product page: https://www.xp-pen.com/product/602.html
- Original model (lsusb ID): [28bd:092d](#)
- USB descriptors: [descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6465565/descriptors.txt)
- Uclogic-probe: [probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6465573/probe.txt)
- Pen coordinates: [pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6465582/pen_coords.txt)
- Pen tilt: [pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6465586/pen_tilt.txt)
- Pen pressure: [pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6465592/pen_pressure.txt)
- Pen buttons: [pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6465595/pen_buttons.txt)
- Frame (red) dials: [frame_dials.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6465598/frame_dials.txt)
- Frame buttons: [frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6465599/frame_buttons.txt)

And everything as a zip: [XPPEN_24-pro-artist_Digimend-full-diagnostics.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/6465607/XPPEN_24-pro-artist_Digimend-full-diagnostics.zip)

(Note: uclogic-probe command line really made the tablet totally disapear as a device. I had to remove all the cables, wait a minute and then reboot the system to get it back to life.)

## Workaround and review on Linux:

Many parts of the tablet already works with the Digimend driver (mapping/coordinate/tilt/stylus button 1) and I documented workarounds (with udev) for the remaining part not supported yet on my [blog post about XP-Pen Artist 24 Pro](https://www.davidrevoy.com/article842/review-of-the-xp-pen-artist-24-pro-on-linux) and on my [video-review of the XP-Pen Artist 24 Pro on Linux](https://www.youtube.com/watch?v=0QEnRWNIjtU) but the workarounds have limitation. 

## References

From my non technical point-of-view, it sounds this tablet share a lot in common with all its siblings at lower size resolution. It was possible to already find many thread here, and I'll link them under as they were all useful and valuable to produce my workarounds (thank you to all participant).

**Issues with full descriptors for reference:**  
- [The XP-Pen Artist Pro 12](https://github.com/DIGImend/digimend-kernel-drivers/issues/351)
- [The XP-Pen Artist Pro 13.3](https://github.com/DIGImend/digimend-kernel-drivers/issues/498)
- [The XP-Pen Artist Pro 15.6](https://github.com/DIGImend/digimend-kernel-drivers/issues/286)
- [The XP-Pen Artist Pro 22](https://github.com/DIGImend/digimend-kernel-drivers/issues/476)
- [The XP-Pen Artist Pro 22R](https://github.com/DIGImend/digimend-kernel-drivers/issues/484)

**Pull Request that might help for reference:**
- [Adds XP Pen Artist 13.3 tablet with fixed stylus button](https://github.com/DIGImend/digimend-kernel-drivers/pull/320)
- [UC-Logic XP-Pen Artist 22HD support](https://github.com/DIGImend/digimend-kernel-drivers/pull/73)
- [Uclogic XP-Pen Aritst 22HD all over again](https://github.com/DIGImend/digimend-kernel-drivers/pull/133)
- [XP-Pen Arist 15.6 Pro Support](https://github.com/DIGImend/digimend-kernel-drivers/pull/291)

kurikaesu, 2021-08-21:
Hi Deevad, I'm looking to add support for your Artist 24 Pro to my pull request here: https://github.com/DIGImend/digimend-kernel-drivers/pull/557

I need you to push the special handshake key to interface 2 then resubmit the data for:
- Pen coordinates
- Pen tilt
- Pen pressure
- Pen buttons
- Frame (red) dials
- Frame buttons

The special handshake key can be sent to the tablet using the python script provided by @Korvox in one of the issues/PRs:
```
#! /usr/bin/env python3

import usb.core
import usb.util

dev = usb.core.find(idVendor=0x28bd, idProduct=0x0909)

if dev.is_kernel_driver_active(2):
    dev.detach_kernel_driver(2)

cfg = dev.get_active_configuration()
intf = cfg[(2,0)]
ei = intf[0]
eo = intf[1]

eo.write(bytes.fromhex('02b00400000000000000'))

while(True):
    print(''.join(format(x, '02x') for x in ei.read(12)))
```

pyusb needs to be installed for the root user since writing on the USB device requires root permissions.
Can be installed with `pip3 install pyusb`

You can ignore the USBTimeoutError that it spits out at the end.

Once the key has been sent to the tablet, when you start capturing the data like Pen Coordinates, the prefix byte should switch to `02` like: `02 f0 00 00 04 00 00 00 00 13`. It is fine if the length of the data differs as it is important to know that it is different.

PS. Love your artwork!

Deevad, 2021-08-30:
Thanks @kurikaesu for the feedback and guidance. I finally decided to test on my main workstation (laptop had a too weak graphic card for the quadHD of the 24Pro anyway); I also saw I had compilation/install issues...

I tried again today, and I tried to get a cleaner method. I used `sudo apt purge digimend-dkms` instead of  `sudo apt remove digimend-dkms` to clean the system, and then I generated the package with 'dpkg-buildpackage -b -uc' and after generation (without errors), I install (after a `cd ..`) with `sudo apt install ./digimend-dkms_11_all.deb`, then I perform a `sudo reboot now` before testing. It sounds like a cleaner method to test.

I also added `28bd:092d` tablet ID to the `xorg.conf` file of your branch to ease testing.

So, after a first run, good news: now I can control the stylus mapping and buttons of the stylus with xsetwacom. I have `MapToOutput` that works, Tilt that works, pressure that works. (Note: it was already the case with digimend v.10). But now I can also assign keys to the three buttons of the stylus via xsetwacom. 

But 

1. The Pad buttons still reports 3 active buttons only (like a stylus?) 
2. The mapping of the stylus on the monitor has a bug: it's like if my display is split into two block horyzontally, as if I had two active areas on the tablet (A from 0 to around 3/4 of the tablet, and a B from 3/4 to the end). However, the vertical is fine. For the horyzontal, If I move stylus from extreme left to extreme right, the cursor first travel the monitor fully from left to right while having my stylus over block A, then reset at left position when reaching block B and redo a full display travel from left to right for this block). I tried to `ResetArea` with xsetwacom, or to `MapToOutput`, I still obtain the same effect (just on another monitor). This is not something that happens with digimend v.10 driver where the mapping takes full tablet area by default.

Thanks if you have an idea how to fix the size of the tablet and how the Pad/Frame can target the buttons.

kurikaesu, 2021-08-30:
Which 3 buttons are the only ones active?

The monitor mapping definitely sounds like a bug but I don't know if it is a bug with the descriptor or not. It could be that the resolution of the digitizer is much higher than the 22R for example and is "overflowing" the number container that I'm using.

One way to check this is to do pen_coords test but with just 2 points, the top left of the digitizer and the bottom right of the digitizer.

Lastly, thank you for your patience with this. Hopefully we are only a few steps away from proper support.

kurikaesu, 2021-09-07:
It has been a while since I last commented but I have something that may interest you. I've been working on userland (not kernel) drivers for my 22R Pro here: https://github.com/kurikaesu/xp-pen-userland

This tool runs as an application in the background (I have it running on login) and creates a working driver for my pen display that leverages the wacom driver for all of the stylus related things like stylus buttons and actual drawing. My driver also currently supports all of the buttons and can map all of the buttons on the fly. I don't have a nice user interface to do it yet as you need to hand-update a JSON config file to change the mappings and I haven't set up mapping of the dials yet.

I will keep you posted with progress but knowing that your 24 Pro is similar enough I thought you may want to try testing my driver out while I work on getting a proper interface written.

Deevad, 2021-09-07:
Hey @kurikaesu ! Oh wow, thank you for sharing it. That sounds great! :+1: 

Sorry for no recent test/news here; I have a deadline for delivering many illustrations and a storyboard by the mid of September, and I'm full time working on it. I'll probably recover testing time budget after that.

kurikaesu, 2021-09-09:
I've added support for the 24 Pro to my userland drivers. Since I don't have a 24 Pro myself to to test that it is working I will have to rely on you to try it out. I have actually tried contacting XP-Pen to see if they would lend me a 24 Pro so that I can test and develop on it but no reply yet. I doubt they will oblige but I wanted to try anyway.

Instructions on how to build the driver and run it are in the readme of the git repository. I've also made progress with the configuration GUI here: https://github.com/kurikaesu/xp-pen-userland-config-util

Deevad, 2021-09-17:
Thanks for merging it in your MR; it will be easier to test.

About the model; I can ask my contact at XP-Pen. I can also ask around community/social media if someone around your location can lend you a 24 Pro. Where are you located? (roughly, at city scale)

kurikaesu, 2021-09-17:
I am in Seattle city, Washington State (Not D.C) in the USA.

I'm still doing a few bits of refining on the driver and I have a new GUI in planning as well to hopefully make things more simpler and less clunky.

Deevad, 2021-09-19:
Thanks for the info. I asked by email (but unfortunately, my contact are mostly marketing/youtube one, not sure they can take this type of decisions) and on social media here:
- https://twitter.com/davidrevoy/status/1439568274908487682
- https://framapiaf.org/@davidrevoy/106958348363699104
Probability to find someone sounds very unlikely, but a small chance is still better than zero chances.

kurikaesu, 2021-09-23:
Good news! I believe I have identified and resolved the issue you were having on my userspace drivers where the stylus was looping around as you moved the stylus to the right.

I was only really able to see the issue and figure it out because I had a tablet that was large enough to cause the problem.

Please test my driver out again (the name has changed because I am now also starting to support Huion) here: https://github.com/kurikaesu/userspace-tablet-driver-daemon

Any issues you find can be posted to the repository as I've started adding my own issues and todo there.

Deevad, 2021-09-24:
@kurikaesu Woohoo! Thanks. I'll test this week-end. :+1: