DIGImend/digimend-kernel-drivers #536: Huion HS-611s

Vertecedoc4545, 2021-05-19:
on the 10 release and on the master branch of the proyect isn't working properly.
on the past releases wasn't recognized as the huion hs611s was recognized as a huion 640p and appered on the wacom tablet seccion on the gnome configuration (gnome 3.38 kernel 5.10, debian sid)
but now on open suse kernel 5.12.0-2 and gnome 40 isn't recognized as tablet on the configuration page also the buttons can't be configurated really only works the presion and the movement of the pen around the tablet nothing else

Vertecedoc4545, 2021-05-31:
On arch linux work but recognicing the huion HS-611 as huion h950p and works as described before

rurigk, 2021-12-20:
On Manjaro KDE
Kernel: x86_64 Linux 5.15.7-1-MANJARO
KDE 5.88.0 / Plasma 5.23.4
With the digimend-kernel-drivers installed the tablet is recognized as Huion H950P only in the Wacom Tablet Settings window
[image]

But with "xsetwacom --list devices" the tablet is labeled correctly
[image]

rurigk, 2021-12-20:
Also "libinput debug-events" don't show any events when i press the last 2 buttons at the botton from the "Express keys"

But it shows the events in evtest "HUION Huion Tablet_HS611 Pad"

```
Event: time 1640006252.968471, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90009
Event: time 1640006252.968471, type 1 (EV_KEY), code 264 (BTN_8), value 1
Event: time 1640006252.968471, -------------- SYN_REPORT ------------
Event: time 1640006253.146486, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90009
Event: time 1640006253.146486, type 1 (EV_KEY), code 264 (BTN_8), value 0
Event: time 1640006253.146486, -------------- SYN_REPORT ------------
Event: time 1640006253.484526, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000a
Event: time 1640006253.484526, type 1 (EV_KEY), code 265 (BTN_9), value 1
Event: time 1640006253.484526, -------------- SYN_REPORT ------------
Event: time 1640006253.710553, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000a
Event: time 1640006253.710553, type 1 (EV_KEY), code 265 (BTN_9), value 0
Event: time 1640006253.710553, -------------- SYN_REPORT ------------
```

The tilt is working for me in Krita

All the media keys are working for me
The 2 special buttons are working for me (the show apps and files)

The features in green boxes are working and the ones in red boxes are not
[image]

Fantailed, 2022-01-04:
It's similar for me. In the Settings window it shows Huion H950P and it only detects 8/10 buttons, exactly as in the image above and the touch strip doesn't work either.

But tilt doesn't work and `xsetwacom list` only shows the stylus and the pad; the touch strip is not listed.

@rurigk can you tell me what you did to get tilt working? I can live with 2 buttons less but tilt is a pretty big deal.

Fantailed, 2022-01-08:
For me it was the same - Manjaro KDE, tablet gets recognized as H950P in settings GUI, only 8 assignable buttons and touch strip not working.

At some point I also got the issue that the GUI was unable to apply any express button settings.

What I did was I ran `kde_wacom_tabletfinder`, where the tablet was recognized as a HS611, manually specified it as a 10-button-device and saved it.
Then I wrote [the following script](https://gist.github.com/Fantailed/be5cc5c0076cbfb8e36223681a677f1c) that detects dbus signals from the GUI (things like tablet added/removed and profile changes), reads the configuration settings from the files produced by the GUI and sets the buttons using `xsetwacom`, which for some reason works for all 10 buttons.

I set that script to autostart and now the tablet works as one would expect, configurable from the GUI, minus the touch strip.

[image]

rurigk, 2022-01-08:
kde_wacom_tabletfinder worked and it detects the tablet as HS611
Your script worked too

albfan, 2022-01-26:
this has nothing to do with kernel, but with libwacom config:

https://github.com/linuxwacom/libwacom/pull/288#issuecomment-1022420947

as many huion share same usb product vendor.

albfan, 2022-01-26:
What looks wrong is checking with evtest and libinput,  strip is mapped with buttons in a combination similar to compose buttons:

strip looks like a mix of button combinations:

|BTN_8 | BTN_9 | BTN_8+BTN_9 | BTN_SOUTH | BTN_SOUTH+BTN_8 | BTN_SOUTH+BTN_9|  BTN_SOUTH+BTN_8+BTN_9| BTN_EAST|
|-|-|-|-|-|-|-|-|

this looks like an error as there would be no way to distinguish real pad buttons 8 and 9 and strip.

Any idea about how strip works?

albfan, 2022-01-27:
https://github.com/DIGImend/usbhid-dump correctly dumps a different button press:

- button 8

```
001:013:000:STREAM             1643326218.519923
 08 E0 01 01 00 01 00 00 00 00 00 00

001:013:000:STREAM             1643326218.701872
 08 E0 01 01 00 00 00 00 00 00 00 00
```

- button 9

```
001:013:000:STREAM             1643326222.210286
 08 E0 01 01 00 02 00 00 00 00 00 00

001:013:000:STREAM             1643326222.344285
 08 E0 01 01 00 00 00 00 00 00 00 00
```

strip:

```
001:013:000:STREAM             1643326227.790416
 08 F0 01 01 00 01 00 00 00 00 00 00

001:013:000:STREAM             1643326229.282055
 08 F0 01 01 00 02 00 00 00 00 00 00

001:013:000:STREAM             1643326230.008159
 08 F0 01 01 00 03 00 00 00 00 00 00

001:013:000:STREAM             1643326230.620300
 08 F0 01 01 00 04 00 00 00 00 00 00

001:013:000:STREAM             1643326231.194433
 08 F0 01 01 00 05 00 00 00 00 00 00

001:013:000:STREAM             1643326231.654437
 08 F0 01 01 00 06 00 00 00 00 00 00

001:013:000:STREAM             1643326232.152338
 08 F0 01 01 00 07 00 00 00 00 00 00

001:013:000:STREAM             1643326232.534453
 08 F0 01 01 00 08 00 00 00 00 00 00

001:013:000:STREAM             1643326233.146471
 08 F0 01 01 00 00 00 00 00 00 00 00
```

gdonval, 2022-06-14:
Just a little heads up: updating the firmware changes the model ID to `006f`. I could get the tablet working with new static `evdev` rule and `hwdb` information with existing drivers but width and height are misreported by the kernel and resolution is missing. Tilt reporting is also weird.

JoseExposito, 2024-05-03:
Hi!

Your issue should be fixed by [DIGImend v13](https://github.com/DIGImend/digimend-kernel-drivers/releases/tag/v13). Please, if you have a chance, give it a try and report any problems found.

Thanks!

closed 2024-05-03 by JoseExposito