DIGImend/digimend-kernel-drivers #537: xp-pen deco v2 right-click broken

linuxgirl22, 2021-05-20:
Hello

I always use these 2 files, to ensure right-click works for xp-pen deco v2, after seeing the solution in a thread here ...  
10-xppen.hwdb into /etc/udev/hwdb.d 
and xp-pen-dec01.tablet into /usr/share/libwacom

Now, after having done a format/reinstall, if I add these 2 files, it breaks xp-pen deco v2 being able to paint at all ... cursor goes to small arrow; no brush shape or able paint on canvas. Only option is to reinstall both linux and lts kernels, where dkms is rebuilt (had tried older kernels, but gcc errors, after an update from gcc 10 to 11 a few days ago) ... 
(3/4) Install DKMS modules
==> dkms install --no-depmod -m digimend -v 10.r0.gec5c16d -k 5.10.38-1-lts
==> dkms install --no-depmod -m digimend -v 10.r0.gec5c16d -k 5.12.5-arch1-1
==> depmod 5.10.38-1-lts
==> depmod 5.12.5-arch1-1
(Anarchy/Mate)

I think there may be a rules.d folder and udev.conf file in udev that wasn't there last time?  
I just can't manage without right-click, so am not sure what to do, and would very much appreciate advice. 
Thanks.

linuxgirl22, 2021-05-20:
Sorry, maybe unclear in opening post ... reinstalling kernels/building dkms just stops the little arrow showing/program not functioning, but there's no right-click ability.

linuxgirl22, 2021-05-20:
Sorry, I try to edit, but spinning circle ... just to add that reboot = broken again/small arrow showing/unable to paint on canvas, even though have removed the 2 files and refreshed dkms.

linuxgirl22, 2021-05-21:
There seems to be nobody here? Just to comment that did do big donation or two previously; not just doing a 'get stuff and run'; just don't have any technical ability to answer any posts etc. Can anyone recommend where is the right place to report these files breaking, as I don't understand why they're breaking or how to fix? I don't know what has changed, but putting these 2 files in the usual places is breaking Sai2 entirely, and I can only reinstall system, but even then, no right-click, so no point. Would be grateful to be pointed to anywhere I can ask further.

nic3-14159, 2021-05-22:
I'm assuming you are using ihewitt's code? It might be better to post something under that pull request (https://github.com/DIGImend/digimend-kernel-drivers/pull/375), since that code isn't yet merged into the main DIGImend repo.

However, I believe there is a completely different method to support this tablet that doesn't involve those two files, based on another open pull request (https://github.com/DIGImend/digimend-kernel-drivers/pull/291). I've gotten that code to work with my tablet (XP-Pen Star 06), and it's been working for me across several updates.

I can try taking a look at the code and see if I can get something that works.

linuxgirl22, 2021-05-22:
Hi nic3-14159 and thank you very much for your response. Yes, you're right about the code used, and thank you for linking to those threads; I will add something. I hadn't been sure if it was wine/kernel/arch issues, so it's good to know what's happening. Trying mint mate right now, and have hit the same issue. Xinput-list shows the three ugtablet entries, and mouse can paint, but just a small white triangle showing up as xp-pen's pen cursor, and not able to paint anything. 

painter@painter:~$ xinput list
⎡ Virtual core pointer                    	                        id=2	        [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	        id=4	        [slave  pointer  (2)]
⎜   ↳ Drevo keyboard Drevo keyboard Consumer Control	id=9	[slave  pointer  (2)]
⎜   ↳ Drevo keyboard Drevo keyboard Mouse     	id=10	[slave  pointer  (2)]
⎜   ↳ Drevo keyboard Drevo keyboard Keyboard  	id=11	[slave  pointer  (2)]
⎜   ↳ Logitech USB Optical Mouse              	        id=13	[slave  pointer  (2)]
⎜   ↳ UC-Logic 21.5" Tablet Monitor Pad pad   	        id=15	[slave  pointer  (2)]
⎜   ↳ UC-Logic 21.5" Tablet Monitor stylus    	        id=14	[slave  pointer  (2)]
⎜   ↳ UGTABLET 10 inch PenTablet Mouse        	id=19	[slave  pointer  (2)]
⎜   ↳ UGTABLET 10 inch PenTablet Pen (0)      	id=22	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	                id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	        id=6	[slave  keyboard (3)]
    ↳ Power Button                            	        id=7	[slave  keyboard (3)]
    ↳ Drevo keyboard Drevo keyboard           	id=8	[slave  keyboard (3)]
    ↳ Drevo keyboard Drevo keyboard System Control	id=12	[slave  keyboard (3)]
    ↳ Drevo keyboard Drevo keyboard Consumer Control	id=17	[slave  keyboard (3)]
    ↳ Drevo keyboard Drevo keyboard Keyboard  	        id=18	[slave  keyboard (3)]
    ↳ Eee PC WMI hotkeys                      	                        id=16	[slave  keyboard (3)]
    ↳ UGTABLET 10 inch PenTablet Keyboard     	        id=20	[slave  keyboard (3)]
    ↳ UGTABLET 10 inch PenTablet              	                id=21	[slave  keyboard (3)]
[using Ugee just as 2nd monitor now]

Very reassuring to hear there's a different method, and great that your tablet is working really well. Sorry, I'm not understanding the other method in #291 ... not very tech, eek. 

Thank you very much about looking at the code. If there's any information I can provide, just let me know, and will provide whatever I can.

linuxgirl22, 2021-05-28:
Hi again. I didn't understand from the other thread what the other method is. I've tried asking on winehq a couple of days ago (no response) and earlier today on linuxquestions (nothing yet). Really puzzled why the changes in the linked thread only hit personally a couple of weeks ago. 

If it's to do with some wayland or kernel or security issue or something else, I can't find anything about this, and don't know where to bug report to, in order to fix and also to prevent any further damage, not only for myself. All I can do for now is not do updates, as the risk of further breakage and no information/fix is too big.

kurikaesu, 2021-09-27:
I've just finished assisting someone to test my userspace driver for this tablet here: https://github.com/kurikaesu/userspace-tablet-driver-daemon

If you are willing to use a userspace driver instead of a kernel driver, please have a look.

linuxgirl22, 2021-09-27:
Thank you very much for posting! There'd been some kind of glitch back then, but it seemed to right itself within about a week, after some updates, if I remember rightly. Have bookmarked the link you've given, incase it happens again. Really appreciate it.