DIGImend/digimend-kernel-drivers #540: Acepen, AP 906 report and questions.

kireita, 2021-06-08:
Hey there,

i recently bought a drawing tablet named [AP 906](https://ace-pen.com/ap-906/) and i have several things to report.

this is how the pen is identified with `lsusb`:
```
Bus 003 Device 002: ID 5543:0091 UC-Logic Technology Corp. 906
```
this is how it is identified with `xinput list`:
```
⎡ Virtual core pointer
⎜   ↳ UPIN.INC 906 Mouse                        id=9    [slave  pointer  (2)]
⎜   ↳ UPIN.INC 906                              id=10   [slave  pointer  (2)]
⎡ Virtual core pointer
    ↳ UPIN.INC 906 Keyboard                     id=11   [slave  keyboard (3)]
```

using it as a wacom tablet:
```
Section "InputClass"
    Identifier "Tablet"
    Driver "wacom"
    MatchDevicePath "/dev/input/event*"
    MatchUSBID "5543:0091"
EndSection
```
works, but the hardware buttons dont work, only pressure, mouse and button 2 work.

using it as a evdev device:
```
Section "InputClass"
       Identifier "UPIN.INC 906"
       MatchIsTablet "on"
       MatchUSBID "5543:0091"
       MatchDevicePath "/dev/input/event*"
       Driver "evdev"
       # Apply custom Options below.
EndSection
```
works, but pressure doesnt work and button 3 has to be remapped because it detects it as left mouse click.

uclogic output:

```
[kireita@SOUCOUYANT ~]$ sudo uclogic-probe 3 2
M 55 00 50 00 49 00 4E 00 2E 00 49 00 4E 00 43 00
P 39 00 30 00 36 00
S 64 12 03 FF 7F FF 7F 03 00 FF 1F D8 13 0A 00 00 00 00 00

[kireita@SOUCOUYANT ~]$ sudo uclogic-probe 3 2 | uclogic-decode
    |           Manufacturer: UPIN.INC
    |                Product: 906
Failed to get string descriptor 0x65: Operation timed out
 64 |        Params block #1: ????????
    |                          Max X: 32767
    |                          Max Y: 32767
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
```

xinput properties:
```
[kireita@SOUCOUYANT ~]$ xinput list-props "UPIN.INC 906"
Device 'UPIN.INC 906':
        Device Enabled (156):   1
        Coordinate Transformation Matrix (158): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        Device Accel Profile (286):     0
        Device Accel Constant Deceleration (287):       1.000000
        Device Accel Adaptive Deceleration (288):       1.000000
        Device Accel Velocity Scaling (289):    10.000000
        Device Product ID (279):        21827, 145
        Device Node (278):      "/dev/input/event5"
        Evdev Axis Inversion (306):     0, 0
        Evdev Axis Calibration (307):   <no items>
        Evdev Axes Swap (308):  0
        Axis Labels (309):      "Abs X" (290), "Abs Y" (291), "Abs Pressure" (303), "Abs Tilt X" (304), "Abs Tilt Y" (305)
        Button Labels (310):    "Button Unknown" (302), "Button Unknown" (302), "Button Unknown" (302), "Button Wheel Up" (162), "Button Wheel Down" (163), "Button Horiz Wheel Left" (164), "Button Horiz Wheel Right" (165)
        Evdev Scrolling Distance (311): 0, 0, 0
        Evdev Middle Button Emulation (312):    0
        Evdev Middle Button Timeout (313):      50
        Evdev Middle Button Button (314):       2
        Evdev Third Button Emulation (315):     0
        Evdev Third Button Emulation Timeout (316):     1000
        Evdev Third Button Emulation Button (317):      3
        Evdev Third Button Emulation Threshold (318):   20
        Evdev Wheel Emulation (319):    0
        Evdev Wheel Emulation Axes (320):       0, 0, 4, 5
        Evdev Wheel Emulation Inertia (321):    10
        Evdev Wheel Emulation Timeout (322):    200
        Evdev Wheel Emulation Button (323):     4
        Evdev Drag Lock Buttons (324):  0
[kireita@SOUCOUYANT ~]$ xinput list-props "UPIN.INC 906 Mouse"
Device 'UPIN.INC 906 Mouse':
        Device Enabled (156):   1
        Coordinate Transformation Matrix (158): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        libinput Natural Scrolling Enabled (292):       0
        libinput Natural Scrolling Enabled Default (293):       0
        libinput Middle Emulation Enabled (294):        0
        libinput Middle Emulation Enabled Default (295):        0
        libinput Calibration Matrix (296):      1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        libinput Calibration Matrix Default (297):      1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        libinput Left Handed Enabled (298):     0
        libinput Left Handed Enabled Default (299):     0
        libinput Send Events Modes Available (275):     1, 0
        libinput Send Events Mode Enabled (276):        0, 0
        libinput Send Events Mode Enabled Default (277):        0, 0
        Device Node (278):      "/dev/input/event4"
        Device Product ID (279):        21827, 145
        libinput Drag Lock Buttons (300):       <no items>
        libinput Horizontal Scroll Enabled (301):       1
[kireita@SOUCOUYANT ~]$ xinput list-props "UPIN.INC 906 Keyboard"
Device 'UPIN.INC 906 Keyboard':
        Device Enabled (156):   1
        Coordinate Transformation Matrix (158): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        libinput Send Events Modes Available (275):     1, 0
        libinput Send Events Mode Enabled (276):        0, 0
        libinput Send Events Mode Enabled Default (277):        0, 0
        Device Node (278):      "/dev/input/event6"
        Device Product ID (279):        21827, 145
```

i would like some help on how to bound it to my left monitor, currently i have 2 monitors 1080p is my left monitor and 4k my right monitor
[image]

ideally i want it just to work with the wacom drivers because KDE has a "graphic tablet" tool which is very handy but the buttons wont work.

if anyone can help me with i would appreciate it very much. since this is a pretty cheap tablet and perfect for people that just started.

thanks in advance

kireita, 2021-06-08:
i am currently running the driver like this:
```
Section "InputClass"
      Identifier "Graphics tablet Keyboard"
      MatchIsKeyboard "on"
      MatchProduct "UPIN.INC 906 Keyboard"
      MatchDevicePath "/dev/input/event*"
      Driver "evdev"
EndSection

Section "InputClass"
       Identifier "Graphics tablet mouse"
       MatchIsPointer "on"
       MatchProduct "UPIN.INC 906 Mouse"
       MatchDevicePath "/dev/input/event*"
	   Driver "evdev"
EndSection

Section "InputClass"
       Identifier "Graphics tablet pen"
	   MatchIsTablet "on"
       MatchProduct "UPIN.INC 906"
       MatchDevicePath "/dev/input/event*"
	   Driver "evdev"
EndSection
```
The only thing i would like is to make it work on my left monitor and have the pressure sensitivity on

nic3-14159, 2021-06-08:
Hello! Could you post the diagnostics as described on [this page](https://digimend.github.io/support/howto/trbl/diagnostics/)? That would be very helpful for getting things like the buttons working properly.

kireita, 2021-06-08:
here they are, but if the pen gets close it inmedeatly starts readin so im sorry if there is inacurate data, just tell me what i need to repeat if something was not clear.

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6617089/descriptors.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6617090/frame_buttons.txt)
[frame_wheel_srolling.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6617091/frame_wheel_srolling.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6617092/hid_report_descriptors.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6617093/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6617094/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6617095/pen_pressure.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6617096/pen_tilt.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6617097/probe.txt)