DIGImend/digimend-kernel-drivers #545: Pls Add support HUION KD200

RoseAround2021, 2021-06-26:
I bought a new handwriting tablet HUION KD200, but found that it does not support Debian 10. Can you provide support? Thank you very much!

nic3-14159, 2021-06-26:
Hello! That's a pretty interesting tablet, and it would be pretty cool to get fully working in Linux. Could you collect and post diagnostics as described on [this page?](https://digimend.github.io/support/howto/trbl/diagnostics/) That would be a great help for adding support for this tablet. Thanks!

Tamas-Toth-ebola, 2021-12-21:
Hi!

I just found this thread now, and I have KD200 also. Could I help with anything to implement the device into DIGImend for properly configure the tablet?

Thanks for your efforts in advance!

nic3-14159, 2021-12-21:
@Tamas-Toth-ebola Hello! It would be great if you could post diagnostics as I outlined in my previous comment, as the original poster hasn't responded. Thanks!

Tamas-Toth-ebola, 2021-12-21:
@nic3-14159 thanks for the fast response! I have to ask some patience in the next days, but hope in one week I could send all those data what you mentioned!

Tamas-Toth-ebola, 2022-01-07:
Dear @nic3-14159 !

Sorry for the late reaction but I had not time for it before.

Firstly I wish you a Happy New Year, and here I'm attaching all the required logs.

[Huion - KD200.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/7831195/Huion.-.KD200.zip)

I tried to catch more samples as you wish and to avoid misunderstands I made 3 discrete report files for the keys/buttons. They contains firstly the top button near the power button also.

Thanks for your efforts and support in advance!

Tamas-Toth-ebola, 2022-01-07:
One more info... I got the following console response when I tried to make the descriptors report, but at the end all info/data are in the corresponding file, so seemed ok.

```
sudo lsusb -v -d $T > descriptors.txt
[sudo] password for ebola: 
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
```

nic3-14159, 2022-01-08:
@Tamas-Toth-ebola After looking at the samples you sent, it doesn't seem like this tablet does anything different from existing Huion tablets supported by DIGImend. I'm not 100% sure about this, but I have a feeling the existing driver will work with the tablet as is without any changes to the code. Could you test everything with the existing driver and report if anything isn't working properly?

K1-K5 as well as the button to the left of the power button should function as tablet shortcut keys, and everything else should work as standard keyboard keys, with the num key toggling the upper and lower functions printed on the keys.

Tamas-Toth-ebola, 2022-01-09:
Hi @nic3-14159 ,

I tried you suggested 'solution' but without success yet. I'm using the latest Ubuntu 21.10, what out-of-box did not handle the keys of the device. I installed the latest .deb package for 'secure the installation' restarted my the machine but at the end the result was the same as before.

#### Ubuntu' default tablet settings

On Ubuntu's default tablet settings I got the following:
[image]

Of course it would be nice if the device itself could be `Huion - Inspiroy Keydial - KD200` instead of `Huion H950P`, but not this is the real problem.

In these settings the pen button mapping seems work with also the tablet button mapping, but these whole settings just seems to bo for mouse mode or something like.

I can map the 2 pen button, and also the 5 `Kx` button as well the left button near the power button. Also `num` button seems to can switch it's state. But sadly none of the other buttons works neither the dial.

#### Krita

All what I wrote is for the default Ubuntu tablet settings. But when I tried to use the tablet with Krita the buttons also seems to be not valid. Lower button on the pen not trigger the eraser _(instead of it is give me the canvas mover)_, but top button trigger the proper brush selector overlay control panel but that is the default right click mouse action so seems nothing different from what I wrote at default Ubuntu settings. Of course neither of the normal keys nor the dial work also.

Maybe I missed something for the proper setup, in this case apologize it and please light it up for me but seems the straight-forward simple usage not work yet.

Thanks for your efforts and help in advance!

AldoLinx, 2022-01-18:
I tested this tablet on Manjaro KDE and it seems to work great. Without touching anything in the configuration seems that the keyboard works with the exception of the numbers and de wheel.

Tamas-Toth-ebola, 2022-01-18:
@AldoLinx hi!

It's a really interesting result as on my Ubuntu + Gnome the special Kx numeric keys can be mapped to custom functions in the system settings, as the special switch key on the left side of the power button also. But neither the dial/whel nor the other keys work.

_(Stock Ubuntu 21.10 + latest DIGImend .deb package from here/GitHub (what later automatically updated from the distro repo...))_

If I will have time for it I will test it with some sterile VirtualBox - Linux guests...

AldoLinx, 2022-01-18:
Some problems I found after testing for an hour. Maybe some of these are related to the KDE tablet settings I don't know.

1- If I move the dial/wheel it locks something that doesn't allow me to click even with my mouse. But if I press the special switch key it goes back to the normal behavior.

2- I can configure all the Kx numeric keys but only K1, K2 and K3 works, for some reason K4 and K5 doesn't do anything.

3- The configuration panel only seems to detect the tablet if its connected with the usb cable. The tablet works fine wirelessly but the settings panel can't detect it correctly.

Sorry for my bad English.

Tamas-Toth-ebola, 2022-01-19:
@AldoLinx your comment drop some light to my test environent. I always used the tablet with it's official own bluetooth dongle through wirelessly. My already sent test/debug info based also on wireless connection.

Will test it through wired connection also if I have time.

Tamas-Toth-ebola, 2022-01-21:
Ok, I had tested on stock 'Manjaro - GNOME' VM yesterday with wireless connection though the tablet's own bluetooth dongle, with the following results.

The normal keys work fine out of box. Without DIGImend driver in GNOME's tablet settings I could see only the tablet, without pen. I could map the special Kx buttons _(but just in one state so no differences with the NUM on/off at the end, while in Windows the 2 state give different results)_ and the switch near the power button.

With DIGImend driver _(installed through AUR)_ I can see the pen also, nut just the tablet, and can map it's button also, but nor for any sensible result, so I need to map them manually through the input device config file _(https://github.com/DIGImend/digimend-kernel-drivers/issues/26#issuecomment-192761796)_ what I had not tried yet.

The only generally not working thing the dial pad. It does not work with or without DIGImend drivers.

So my final opinion with that test environment:

1. The biggest problem is the unworkable dial wheel.
2. If we could map the Kx keys in the proper way, when in NUM on state they could give us normal number keys, that would be the proper result, but basically we can use them.
3. If tablet settings could show us the proper tablet name that would be nice, instead of H950, but of course this is just a 'visual' problem.
4. If there could be GUI solution for maping all keys _(especially the pen buttons what can be remapped but just onto predefined events instead of custom keypresses as for eg.: 'E' for erase in Krita)_, that would be nice.
5. _(GUI screen mapping could be the ice on the cake.)_

Now I have to fine the problem with my Ubuntu workstation. Will check the tablet with stock Ubuntu VM. If it gave me the same result as Manjaro, I have to find what hinders it's operation in my working environment. Maybe any HID relevant software as Solaar for my Logitech input devices, who knows?!... and how to solve it?!...

Are there any debug options through this way? Any opinions?

Tamas-Toth-ebola, 2022-01-22:
Yesterday I tested the tablet with latest stock Ubuntu _(21.10)_; Debian _(11.2)_; and Manjaro _(21.2.1)_ again.

On Debian and Ubuntu I got the same results _(not surprisingly)_, both the pen and the tablet seems without any custom settings in the offical GNOME tablet settings. DIGImend driver basically not needed for that, but sadly the default normal keys not work. Neither of them. I can map the Kx keys as I mentionesd before, can map the pen, but not onto any sensible thing. Sadly the diap wheel does not work. Neither with the stock OS, nor with DIGImend driver.

**So on Debian based system normal keys and the dial pad does not work, but the other thing yes. Did not try custom button map with DIGIMEND driver.**

On Manjaro I got the same result as before. Without DIGImend stock GNOME tablet settings just show me the tablet settings but not the pen settings. After installing DIGImend through AUR, I got the pen settings also.

**On Manjaro all the normal keys work fin out of box, but dial wheel does not with or without DIGImend. Did not try custom button map with DIGIMEND driver.**

I'm here now. Sadly on Ubuntu, I don't know but default/normal keys deos not work, as deal wheel also not. Neither with nor without DIGImend driver.

nic3-14159, 2022-01-24:
@Tamas-Toth-ebola 
> The normal keys work fine out of box. Without DIGImend driver in GNOME's tablet settings I could see only the tablet, without pen. I could map the special Kx buttons _(but just in one state so no [...]

The Windows driver must be doing something special to allow that then. From the diagnostics you posted, the Kx buttons always seemed to send tablet button events, regardless of the state of the NUM key. One of the [FAQs on Huion's site](https://support.huion.com/en/support/solutions/articles/44002135144) says something interesting about those keys:
> Whether the driver is installed or not, keys K1 to K5 are served as letter keys T, G, O, P and L correspondingly by default.

So it seems like they have a mode where those keys operate as a keyboard and send standard keyboard events and not tablet button events. Without a driver, the keyboard portion should work with the generic HID drivers and work as T/1, G/2, O/3, P/4, and L/5. The wording seems to indicate that the Windows driver leaves those keys in the generic keyboard mode by default, but I'm not sure what the behavior is once you add mappings.

In Windows, are you able to add mappings to the Kx buttons in both the NUM on state and NUM off state, or only the the NUM off state? It it's the latter, do the Kx still work as 1-5 with NUM on (and the mapping with num off)? If this is the case, we probably have to do more research into what the Windows driver does.

> The only generally not working thing the dial pad. It does not work with or without DIGImend drivers.
> 
> So my final opinion with that test environment:
>
>     1. The biggest problem is the unworkable dial wheel.

Could you try running `sudo evtest`, select the device corresponding to the pad interface on your tablet, then use the dial? It should print wheel events if the DIGImend kernel driver is working correctly.

>     3. If tablet settings could show us the proper tablet name that would be nice, instead of H950, but of course this is just a 'visual' problem.

That should be fixable, but I think that also requires some work in libwacom to get the names in the KDE and GNOME GUI tools to show up correctly.

>     4. If there could be GUI solution for maping all keys _(especially the pen buttons what can be remapped but just onto predefined events instead of custom keypresses as for eg.: 'E' for erase in Krita)_, that would be nice.

This would need to be handled in other software, as this repo is only meant to be a kernel driver that gets the kernel to properly recognize events and send them to userspace (although perhaps the DIGImend project could work on such a software). The GNOME and KDE desktops have some GUIs for configuring tablet settings, but I believe some work needs to be done in libwacom to get them working properly with this tablet. The best thing I've seen for a universal GUI tool for remapping keys/buttons is [input-remapper](https://github.com/sezanzeb/input-remapper)

>     5. _(GUI screen mapping could be the ice on the cake.)_
Again, this would need to be done outside of this repo. I think KDE has this feature, not sure about GNOME. I don't know of any universal tool for this.

> Are there any debug options through this way? Any opinions?

There are several layers to the whole input system in Linux, and all of them have to work correctly for tablets to work. I think the DIGImend driver isn't working correctly with the keyboard/buttons, so I'll have to look into it some more. The Ubuntu settings stuff probably doesn't work because this tablet isn't in libwacom (most tablet related stuff on Linux is geared toward Wacom tablets and the tablets supported by DIGImend are sort of wedged in. That also does somewhat steer/limit design decisions because what we do needs to be compatible with whatever the Linux Wacom project does)

There are some debug options [here](https://digimend.github.io/support/howto/trbl/locating_failure/) for figuring why things don't work. I'd recommend starting with the "Debugfs interface" and the "Evtest tool" sections, as those should give an indication if the DIGImend driver (one of the first laters in the whole input stack) is working correctly.

Tamas-Toth-ebola, 2022-01-24:
Thanks @nic3-14159 for your detailed answers!

- All GUI relevant information is totally understandalbe. Your software theoretically is really responsible only to handle the device on driver level. I just mentioned them because with those user level options could be the Linux tablet handling equal or basically even better than on Windows. _(Anyway I do not use Windows for more then 10 years, while this actual tablet is my daughter who use it on Windows.)_
- Will test the Windows working mode through Huion's driver, what you asked.
- Also will give you the required dial wheel logs. _(Just asking that the pack I sent before does not contain the required informations?)_
- If I have enough time will try to debug something. Now I just know that normal keys working fine on Manjaro, but sady on Ubuntu not at all, while the dial wheel is the other problematic, but that one does not work on even Manjaro.

Thanks again for your efforts!

Tamas-Toth-ebola, 2022-01-26:
@nic3-14159 I had a quick idea.

The difference between the tested Manjaro and Ubuntu basically was 'just' the kernel. Latest Manjaro uses 5.15 while latest Ubuntu uses just 5.13. Theoretically this also could be a reason why on Manjaro the normal keys work and on Ubuntu not.

Sadly this was just an idea as after I have upgraded my Ubuntu's kernel to the pruposed 5.15 the situation is the same as before. Normal keys still not work while on Manjaro it does.

So not the kernel difference is the reason of the working/not working normal keys.

inochisa, 2023-01-06:
Hi @Tamas-Toth-ebola.

I am now working on support of the KD200 tablet. It seems when use wireless, the tablet report a different device id. As my device reports `256c:0064` when using usb cable, but it reports `256c:8251` when using wireless. Do you have the same situation?

Tamas-Toth-ebola, 2023-01-10:
Hi @inochisa !

Sorry for the delay but my PC _(Intel - NUC)_ was in service in the last some days and I did not want to play the tablet on my temp. machine. But now as I got my device back, I cheked it.

In wired mode:
```
Bus 001 Device 008: ID 256c:006d HUION Huion Tablet_KD200
```
In wireless mode:
```
Bus 001 Device 009: ID 256c:006d HUION Huion_KD200_Dongle
```

As you can see the 2 devID are the same but the device name is different a bit. I hope it could help you, and for us :smile:

inochisa, 2023-01-13:
Hi @Tamas-Toth-ebola, thank you so much for your efforts. These are enough for me.

As your informations shows, it seems there are no major difference between our tablets. So I think I can use my own tablet to finish all the test.

mproffitt, 2023-03-11:
I bought this tablet this week despite knowing of this thread and am now also trying to understand how to make it work.

```
Ubuntu Studio 22.10
Kernel 5.19.0-1018-lowlatency #19-Ubuntu SMP PREEMPT_DYNAMIC
```

Regarding the usage of built-in wireless over using the provided dongle, the device reports differently to the USB IDs and instead contains a different bus and product number. This is also evident inside the `hid-recorder` info from @Tamas-Toth-ebola. (`I: 5 256c 8251`)

This, I believe is the relevant section from `udevadm monitor --udev --env` when using built in wireless.

```
UDEV  [42142.131897] add      /devices/virtual/misc/uhid/0005:256C:8251.0039 (hid)
ACTION=add
DEVPATH=/devices/virtual/misc/uhid/0005:256C:8251.0039
SUBSYSTEM=hid
HID_ID=0005:0000256C:00008251
HID_NAME=Inspiroy Keydial-096
HID_PHYS=68:54:5a:90:ae:15
HID_UNIQ=20:20:12:01:04:48
MODALIAS=hid:b0005g0001v0000256Cp00008251
SEQNUM=15104
USEC_INITIALIZED=42142125905
DRIVER=hid-generic
```

Here there is both a different bus id (`0005` over `0003`) and a different product ID `8251` over `006d`

Adding the device to the `xorg.conf` actually allows this to function in the same manner as the `uclogic` driver using a custom tablet mapping in KDE but again without the dial. This is livable, but not ideal.

Custom tablet config in (`~/.config/tabletdblocalrc`)
```
[8251]
layout=unknown
model=unknown
name=Inspiroy Keydial-096 S
padbuttons=8
statusleds=4
touchring=false
touchsensorid=
touchstripl=false
touchstripr=false
wheel=false
```

One thing I notice on USB, using the PR from @inochisa is that the dial does show up as an input device but I can't see any way of binding to that. 

```
[46543.199683] input: HUION Huion Tablet_KD200 Pen as /devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb5/5-2/5-2.1/5-2.1:1.0/0003:256C:006D.004A/input/input176
[46543.199782] input: HUION Huion Tablet_KD200 Pad as /devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb5/5-2/5-2.1/5-2.1:1.0/0003:256C:006D.004A/input/input177
[46543.199843] input: HUION Huion Tablet_KD200 Touch Strip as /devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb5/5-2/5-2.1/5-2.1:1.0/0003:256C:006D.004A/input/input178
[46543.199899] input: HUION Huion Tablet_KD200 Dial as /devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb5/5-2/5-2.1/5-2.1:1.0/0003:256C:006D.004A/input/input179
```

This last device does not show up when using the generic driver over builtin bluetooth wireless.

inochisa, 2023-03-17:
> Regarding the usage of built-in wireless over using the provided dongle, the device reports differently to the USB IDs and instead contains a different bus and product number. This is also evident inside the `hid-recorder` info from @Tamas-Toth-ebola. (`I: 5 256c 8251`)

The bluetooth interface can not use since I did not find a way to send the ble att packet while initializing hid interface.

> Adding the device to the `xorg.conf` actually allows this to function in the same manner as the `uclogic` driver using a custom tablet mapping in KDE but again without the dial. This is livable, but not ideal.

The dial of KD200 can be disabled by sending usb ctrl message. You can enable it by send  usb ctrl message `0xd3` (or `0xd4`, I forgot which one is correct), or you can use windows driver to enable it.

> One thing I notice on USB, using the PR from @inochisa is that the dial does show up as an input device but I can't see any way of binding to that.

I have prepared a new code based on parsing report descriptor, but these day I am working on another important project. I plan to restart this in one month or two.

JoseExposito, 2024-05-03:
Hi!

Your issue should be fixed by [DIGImend v13](https://github.com/DIGImend/digimend-kernel-drivers/releases/tag/v13). Please, if you have a chance, give it a try and report any problems found.

Thanks!

closed 2024-05-03 by JoseExposito