DIGImend/digimend-kernel-drivers #546: disk full when running make

rodralvarez, 2021-07-05:
Hi, I have ~18 Gb left in the / partition and when running 'sudo make' it takes a long time and the disk has no space left so it fails.
After running sudo make I have to run sudo make clean in order to recover recover the 18Gb. 
Is this expected?

nic3-14159, 2021-07-05:
That is certainly not expected. So after running make, it shows up as 0 GB free on /, and back to 18GB free after running make clean?

nic3-14159, 2021-07-05:
Maybe try building DIGImend again using `make --debug` and posting the output? That might give a clue to what is going on. The fact that it takes a long time and then fails with no disk space left makes me think it somehow entered an infinite loop, continually creating files and filling up the disk space, or something like that.

rodralvarez, 2021-07-07:
Hi, thanks a lot for your reply, sorry the delay on this,

> That is certainly not expected. So after running make, it shows up as 0 GB free on /, and back to 18GB free after running make clean?

Yes, I mean not exactly 0 GB left, but ~200kb free.

> Maybe try building DIGImend again using make --debug and posting the output? That might give a clue to what is going on. The fact that it takes a long time and then fails with no disk space left [...]

Please see the attached file, I had to stop make --debug, otherwise it'll go on and on a for a few hours filling up the disk. I hope that helps.

[make--debug_output.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/6779570/make--debug_output.txt)

nic3-14159, 2021-07-08:
It looks like you are building from the DIGImend-kernel-drivers-for-Huion repo. That repo looks fairly out of date, and I believe there were some updates to the Makefile in the digimend-kernel-drivers repo (this repo) that fixed some issues with building on more up to date systems. I suspect the line 
`$(MAKE) -C $(KDIR) SUBDIRS=$(PWD) $@`
in the Makefile (line 22) may be causing your issue, based on some build failures I've seen before. However, none were as serious as going on forever and filling up the disk. It probably depends on the distro and how it is set up. If you want you could try changing that line to
`$(MAKE) -C $(KDIR) M=$(PWD) $@`
and see if that works, but I think a better solution would be building from the latest release of this repo. As far as I can tell, that Huion repo is exactly the same as this repo, just out of date.

Hope that helps

rodralvarez, 2021-07-08:
Hi, thanks a lot, I downloaded the latest version of the tar.gz and it worked, I previously used git clone --recurse-submodules, I have no clue why it downloaded and old version.
The pressure of the pen is still not recognized, but that's another issue, I'll try to fix it and look at previous issues posted here.

Thanks.

closed 2021-07-10 by rodralvarez