DIGImend/digimend-kernel-drivers #547: Huion Inspiroy 950P has vendor/product id of another tablet; not detected.

aero31aero, 2021-07-11:
The pen tablet works (tried in Krita, the pressure sensitivity works as expected) but xsetwacom, kde settings, etc do not detect it as a pen input or allow changing the buttons or the target area.

The device id in `lsusb` output (`256:006d`) matches [HS610](https://digimend.github.io/tablets/Huion_HS610/) which is an unsupported device, however I purchased a [H950P](https://digimend.github.io/tablets/Huion_H950P/).

What should I do to debug this further?

---

Other outputs:

```
$ sudo uclogic-probe | uclogic-decode
 c8 |        Params block #2: ??k?????
    |                          Max X: 44200
    |                          Max Y: 27600
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |       Firmware version: HUION_T172_190325
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
```

```
$ sudo udevadm info
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:256C:006D.000F/input/input49/event17
DEVNAME=/dev/input/event17
MAJOR=13
MINOR=81
SUBSYSTEM=input
USEC_INITIALIZED=220660890069
ID_INPUT=1
ID_INPUT_TABLET=1
ID_INPUT_TABLET_PAD=1
ID_VENDOR=256c
ID_VENDOR_ENC=256c
ID_VENDOR_ID=256c
ID_MODEL=006d
ID_MODEL_ENC=006d
ID_MODEL_ID=006d
ID_REVISION=0000
ID_SERIAL=256c_006d
ID_TYPE=hid
ID_BUS=usb
ID_USB_INTERFACES=:030102:
ID_USB_INTERFACE_NUM=00
ID_USB_DRIVER=usbhid
ID_PATH=pci-0000:00:14.0-usb-0:2:1.0
ID_PATH_TAG=pci-0000_00_14_0-usb-0_2_1_0
LIBINPUT_DEVICE_GROUP=3/256c/6d:usb-0000:00:14.0-2
DEVLINKS=/dev/input/by-id/usb-256c_006d-event-mouse /dev/input/by-path/pci-0000:00:14.0-usb-0:2:1.0-event-mouse
```

nic3-14159, 2021-07-11:
Huion likes to do a rather annoying thing where they assign the same USB IDs to a bunch of their tablets. As far as I know, there are only 2: 256c:006e and 256c:006d. From some other issues it looks like Huion has different revisions or firmware for this tablet that use different USB ids. 256c:006d is a newer ID, and should be supported as of the v10 release, but something must be causing it to fail with your tablet. 

Could you try running `sudo digimend-debug 1` to turn on some debug messsges, plug in your tablet, and then post the output of `dmesg` ?

aero31aero, 2021-07-11:
I should have the latest version of digimend drivers as they are shipped with the Arch kernel, but I went ahead and clone;make;make installed the latest git tree as well. Same result.

dmesg output with debugging on:

```
$ sudo dmesg # log after plugging in the tablet
[17415.382658] usb 1-2: new full-speed USB device number 8 using xhci_hcd
[17415.600824] usb 1-2: New USB device found, idVendor=256c, idProduct=006d, bcdDevice= 0.00
[17415.600833] usb 1-2: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[17415.611314] hid_uclogic:uclogic_params_huion_init:862: uclogic 0003:256C:006D.0007: pen v2 parameters found
[17415.611364] hid_uclogic:uclogic_probe:224: uclogic 0003:256C:006D.0007: parameters:
[17415.611371] hid_uclogic:uclogic_params_hid_dbg:116: uclogic 0003:256C:006D.0007: .invalid = false
[17415.611378] hid_uclogic:uclogic_params_hid_dbg:118: uclogic 0003:256C:006D.0007: .desc_ptr = 0000000000000000
[17415.611384] hid_uclogic:uclogic_params_hid_dbg:119: uclogic 0003:256C:006D.0007: .desc_size = 0
[17415.611389] hid_uclogic:uclogic_params_hid_dbg:120: uclogic 0003:256C:006D.0007: .pen = {
[17415.611395] hid_uclogic:uclogic_params_pen_hid_dbg:59: uclogic 0003:256C:006D.0007: 	.usage_invalid = false
[17415.611401] hid_uclogic:uclogic_params_pen_hid_dbg:61: uclogic 0003:256C:006D.0007: 	.desc_ptr = 00000000fc4f6a17
[17415.611406] hid_uclogic:uclogic_params_pen_hid_dbg:62: uclogic 0003:256C:006D.0007: 	.desc_size = 115
[17415.611411] hid_uclogic:uclogic_params_pen_hid_dbg:63: uclogic 0003:256C:006D.0007: 	.id = 8
[17415.611417] hid_uclogic:uclogic_params_pen_hid_dbg:64: uclogic 0003:256C:006D.0007: 	.subreport_list = {
[17415.611424] hid_uclogic:uclogic_params_pen_hid_dbg:66: uclogic 0003:256C:006D.0007: 		{0xe0, 247},
[17415.611430] hid_uclogic:uclogic_params_pen_hid_dbg:66: uclogic 0003:256C:006D.0007: 		{0xf0, 248},
[17415.611438] hid_uclogic:uclogic_params_pen_hid_dbg:66: uclogic 0003:256C:006D.0007: 		{0xf1, 249}
[17415.611444] hid_uclogic:uclogic_params_pen_hid_dbg:71: uclogic 0003:256C:006D.0007: 	}
[17415.611449] hid_uclogic:uclogic_params_pen_hid_dbg:72: uclogic 0003:256C:006D.0007: 	.inrange = none
[17415.611454] hid_uclogic:uclogic_params_pen_hid_dbg:74: uclogic 0003:256C:006D.0007: 	.fragmented_hires = true
[17415.611460] hid_uclogic:uclogic_params_pen_hid_dbg:76: uclogic 0003:256C:006D.0007: 	.tilt_y_flipped = true
[17415.611467] hid_uclogic:uclogic_params_hid_dbg:122: uclogic 0003:256C:006D.0007: 	}
[17415.611472] hid_uclogic:uclogic_params_hid_dbg:123: uclogic 0003:256C:006D.0007: .frame_list = {
[17415.611478] hid_uclogic:uclogic_params_hid_dbg:125: uclogic 0003:256C:006D.0007: 	{
[17415.611483] hid_uclogic:uclogic_params_frame_hid_dbg:91: uclogic 0003:256C:006D.0007: 		.desc_ptr = 00000000c6e7d7c9
[17415.611489] hid_uclogic:uclogic_params_frame_hid_dbg:92: uclogic 0003:256C:006D.0007: 		.desc_size = 70
[17415.611494] hid_uclogic:uclogic_params_frame_hid_dbg:93: uclogic 0003:256C:006D.0007: 		.id = 247
[17415.611500] hid_uclogic:uclogic_params_frame_hid_dbg:94: uclogic 0003:256C:006D.0007: 		.suffix = (null)
[17415.611505] hid_uclogic:uclogic_params_frame_hid_dbg:95: uclogic 0003:256C:006D.0007: 		.re_lsb = 0
[17415.611513] hid_uclogic:uclogic_params_frame_hid_dbg:96: uclogic 0003:256C:006D.0007: 		.dev_id_byte = 0
[17415.611518] hid_uclogic:uclogic_params_frame_hid_dbg:97: uclogic 0003:256C:006D.0007: 		.touch_byte = 0
[17415.611525] hid_uclogic:uclogic_params_frame_hid_dbg:98: uclogic 0003:256C:006D.0007: 		.touch_max = 0
[17415.611530] hid_uclogic:uclogic_params_frame_hid_dbg:99: uclogic 0003:256C:006D.0007: 		.touch_flip_at = 0
[17415.611536] hid_uclogic:uclogic_params_frame_hid_dbg:101: uclogic 0003:256C:006D.0007: 		.bitmap_dial_byte = 0
[17415.611541] hid_uclogic:uclogic_params_hid_dbg:127: uclogic 0003:256C:006D.0007: 	},
[17415.611546] hid_uclogic:uclogic_params_hid_dbg:125: uclogic 0003:256C:006D.0007: 	{
[17415.611550] hid_uclogic:uclogic_params_frame_hid_dbg:91: uclogic 0003:256C:006D.0007: 		.desc_ptr = 00000000b58f29b5
[17415.611554] hid_uclogic:uclogic_params_frame_hid_dbg:92: uclogic 0003:256C:006D.0007: 		.desc_size = 79
[17415.611557] hid_uclogic:uclogic_params_frame_hid_dbg:93: uclogic 0003:256C:006D.0007: 		.id = 248
[17415.611561] hid_uclogic:uclogic_params_frame_hid_dbg:94: uclogic 0003:256C:006D.0007: 		.suffix = Touch Strip
[17415.611565] hid_uclogic:uclogic_params_frame_hid_dbg:95: uclogic 0003:256C:006D.0007: 		.re_lsb = 0
[17415.611569] hid_uclogic:uclogic_params_frame_hid_dbg:96: uclogic 0003:256C:006D.0007: 		.dev_id_byte = 4
[17415.611572] hid_uclogic:uclogic_params_frame_hid_dbg:97: uclogic 0003:256C:006D.0007: 		.touch_byte = 5
[17415.611576] hid_uclogic:uclogic_params_frame_hid_dbg:98: uclogic 0003:256C:006D.0007: 		.touch_max = 8
[17415.611579] hid_uclogic:uclogic_params_frame_hid_dbg:99: uclogic 0003:256C:006D.0007: 		.touch_flip_at = 0
[17415.611583] hid_uclogic:uclogic_params_frame_hid_dbg:101: uclogic 0003:256C:006D.0007: 		.bitmap_dial_byte = 0
[17415.611586] hid_uclogic:uclogic_params_hid_dbg:127: uclogic 0003:256C:006D.0007: 	},
[17415.611590] hid_uclogic:uclogic_params_hid_dbg:125: uclogic 0003:256C:006D.0007: 	{
[17415.611594] hid_uclogic:uclogic_params_frame_hid_dbg:91: uclogic 0003:256C:006D.0007: 		.desc_ptr = 00000000fde70e54
[17415.611597] hid_uclogic:uclogic_params_frame_hid_dbg:92: uclogic 0003:256C:006D.0007: 		.desc_size = 83
[... cut ...]
[17415.618803] hid_uclogic:uclogic_params_hid_dbg:127: uclogic 0003:256C:006D.0008: 	}
[17415.618810] hid_uclogic:uclogic_params_hid_dbg:130: uclogic 0003:256C:006D.0008: }
[17415.619093] uclogic 0003:256C:006D.0008: No inputs registered, leaving
[17415.619231] uclogic 0003:256C:006D.0008: hidraw4: USB HID v1.11 Device [HID 256c:006d] on usb-0000:00:14.0-2/input1
```

Ultimately, it says: `[17415.619093] uclogic 0003:256C:006D.0008: No inputs registered, leaving`. My hunch is the driver sees the usb product id and decides that its not something it supports.

aero31aero, 2021-07-11:
I can confirm that my issue is different from #308 since the latest tree should recognize my device id `#define USB_DEVICE_ID_HUION_TABLET2	0x006d`.

nic3-14159, 2021-07-11:
Just checking, you did run `sudo modprobe -r hid-kye hid-uclogic hid-polostar hid-viewsonic` or rebooted after installing, right? Otherwise the kernel could still be using the old driver. Also, although the digimend drivers are eventually merged into the kernel tree, I don't think v10 has been merged yet, so the driver in the Arch kernel likely would not have worked with the tablet.

aero31aero, 2021-07-11:
I did. Just to be sure, I added a debug message of my own and it got printed so I'm definitely on the latest version.

Just in case I messed up earlier, here's the new log:

```
[  182.509749] usb 1-2: new full-speed USB device number 11 using xhci_hcd
[  182.718798] usb 1-2: New USB device found, idVendor=256c, idProduct=006d, bcdDevice= 0.00
[  182.718806] usb 1-2: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[  182.730317] hid_uclogic:uclogic_params_huion_init:862: uclogic 0003:256C:006D.0010: pen v2 parameters found
[  182.730327] hid_uclogic:uclogic_params_init:1182: uclogic 0003:256C:006D.0010: DEBUG --------------- rc = 0
[  182.730332] hid_uclogic:uclogic_probe:224: uclogic 0003:256C:006D.0010: parameters:
[  182.730336] hid_uclogic:uclogic_params_hid_dbg:116: uclogic 0003:256C:006D.0010: .invalid = false
[  182.730341] hid_uclogic:uclogic_params_hid_dbg:118: uclogic 0003:256C:006D.0010: .desc_ptr = 0000000000000000
[  182.730345] hid_uclogic:uclogic_params_hid_dbg:119: uclogic 0003:256C:006D.0010: .desc_size = 0
[  182.730348] hid_uclogic:uclogic_params_hid_dbg:120: uclogic 0003:256C:006D.0010: .pen = {
[  182.730352] hid_uclogic:uclogic_params_pen_hid_dbg:58: uclogic 0003:256C:006D.0010: DEBUG --------------- HELLO CUSTOM VERSION
[  182.730356] hid_uclogic:uclogic_params_pen_hid_dbg:59: uclogic 0003:256C:006D.0010: 	.usage_invalid = false
[  182.730360] hid_uclogic:uclogic_params_pen_hid_dbg:61: uclogic 0003:256C:006D.0010: 	.desc_ptr = 0000000033388785
[  182.730364] hid_uclogic:uclogic_params_pen_hid_dbg:62: uclogic 0003:256C:006D.0010: 	.desc_size = 115
[  182.730368] hid_uclogic:uclogic_params_pen_hid_dbg:63: uclogic 0003:256C:006D.0010: 	.id = 8
[  182.730371] hid_uclogic:uclogic_params_pen_hid_dbg:64: uclogic 0003:256C:006D.0010: 	.subreport_list = {
[  182.730376] hid_uclogic:uclogic_params_pen_hid_dbg:66: uclogic 0003:256C:006D.0010: 		{0xe0, 247},
[  182.730380] hid_uclogic:uclogic_params_pen_hid_dbg:66: uclogic 0003:256C:006D.0010: 		{0xf0, 248},
[  182.730384] hid_uclogic:uclogic_params_pen_hid_dbg:66: uclogic 0003:256C:006D.0010: 		{0xf1, 249}
[  182.730387] hid_uclogic:uclogic_params_pen_hid_dbg:71: uclogic 0003:256C:006D.0010: 	}
[  182.730391] hid_uclogic:uclogic_params_pen_hid_dbg:72: uclogic 0003:256C:006D.0010: 	.inrange = none
[  182.730395] hid_uclogic:uclogic_params_pen_hid_dbg:74: uclogic 0003:256C:006D.0010: 	.fragmented_hires = true
[  182.730398] hid_uclogic:uclogic_params_pen_hid_dbg:76: uclogic 0003:256C:006D.0010: 	.tilt_y_flipped = true
[  182.730401] hid_uclogic:uclogic_params_hid_dbg:122: uclogic 0003:256C:006D.0010: 	}
[  182.730404] hid_uclogic:uclogic_params_hid_dbg:123: uclogic 0003:256C:006D.0010: .frame_list = {
[  182.730408] hid_uclogic:uclogic_params_hid_dbg:125: uclogic 0003:256C:006D.0010: 	{
[  182.730412] hid_uclogic:uclogic_params_frame_hid_dbg:91: uclogic 0003:256C:006D.0010: 		.desc_ptr = 000000006f728bd4
[  182.730415] hid_uclogic:uclogic_params_frame_hid_dbg:92: uclogic 0003:256C:006D.0010: 		.desc_size = 70
[  182.730419] hid_uclogic:uclogic_params_frame_hid_dbg:93: uclogic 0003:256C:006D.0010: 		.id = 247
[  182.730423] hid_uclogic:uclogic_params_frame_hid_dbg:94: uclogic 0003:256C:006D.0010: 		.suffix = (null)
[  182.730426] hid_uclogic:uclogic_params_frame_hid_dbg:95: uclogic 0003:256C:006D.0010: 		.re_lsb = 0
[  182.730429] hid_uclogic:uclogic_params_frame_hid_dbg:96: uclogic 0003:256C:006D.0010: 		.dev_id_byte = 0
[  182.730433] hid_uclogic:uclogic_params_frame_hid_dbg:97: uclogic 0003:256C:006D.0010: 		.touch_byte = 0
[  182.730436] hid_uclogic:uclogic_params_frame_hid_dbg:98: uclogic 0003:256C:006D.0010: 		.touch_max = 0
[  182.730440] hid_uclogic:uclogic_params_frame_hid_dbg:99: uclogic 0003:256C:006D.0010: 		.touch_flip_at = 0
[  182.730443] hid_uclogic:uclogic_params_frame_hid_dbg:101: uclogic 0003:256C:006D.0010: 		.bitmap_dial_byte = 0
[  182.730447] hid_uclogic:uclogic_params_hid_dbg:127: uclogic 0003:256C:006D.0010: 	},
[  182.730450] hid_uclogic:uclogic_params_hid_dbg:125: uclogic 0003:256C:006D.0010: 	{
[  182.730453] hid_uclogic:uclogic_params_frame_hid_dbg:91: uclogic 0003:256C:006D.0010: 		.desc_ptr = 0000000063712f69
[  182.730457] hid_uclogic:uclogic_params_frame_hid_dbg:92: uclogic 0003:256C:006D.0010: 		.desc_size = 79
[  182.730460] hid_uclogic:uclogic_params_frame_hid_dbg:93: uclogic 0003:256C:006D.0010: 		.id = 248
[  182.730464] hid_uclogic:uclogic_params_frame_hid_dbg:94: uclogic 0003:256C:006D.0010: 		.suffix = Touch Strip
[  182.730467] hid_uclogic:uclogic_params_frame_hid_dbg:95: uclogic 0003:256C:006D.0010: 		.re_lsb = 0
[  182.730470] hid_uclogic:uclogic_params_frame_hid_dbg:96: uclogic 0003:256C:006D.0010: 		.dev_id_byte = 4
[  182.730474] hid_uclogic:uclogic_params_frame_hid_dbg:97: uclogic 0003:256C:006D.0010: 		.touch_byte = 5
[  182.730477] hid_uclogic:uclogic_params_frame_hid_dbg:98: uclogic 0003:256C:006D.0010: 		.touch_max = 8
[  182.730480] hid_uclogic:uclogic_params_frame_hid_dbg:99: uclogic 0003:256C:006D.0010: 		.touch_flip_at = 0
[  182.730484] hid_uclogic:uclogic_params_frame_hid_dbg:101: uclogic 0003:256C:006D.0010: 		.bitmap_dial_byte = 0
[  182.730487] hid_uclogic:uclogic_params_hid_dbg:127: uclogic 0003:256C:006D.0010: 	},
[  182.730490] hid_uclogic:uclogic_params_hid_dbg:125: uclogic 0003:256C:006D.0010: 	{
[  182.730494] hid_uclogic:uclogic_params_frame_hid_dbg:91: uclogic 0003:256C:006D.0010: 		.desc_ptr = 00000000e30acd7e
[... cut ...]
[  182.737658] hid_uclogic:uclogic_params_frame_hid_dbg:99: uclogic 0003:256C:006D.0011: 		.touch_flip_at = 0
[  182.737662] hid_uclogic:uclogic_params_frame_hid_dbg:101: uclogic 0003:256C:006D.0011: 		.bitmap_dial_byte = 0
[  182.737665] hid_uclogic:uclogic_params_hid_dbg:127: uclogic 0003:256C:006D.0011: 	}
[  182.737668] hid_uclogic:uclogic_params_hid_dbg:130: uclogic 0003:256C:006D.0011: }
[  182.737855] uclogic 0003:256C:006D.0011: No inputs registered, leaving
[  182.737967] uclogic 0003:256C:006D.0011: hidraw4: USB HID v1.11 Device [HID 256c:006d] on usb-0000:00:14.0-2/input1
```

aero31aero, 2021-07-14:
I have more info that I don't know what to make of. In Krita's tablet testing tool, if I press the tablet's buttons, I get this output:

```
Mouse press X=168 Y=158 B=1
Mouse release
Mouse press X=168 Y=158 B=4
Mouse release
Mouse press X=168 Y=158 B=2
Mouse release
Mouse press X=168 Y=158 B=8
Mouse release
Mouse press X=168 Y=158 B=16
Mouse release
Mouse press X=168 Y=158 B=32
Mouse release
Mouse press X=168 Y=158 B=64
Mouse release
Mouse press X=168 Y=158 B=128
Mouse release
```

So, it looks like for each button press the tablet is sending 1 byte with the bit corresponding to the button being on. This could help in hacking together some way to parse which button is being pressed but I would love not to resort to writing some hacky script to do this if xsetwacom can be made to work.

japareaggae, 2021-09-07:
Hello, I'm also affected by this issue on my H950P and I've been digging around trying to figure out what part of the input stack is causing the problem.

I did notice something interesting: using `libinput debug-events`, it seems that libinput knows that the buttons belong to a tablet:

<details>

```
-event29  TABLET_PAD_BUTTON +0.000s	  0 pressed (mode 0)
 event29  TABLET_PAD_BUTTON +0.100s	  0 released (mode 0)
 event29  TABLET_PAD_BUTTON +0.344s	  1 pressed (mode 0)
 event29  TABLET_PAD_BUTTON +0.470s	  1 released (mode 0)
 event29  TABLET_PAD_BUTTON +0.726s	  2 pressed (mode 0)
 event29  TABLET_PAD_BUTTON +0.860s	  2 released (mode 0)
 event29  TABLET_PAD_BUTTON +1.076s	  3 pressed (mode 0)
 event29  TABLET_PAD_BUTTON +1.210s	  3 released (mode 0)
 event29  TABLET_PAD_BUTTON +1.638s	  4 pressed (mode 0)
 event29  TABLET_PAD_BUTTON +1.742s	  4 released (mode 0)
 event29  TABLET_PAD_BUTTON +2.054s	  5 pressed (mode 0)
 event29  TABLET_PAD_BUTTON +2.186s	  5 released (mode 0)
 event29  TABLET_PAD_BUTTON +2.474s	  6 pressed (mode 0)
 event29  TABLET_PAD_BUTTON +2.590s	  6 released (mode 0)
 event29  TABLET_PAD_BUTTON +2.846s	  7 pressed (mode 0)
 event29  TABLET_PAD_BUTTON +2.988s	  7 released (mode 0)
```

</details>

Moving down the stack with `evtest`, it also knows that the buttons are from a tablet: 

<details>

```
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6d version 0x110
Input device name: "HUION Huion Tablet_H950P Pad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 256 (BTN_0)
    Event code 257 (BTN_1)
    Event code 258 (BTN_2)
    Event code 259 (BTN_3)
    Event code 260 (BTN_4)
    Event code 261 (BTN_5)
    Event code 262 (BTN_6)
    Event code 263 (BTN_7)
    Event code 264 (BTN_8)
    Event code 265 (BTN_9)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 331 (BTN_STYLUS)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max        1
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max        1
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
Event: time 1630984983.892473, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1630984983.892473, type 1 (EV_KEY), code 256 (BTN_0), value 1
Event: time 1630984983.892473, -------------- SYN_REPORT ------------
Event: time 1630984984.024470, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1630984984.024470, type 1 (EV_KEY), code 256 (BTN_0), value 0
Event: time 1630984984.024470, -------------- SYN_REPORT ------------
Event: time 1630984985.622494, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1630984985.622494, type 1 (EV_KEY), code 257 (BTN_1), value 1
Event: time 1630984985.622494, -------------- SYN_REPORT ------------
Event: time 1630984985.740493, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1630984985.740493, type 1 (EV_KEY), code 257 (BTN_1), value 0
Event: time 1630984985.740493, -------------- SYN_REPORT ------------
Event: time 1630984986.168498, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1630984986.168498, type 1 (EV_KEY), code 258 (BTN_2), value 1
Event: time 1630984986.168498, -------------- SYN_REPORT ------------
Event: time 1630984986.296499, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1630984986.296499, type 1 (EV_KEY), code 258 (BTN_2), value 0
[...]
```

</details>

Which makes me think that this is not a kernel driver problem, but something lurking somewhere up the stack.

I also noticed some weird behavior outside of Krita: 

* GNOME on Wayland detects the tablet, but the "Map Buttons..." option does not detect the tablet's buttons (in fact, it seems to hang and refuse all input until you unplug the tablet) -- will test on Xorg later
* KDE Plasma seems to treat the tablet as a gamepad. kcm_wacomtablet completely ignores it, even if xf86-input-wacom is installed

By the way, the title for this issue probably should be updated since the tablet is detected by the kernel, it's just kind of broken.

aero31aero, 2021-09-07:
Updated the issue title, but anyone would have to look at the content here anyway.

BTW, thanks to your info I am now inclined to test my tablet with evtest as well. Maybe there's some way to remap the tablet's buttons to a virtual wacom tablet input.

japareaggae, 2021-09-07:
I got the tablet options to work on KDE (Xorg) now! What you need is the [xorg.conf file](https://github.com/DIGImend/digimend-kernel-drivers/blob/master/xorg.conf) from this repository, so that xf86-input-wacom can detect it. After you do that, make sure you have kcm_wacomtablet installed, and you should be able to configure the tablet from System Settings > Input Devices > Graphic Tablet. xsetwacom also works fine after doing this, so this should probably work for all Xorg environments. (You can just drop that xorg.conf file inside the /etc/X11/xorg.conf.d directory.)

I haven't found a good solution yet for Wayland, so for now I'm using [key-mapper](https://github.com/sezanzeb/key-mapper) to map the tablet buttons to keyboard shortcuts. It should work fine on any environment since it's an evdev-level remapper.

squeaktoy, 2023-10-08:
> I got the tablet options to work on KDE (Xorg) now! What you need is the [xorg.conf file](https://github.com/DIGImend/digimend-kernel-drivers/blob/master/xorg.conf) from this repository, so that xf86-input-wacom can detect it. After you do that, make sure you have kcm_wacomtablet installed, and you should be able to configure the tablet from System Settings > Input Devices > Graphic Tablet. xsetwacom also works fine after doing this, so this should probably work for all Xorg environments. (You can just drop that xorg.conf file inside the /etc/X11/xorg.conf.d directory.)
> 
> I haven't found a good solution yet for Wayland/Sway, so for now I'm using [key-mapper](https://github.com/sezanzeb/key-mapper) to map the tablet buttons to keyboard shortcuts. It should work fine on any environment since it's an evdev-level remapper.

Unfortunately this doesn't work anymore on Linux 6.1

japareaggae, 2023-10-08:
I've just tested this setup on a Debian live image running kernel version 6.1 and it seems to be working fine, maybe there's something wrong with your configuration or with your tablet?