DIGImend/digimend-kernel-drivers #552: How can i Map the drawing surface to a specific screen area ? (Huion Inspiroy H950P x Linux Mint 20.1 Ulyssa)

Wilfried-1, 2021-07-31:
Hi,

I have a 49 inch monitor (3840x1080 - 32:9 ratio) and my huion tablet's drawing area is matching the whole screen area.

What i would like to do, is to have the drawing surface only matching the left half of my screen so that the ratio will not be distorted and the input will not be stretched out.

When i draw, i usually have Krita on the left half of my screen, and a web browser on my right half.

Good to know : 3840 / 2 = 1920

So when i have krita on the left half, i automatically have a window of 1920x1080 with a ratio of 16:9 (which perfectly fits the tablet's drawing surface's ratio)

I'll link a ScreenShot of what menus I have after installingthe drivers on linux (i can't define the size of the screen area i want to be matching with the drawing surface).

[image]

Wilfried-1, 2021-07-31:
Ok so i found the solution :

1. Go into terminal, then ask it your current settings : 
`xsetwacom list devices`
HUION Huion Tablet_H950P Pad pad	id: 23	type: PAD       
HUION Huion Tablet_H950P Touch Strip pad	id: 26	type: PAD       
HUION Huion Tablet_H950P Pen stylus	id: 27	type: STYLUS 

2. Find the area currently set for your tablet (i tried everyone of them until i found a numerical return
`xsetwacom get  23 Area`
Property 'Wacom Tablet Area' does not exist on device.
`xsetwacom get  26 Area`
Property 'Wacom Tablet Area' does not exist on device.
`xsetwacom get  27 Area`
0 0 44200 27600

3. Double the X axis (it's counter intuitive, but basically you need to tell Linux that your tablet is 2x larger than it really is, so that way Linux will attribute your full screen on 2x the size of your tablet, like your current tablet = Left half of your screen, and the second one, that doesn't exist, = the Right half of your screen) -> The numerical return is made that way : x1 y1 x2 y2 and we'll just double the x2 (which correspond to the end of the drawing surface of the tablet)
`xsetwacom --set 27 area 0 0 88400 27600`

Then you're done :)

PS : Open your drawing software before doing this, and draw one circle, then after you entered the new settings, draw another circle on the side of the first circle

Maybe train yoruself for 1 or 2 min so that your hand is more used to this new ratio, and then re-draw a last circle on the other side of the first one that you made with the very first settings you had so you can compare the difference :)

Have nice drawings !

closed 2021-07-31 by Wilfried-1

Kangaroux, 2021-09-12:
@Wilfried-1 Works great, thanks!

Wilfried-1, 2021-09-13:
> @Wilfried-1 Works great, thanks!

I'm glad it could help someone ! Feel free to send me what you painted !