DIGImend/digimend-kernel-drivers #554: Support for XP-Pen Artist 12 (not pro) ?

zenny, 2021-08-17:
Hi,

I could not figure out any way that I can use this driver instead of the proprietary driver from xp-pen (which too is crashing without any debug possibilities).

I tried to do the following:

1. Appended `hid-ids.h` with:
```#define USB_DEVICE_ID_UGEE_XPPEN_TABLET_A12	0x080a`
Ref: `Bus 003 Device 002: ID 28bd:080a UGTABLET 11.6 inch PenDisplay```

2. Appended `hid-uclogic-params.c` with:
```
case VID_PID(USB_VENDOR_ID_UGEE,
			USB_DEVICE_ID_UGEE_XPPEN_TABLET_A12):
```

3. Appended `hid-uclogic-core.c` with:
```
{ HID_USB_DEVICE(USB_VENDOR_ID_UGEE,
				USB_DEVICE_ID_UGEE_XPPEN_TABLET_A12) },
```

4. Appended the USBID to `xorg.conf`:
```
Section "InputClass"
        Identifier "Ugee/XP-Pen tablets with Wacom driver"
        MatchUSBID "28bd:007[1458]|28bd:0094|28bd:0042|**28bd:080a**|5543:004[57]|5543:0081|5543:0004|5543:3031"
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
EndSection
```

5. Ran `sudo make dkms_install`:
```
$ sudo make dkms_install
dkms add .

Creating symlink /var/lib/dkms/digimend/11/source ->
                 /usr/src/digimend-11

DKMS: add completed.
dkms build digimend/11

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.4.138_1 KVERSION=5.4.138_1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.138_1 (x86_64)
Consult /var/lib/dkms/digimend/11/build/make.log for more information.
make: *** [Makefile:101: dkms_modules_install] Error 10
```

6. The `/var/lib/dkms/digimend/11/build/make.log` shows:
```
$ cat /var/lib/dkms/digimend/11/build/make.log
DKMS make.log for digimend-11 for kernel 5.4.138_1 (x86_64)
Tue Aug 17 11:12:24 AM CEST 2021
make[1]: Entering directory '/var/lib/dkms/digimend/11/build'
make -C /lib/modules/5.4.138_1/build M=/var/lib/dkms/digimend/11/build modules
make[2]: Entering directory '/usr/src/kernel-headers-5.4.138_1'
  CC [M]  /var/lib/dkms/digimend/11/build/hid-kye.o
  CC [M]  /var/lib/dkms/digimend/11/build/hid-uclogic-core.o
  CC [M]  /var/lib/dkms/digimend/11/build/hid-uclogic-rdesc.o
  CC [M]  /var/lib/dkms/digimend/11/build/hid-uclogic-params.o
In file included from /var/lib/dkms/digimend/11/build/hid-kye.c:21:
/var/lib/dkms/digimend/11/build/hid-ids.h:62:11: error: expected '=', ',', ';', 'asm' or '__attribute__' before '--' token
   62 | xsetwacom --set "$tabletstylus" Area 100 120 32794 32797
      |           ^~
In file included from /var/lib/dkms/digimend/11/build/hid-uclogic-params.c:19:
[... 218 lines cut ...]
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:262: /var/lib/dkms/digimend/11/build/hid-uclogic-core.o] Error 1
make[3]: *** [scripts/Makefile.build:262: /var/lib/dkms/digimend/11/build/hid-uclogic-params.o] Error 1
make[2]: *** [Makefile:1734: /var/lib/dkms/digimend/11/build] Error 2
make[2]: Leaving directory '/usr/src/kernel-headers-5.4.138_1'
make[1]: *** [Makefile:25: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/digimend/11/build'
```

7. Any input to resolve this issue? Some outpus about the xp-pen artist 12 tablet are follows:

``` 
$ xinput list | grep UG
⎜   ↳ UGTABLET 11.6 inch PenDisplay Mouse       id=8    [slave  pointer  (2)]
⎜   ↳ UGTABLET 11.6 inch PenDisplay             id=9    [slave  pointer  (2)]
    ↳ UGTABLET 11.6 inch PenDisplay Keyboard    id=10   [slave  keyboard (3)]
```
```
$ lsusb | grep UG
Bus 003 Device 002: ID 28bd:080a UGTABLET 11.6 inch PenDisplay
```

The tablet pointer gets detected with default kernel drivers that comes with v5.4.138 in tty console, but does not respond inside X! Hopefully this driver may work. Thanks in advance.

/z

nic3-14159, 2021-08-17:
Hello! Does the driver compile at all before you make any changes? Also, could you also post the actual code you wrote somewhere? I can't quite tell what changes you made.

I did notice the line
` 62 | xsetwacom --set "$tabletstylus" Area 100 120 32794 32797`
in the make log, which is unexpected. It seems like somehow there is an xsetwacom command in hid-ids.h? That shouldn't be there if there is.

zenny, 2021-08-17:
> Hello! Does the driver compile at all before you make any changes?

Yes as of below:

```
$ make
make -C /lib/modules/5.4.138_1/build M=/xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers modules
make[1]: Entering directory '/usr/src/kernel-headers-5.4.138_1'
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-kye.o
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-uclogic-core.o
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-uclogic-rdesc.o
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-uclogic-params.o
  LD [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-uclogic.o
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-polostar.o
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-viewsonic.o
  Building modules, stage 2.
  MODPOST 4 modules
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-kye.mod.o
  LD [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-kye.ko
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-polostar.mod.o
  LD [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-polostar.ko
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-uclogic.mod.o
  LD [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-uclogic.ko
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-viewsonic.mod.o
  LD [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-viewsonic.ko
make[1]: Leaving directory '/usr/src/kernel-headers-5.4.138_1'
```

> Also, could you also post the actual code you wrote somewhere? I can't quite tell what changes you made.

I appended (added lines):

1. Appended line 60 in `hid-ids.h` with:
```#define USB_DEVICE_ID_UGEE_XPPEN_TABLET_A12	0x080a```

Ref: `Bus 003 Device 002: ID 28bd:080a UGTABLET 11.6 inch PenDisplay`

2. Appended line 1195-1196 in `hid-uclogic-params.c` with:
```
case VID_PID(USB_VENDOR_ID_UGEE,
			USB_DEVICE_ID_UGEE_XPPEN_TABLET_A12):
```

3. Appended 537-538 in `hid-uclogic-core.c` with:
```
{ HID_USB_DEVICE(USB_VENDOR_ID_UGEE,
				USB_DEVICE_ID_UGEE_XPPEN_TABLET_A12) },
```

4. Appended the USBID `28bd:080a` of the tablet to line 26 of  `xorg.conf` to look like:
```
MatchUSBID "28bd:007[1458]|28bd:0094|28bd:0042|28bd:080a|5543:004[57]|5543:0081|5543:0004|5543:3031"
```

> 
> I did notice the line
> ` 62 | xsetwacom --set "$tabletstylus" Area 100 120 32794 32797`
> in the make log, which is unexpected. It seems like somehow there is an xsetwacom command in hid-ids.h? That shouldn't be there if there is.

I could find no trace of the ` 62 | xsetwacom --set "$tabletstylus" Area 100 120 32794 32797` in `hid-ids.h`

@nic3-14159 Thanks for the pointer.

However, compiling again went through, but yet to see whether the stylus works inside X or not. I shall get back after I reboot.

zenny, 2021-08-17:
> Hello! Does the driver compile at all before you make any changes?

Yes as of below:

```
$ make
make -C /lib/modules/5.4.138_1/build M=/xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers modules
make[1]: Entering directory '/usr/src/kernel-headers-5.4.138_1'
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-kye.o
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-uclogic-core.o
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-uclogic-rdesc.o
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-uclogic-params.o
  LD [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-uclogic.o
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-polostar.o
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-viewsonic.o
  Building modules, stage 2.
  MODPOST 4 modules
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-kye.mod.o
  LD [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-kye.ko
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-polostar.mod.o
  LD [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-polostar.ko
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-uclogic.mod.o
  LD [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-uclogic.ko
  CC [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-viewsonic.mod.o
  LD [M]  /xtbmr/HOMEPOOL/HOME/VOIDHOME.BAK/zenny/Downloads/gitrepos/digimend-kernel-drivers/hid-viewsonic.ko
make[1]: Leaving directory '/usr/src/kernel-headers-5.4.138_1'
```

> Also, could you also post the actual code you wrote somewhere? I can't quite tell what changes you made.

I appended (added lines):

1. Appended line 60 in `hid-ids.h` with:
```#define USB_DEVICE_ID_UGEE_XPPEN_TABLET_A12	0x080a```

Ref: `Bus 003 Device 002: ID 28bd:080a UGTABLET 11.6 inch PenDisplay`

2. Appended line 1195-1196 in `hid-uclogic-params.c` with:
```
case VID_PID(USB_VENDOR_ID_UGEE,
			USB_DEVICE_ID_UGEE_XPPEN_TABLET_A12):
```

3. Appended 537-538 in `hid-uclogic-core.c` with:
```
{ HID_USB_DEVICE(USB_VENDOR_ID_UGEE,
				USB_DEVICE_ID_UGEE_XPPEN_TABLET_A12) },
```

4. Appended the USBID `28bd:080a` of the tablet to line 26 of  `xorg.conf` to look like:
```
MatchUSBID "28bd:007[1458]|28bd:0094|28bd:0042|28bd:080a|5543:004[57]|5543:0081|5543:0004|5543:3031"
```

> 
> I did notice the line
> ` 62 | xsetwacom --set "$tabletstylus" Area 100 120 32794 32797`
> in the make log, which is unexpected. It seems like somehow there is an xsetwacom command in hid-ids.h? That shouldn't be there if there is.

I could find no trace of the ` 62 | xsetwacom --set "$tabletstylus" Area 100 120 32794 32797` in `hid-ids.h`

@nic3-14159 Thanks for the pointer.

However, compiling again went through, but yet to see whether the stylus works inside X or not.

After a reboot, `xinput list` fails to detect the tablet completly:

```$ lsusb | grep UG
Bus 001 Device 003: ID 28bd:080a UGTABLET 11.6 inch PenDisplay

[... 122 lines cut ...]
   832x624       74.55
   800x600       72.19    75.00    60.32    56.25
   720x576       50.00
   720x480       60.00    59.94
   640x480       75.00    72.81    66.67    60.00    59.94
   720x400       70.08
DVI-0 connected 1920x1080+0+0 (normal left inverted right x axis y axis) 256mm x 144mm
   1920x1080     60.00*+  62.81    59.94
   1920x1080i   125.62
   1680x1050     61.00
   1280x1024     60.02
   1440x900      59.90
   1360x768      59.95
   1280x720      60.00    59.94
   1024x768      60.00
   1152x648      60.01
   800x600       60.32
   720x480       60.00    59.94
   640x480       60.00    59.94
VGA-0 disconnected (normal left inverted right x axis y axis)
```

The main screen is connected to HDMI-0 and the tablet to the DVI-0 port of the same card, mirrored display mode (tried with top of each other by running `xrandr --output HDMI-0 --primary --mode 1920x1080 --pos 0x0 --rotate normal --output DVI-0 --mode 1920x1080 --pos 0x1080 --rotate normal --output VGA-0 --off`, but no go).

Any inputs appreciated! Cheers,

nic3-14159, 2021-08-17:
Based on where you added the case statement in hid-uclogic-params.c, xinput likely does not recognize your tablet due to issue #550. (Not that the place you added the case statement is wrong, there just happens to be an issue that only seems to affect tablet there) I have done some debugging with the person who filed that issue, and there is now a potential fix for that (outlined in #553).

Essentially, you just need to add the line
`hdev->quirks |= HID_QUIRK_HIDINPUT_FORCE;`
at line 206 in hid-uclogic-core.c

See if that makes xinput recognize your tablet.

Also, I wouldn't be surprised if the tablet buttons don't work properly. It seems like a lot of newer XP-Pen tablets require a new initialization procedure to enable the buttons properly (otherwise they send keyboard events and not tablet button events) There is a pull request (#291) for that procedure, but it hasn't been merged yet and may need to be adapted to work with different models.

zenny, 2021-08-17:
> Based on where you added the case statement in hid-uclogic-params.c, xinput likely does not recognize your tablet due to issue #550. (Not that the place you added the case statement is necessarily wrong, there just happens to be an issue that only seems to affect tablets there) I have done some debugging with the person who filed that issue, and there is now a potential fix for that (outlined in #553).
> 
> Essentially, you just need to add
> `hdev->quirks |= HID_QUIRK_HIDINPUT_FORCE;`
> at line 206 in hid-uclogic-core.c

Thanks I appended the line 206 to `hdi-uclogic-core.c`, and recompiled and installed.
> 
> See if that makes xinput recognize your tablet.

Still not detected by xinput, fyi.

> Also, I wouldn't be surprised if the tablet buttons don't work properly. It seems like a lot of newer XP-Pen tablets require a new initialization procedure to enable the buttons properly (otherwise [...]

I guess XP-Pen Artist 12 could be their earlier releases?!

nic3-14159, 2021-08-17:
>Still not detected by xinput, fyi.

Just checking, did you either reboot or run `sudo modprobe -r hid-uclogic` after installing the driver with the fix I suggested? If you didn't, the kernel could still be running the old driver without the fix.

Also, I took a look at the logs you posted in that other issue, and it does appear that the frame buttons emit keyboard events. That, along with some details in the USB device descriptor, tells me that the Artist 12 is indeed one of these newer tablets that need the new init procedure for the frame buttons to work properly.

zenny, 2021-08-17:
> > Still not detected by xinput, fyi.
> 
> Just checking, did you either reboot or run `sudo modprobe -r hid-uclogic` after installing the driver with the fix I suggested? If not, the kernel could still be running the old driver without the fix.

Rebooted and no go either.

> Also, I took a look at the logs you posted in that other issue, and it does appear that the frame buttons emit keyboard events. That, along with some details in the USB device descriptor, tells me [...]

Is there any way to overcome this?

I see. Then I have a wrong guess. In my case even the pen stylus pointers are not detected inside X (but ok in tty console) even after appending (forcing) as of below inside the xorg.conf:

```
$ cat /usr/share/X11/xorg.conf.d/60-xppen.conf
Section "InputClass"
        Identifier "XP-Pen Artist 12 Keyboard"
        MatchUSBID "28bd:080a"
        MatchIsKeyboard "on"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
EndSection

Section "InputClass"
        Identifier "XP-Pen Artist 12"
        MatchIsTablet "on"
        Driver "wacom"
        MatchUSBID "28bd:080a"
        MatchDevicePath "/dev/input/event*"
EndSection

Section "InputClass"
        Identifier "XP-Pen Artist 12 Mouse"
        MatchUSBID "28bd:080a"
        MatchIsPointer "on"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
EndSection
```

Anyway, so nice of you for prompt inputs.

zenny, 2021-08-17:
Meanwhile I just saw the udev.rules file with some strange values. Does this need to be changed, too? If so, to what?

https://github.com/DIGImend/digimend-kernel-drivers/pull/291#issuecomment-529033736 points to `hwdb.d` Is that something to do with that makes the stylus not work inside X?

kurikaesu, 2021-08-23:
Could you do a `sudo libinput record` and then look for something like this?:

```
  udev:
    properties:
    - ID_INPUT=1
    - ID_INPUT_HEIGHT_MM=129
    - ID_INPUT_KEY=0
    - ID_INPUT_TABLET=1
    - ID_INPUT_TABLET_PAD=1
    - ID_INPUT_WIDTH_MM=230
    - LIBINPUT_DEVICE_GROUP=3/28bd/909:usb-0000:0a:00.3-2
```

What I'm interested in is if udev is listing the device as a mouse.

Secondly, if you plug in the tablet and do a `dmesg` you should be able to see if the driver actually detects the device. The uclogic lines below are digimend driver outputs.:
```
[89715.950719] usb 3-2.3: new full-speed USB device number 18 using xhci_hcd
[89716.063518] usb 3-2.3: New USB device found, idVendor=28bd, idProduct=0909, bcdDevice= 0.00
[89716.063524] usb 3-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[89716.063526] usb 3-2.3: Product: 9 inch PenTablet
[89716.063529] usb 3-2.3: SerialNumber: 0000000000
[89716.219139] uclogic 0003:28BD:0909.0026: interface is invalid, ignoring
[89716.219730] uclogic 0003:28BD:0909.0027: interface is invalid, ignoring
[89716.227163] input: 9 inch PenTablet Pen as /devices/pci0000:00/0000:00:07.1/0000:0a:00.3/usb3/3-2/3-2.3/3-2.3:1.2/0003:28BD:0909.0028/input/input58
[89716.286719] input: 9 inch PenTablet Pad as /devices/pci0000:00/0000:00:07.1/0000:0a:00.3/usb3/3-2/3-2.3/3-2.3:1.2/0003:28BD:0909.0028/input/input59
[89716.286839] uclogic 0003:28BD:0909.0028: input,hidraw0: USB HID v1.00 Keypad [9 inch PenTablet] on usb-0000:0a:00.3-2.3/input2
[89716.312365] uclogic 0003:28BD:0909.0026: interface is invalid, ignoring
[89716.314353] uclogic 0003:28BD:0909.0027: interface is invalid, ignoring
```

zenny, 2021-08-28:
> Could you do a `sudo libinput record` and then look for something like this?:
>
> ```
>   udev:
>     properties:
>     - ID_INPUT=1
>     - ID_INPUT_HEIGHT_MM=129
>     - ID_INPUT_KEY=0
>     - [...]

I have posted the outcomes of `libinput record` in http://ix.io/3xbL. The further outputs are pressing of the buttons on the tablet.

> Secondly, if you plug in the tablet and do a `dmesg` you should be able to see if the driver actually detects the device. The uclogic lines below are digimend driver outputs.:
>
> ```
> [89715.950719] [...]

Meanwhile, I upgraded the linux kernel to v.5.4.142, the libwacom driver detects (without digimend compilation) the tablet and the stylus without digimend, but the the seven buttons on the tablet itself does not respond to, though `libinput record` detects the events when buttons pressed as seen in http://ix.io/3xbL.