DIGImend/digimend-kernel-drivers #555: Gaomon m6: for only 1/3 can be recognized

kamayi2008, 2021-08-17:
I am using Gaomon m6, the pressure works fine, I haven't try setting the buttons. The annoying problem is that only about 1/3 of the tablelet can be recognized on linux, and it happens on windows everytime after re-login. I tryed 'xsetwacom set  MapToOutput' and 'xesetwacom set Area', failed.
So anyone can help this? Thanks

nic3-14159, 2021-08-20:
Hello! Could you try running `sudo usbhid-dump -es -m VID:PID`, where VID:PID is the id for the tablet found in the output of `lsusb`? See if the tablet produces reports while moving the pen over the area that isn't being recognized.

nukawa, 2021-08-23:
Sir, sorry to disturb you, I also have Gaomon m6 recently, is your m6 workable?? I mean even pen only excluding buttons...etc.
So far, my system (Ubuntu20.04) could detect this, but it is not usable totally. Could you give me any advice? Thanks a lot.
 I just hope it could be available like normal mouse at least now.

nukawa, 2021-08-28:
Dear sir, 
I have followed to collect the diagnosis of my Gaomon M6 for your reference.

VID: "256c"
PID: "0064"
vendor: Gaomon
product: M6
Product page: [https://www.gaomon.cn/Pen_Tablet/M6.html](url)

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7070694/descriptors.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7070695/hid_report_descriptors.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7070696/pen_buttons.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7070697/pen_pressure.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7070698/pen_tilt.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7070699/pen_coords.txt)

Unfortunately, there is no records of "tee frame_dials" & "tee frame_buttons", so these two files are unavailable. Btw, this did not bundle any mouse.
If you have any questions or comments, please let me know, thanks for your help.

nukawa, 2021-08-28:
Dear LazyCatW, 

I have tried your method to do, but it seems still not workable. I might not have enough knowledge to solve this due to brand-new Linux user.

Thanks for your help and comments.

Littlesharkcug, 2021-12-26:
> 

Did it has beed solved？

nukawa, 2021-12-27:
Hi, @Littlesharkcug,
May I know what happened to you? Is it usable partially? For me, it is totally unworkable so far in Ubuntu 20.
I still try finding something available to make it usable, even partially.

Littlesharkcug, 2021-12-28:
> Hi, @Littlesharkcug, May I know what happened to you? Is it usable partially? For me, it is totally unworkable so far in Ubuntu 20. I still try finding something available to make it usable, even partially.

I want to use my M6 in Manjaro.It is completely unavailable  till now

nukawa, 2021-12-28:
Oh, so sad....
It seems to be unavailable for these OS....
Now I am interesting in kamayi2008's situations, M6 seems workable partially.

JoseExposito, 2024-05-03:
Hi!

Your issue should be fixed by [DIGImend v13](https://github.com/DIGImend/digimend-kernel-drivers/releases/tag/v13). Please, if you have a chance, give it a try and report any problems found.

Thanks!

closed 2024-05-03 by JoseExposito