DIGImend/digimend-kernel-drivers #556: XP-Pen Artist 12 Diagnostics

zenny, 2021-08-17:
Hi,

Please find below the diagnostics of  XP-Pen Artist 12 to make things easier to make work with DIGImend driver as described in #554. The diagnostics follow suit to the details in https://digimend.github.io/support/howto/trbl/diagnostics/.

1Product page with proprietary linux driver: https://www.xp-pen.com/download-68.html

```
$ lsusb
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 009 Device 003: ID 045e:00b0 Microsoft Corp. Digital Media Pro Keyboard
Bus 009 Device 002: ID 045e:0083 Microsoft Corp. Basic Optical Mouse
Bus 009 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 005: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 003 Device 004: ID 413c:301a Dell Computer Corp. Dell MS116 USB Optical Mouse
Bus 003 Device 003: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 002: ID 1a2c:405e China Resource Semico Co., Ltd USB Keyboard
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 28bd:080a UGTABLET 11.6 inch PenDisplay
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

The tablet is `Bus 001 Device 003: ID 28bd:080a UGTABLET 11.6 inch PenDisplay`

2. Descriptors.txt (http://ix.io/3whe)
```
$ T=28bd:080a
$ sudo lsusb -v -d $T > descriptors.txt
Password:
can't get debug descriptor: Resource temporarily unavailable
$ cat descriptors.txt

Bus 001 Device 003: ID 28bd:080a UGTABLET 11.6 inch PenDisplay
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x28bd
  idProduct          0x080a
  bcdDevice            0.00
  iManufacturer           1 UGTABLET
  iProduct                2 11.6 inch PenDisplay
  iSerial                 3
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x005b
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower               70mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
[... 167 lines cut ...]
001:003:001:STREAM             1629219386.149375
 07 98 97 03 E7 35 00 00

001:003:001:STREAM             1629219389.159478
 07 90 A0 28 92 1E 00 00

001:003:001:STREAM             1629219389.164102
 07 90 A0 28 92 1E 00 00
...
```

7. Pen tilt angles (http://ix.io/3who)
`$ sudo usbhid-dump -es -m $T | tee pen_tilt.txt`

8. Pen pressure (http://ix.io/3whq)
`$ sudo usbhid-dump -es -m $T | tee pen_pressure.txt`

9. Pen buttons (http://ix.io/3whr)
`$ sudo usbhid-dump -es -m $T | tee pen_buttons.txt`

10. No Frame dials 
`$ sudo usbhid-dump -es -m $T | tee frame_dials.txt`

11. Frame buttons (http://ix.io/3whs)
`$ sudo usbhid-dump -es -m $T | tee frame_buttons.txt`

kurikaesu, 2021-08-22:
Could you look at my comment here https://github.com/DIGImend/digimend-kernel-drivers/issues/535#issuecomment-903189712 and do the same for your Artist 12? Reply in this issue of course :smile:

zenny, 2021-08-28:
> Could you look at my comment here [#535 (comment)](https://github.com/DIGImend/digimend-kernel-drivers/issues/535#issuecomment-903189712) and do the same for your Artist 12? Reply in this issue of course smile

@kurikaesu Thanks for pointing to a possible solution. However, with an upgrade to the latest kernel (v5.4.142) appears to have the libwacom support to the tablet by default, but it fails to detect physical buttons on the tablet itself, fyi, though `libinput record` records the events.

Most surprisingly, `xinput list` does not detect the "keyboard" part of the tablet (most reasonably the physical buttons on the pad), 

```
$ xinput list | grep UG
⎜   ↳ UGTABLET 11.6 inch PenDisplay Mouse touch id=8    [slave  pointer  (2)]
⎜   ↳ UGTABLET 11.6 inch PenDisplay stylus      id=9    [slave  pointer  (2)]
⎜   ↳ UGTABLET 11.6 inch PenDisplay pad         id=10   [slave  pointer  (2)]
⎜   ↳ UGTABLET 11.6 inch PenDisplay eraser      id=16   [slave  pointer  (2)]
```

But, `libinput record` detects the keyboard component of the pad and the key presses, too (details in http://ix.io/3xbL):

```
$ libinput record
Available devices:
/dev/input/event2:      UGTABLET 11.6 inch PenDisplay Mouse
/dev/input/event3:      UGTABLET 11.6 inch PenDisplay
/dev/input/event4:      UGTABLET 11.6 inch PenDisplay
/dev/input/event5:      UGTABLET 11.6 inch PenDisplay Keyboard
```

Just confused!

kurikaesu, 2021-08-28:
This is what I'm trying to help you fix by getting you to do the key handshake and dump of probe information.
The libwacom support will only work for the stylus/digitizer because the pad buttons are outputting keyboard events. If you want the pads buttons to work with libwacom then a driver like digimend needs to re-interpret the keyboard to pad buttons.

I do not know if the Artist 12 requires the activation key being sent to the correct interface and this can only really be confirmed by installing the proprietary linux drivers, changing the pad-keys to a custom binding and then capturing the USB packets using something like wireshark. From the captures we can inspect if the proprietary drivers are sending a `02b00400000000000000` or equivalent to the device and forcing it into a "uniform interface mode".

This is why I suggested following what I had written in #535 to help debug the issue, or if you want to check if you actually need the activation key then doing the wireshark dump is the right avenue.

Either way, once we know your tablet requires an activation key or not will help determine how to move forward.

zenny, 2021-08-28:
> This is what I'm trying to help you fix by getting you to do the key handshake and dump of probe information.

So nice of you!

> The libwacom support will only work for the stylus/digitizer because the pad buttons are outputting keyboard events. If you want the pads buttons to work with libwacom then a driver like digimend needs to re-interpret the keyboard to pad buttons.

Do you mean that I still need to compile digimend to make the pad buttons to work?

> I do not know if the Artist 12 requires the activation key being sent to the correct interface and this can only really be confirmed by installing the proprietary linux drivers, changing the [...]

As your advice, I created:

```$ cat keyboard-debug.py
#! /usr/bin/env python3
# https://github.com/DIGImend/digimend-kernel-drivers/issues/535#issuecomment-903189712

import usb.core
import usb.util

dev = usb.core.find(idVendor=0x28bd, idProduct=0x080a)

if dev.is_kernel_driver_active(2):
    dev.detach_kernel_driver(2)

cfg = dev.get_active_configuration()
intf = cfg[(2,0)]
ei = intf[0]
eo = intf[1]

eo.write(bytes.fromhex('02b00400000000000000'))

while(True):
    print(''.join(format(x, '02x') for x in ei.read(12)))
```

But the process died immediately without me pressing any buttons, fyi. 

```
]$ sudo python ./keyboard-debug.py
02b10400000000000000
Traceback (most recent call last):
  File "/home/zenny/Downloads/xp-pen-diagnostics/./keyboard-debug.py", line 20, in <module>
    print(''.join(format(x, '02x') for x in ei.read(12)))
  File "/usr/lib/python3.9/site-packages/usb/core.py", line 423, in read
    return self.device.read(self, size_or_buffer, timeout)
  File "/usr/lib/python3.9/site-packages/usb/core.py", line 1029, in read
    ret = fn(
  File "/usr/lib/python3.9/site-packages/usb/backend/libusb1.py", line 864, in intr_read
    return self.__read(self.lib.libusb_interrupt_transfer,
  File "/usr/lib/python3.9/site-packages/usb/backend/libusb1.py", line 954, in __read
    _check(retval)
  File "/usr/lib/python3.9/site-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 110] Operation timed out
```

> or if you want to check if you actually need the activation key then doing the wireshark dump is the right avenue.
> 
> Either way, once we know your tablet requires an activation key or not will help determine how to move forward.

kurikaesu, 2021-08-28:
Alright, from the response of the keyboard-debug.py it looks like your Artist 12 responds the same was as my Artist 13.3 Pro (and other Pro tablets).

The reponse of `02b10400000000000000` indicates that the tablet does have an unlock key and the unlock key we used `02b00400000000000000` was correct.

All of that USBTimeout error is fine, it is expected.

What you will need to do is re-run the keyboard-debug.py and get it to spit out `02b10400000000000000` again and allow it to exit/fail out. Once it has failed out, then run the steps after the heading `Collect raw input samples`: http://digimend.github.io/support/howto/trbl/diagnostics/

Having used the unlock key prior, the streams should start producing events on the 02 interface like:
`02 f0 00 00 04 00 00 00 00 13`.

We will need to get data for all of the tablet. You can see what information is most important for me here under the headings where it says `USB Packets - interface 2` (There is one section for the pen/digitizer and another for the frame buttons/express keys): https://github.com/kurikaesu/DIGImend-research/blob/main/xp-pen-13.3-and-12-pro.md

Once you have that information in hand and can add it as a comment here, I should be able to write a digimend driver very quickly to support your tablet.

zenny, 2021-08-28:
> Alright, from the response of the keyboard-debug.py it looks like your Artist 12 responds the same was as my Artist 13.3 Pro (and other Pro tablets).
>
> The reponse of `02b10400000000000000` indicates [...]

Do you need `usbhid-dump` or something else from `tcpdump`?

> Having used the unlock key prior, the streams should start producing events on the 02 interface like:
> `02 f0 00 00 04 00 00 00 00 13`.
>
> We will need to get data for all of the tablet. You can see [...]

zenny, 2021-08-28:
@kurikaesu As you have asked for raw data outpute after the reponse of 02b10400000000000000 as follows:
frame_wheel_scrolling.txt http://ix.io/3xfE
pen_coords.txt http://ix.io/3xfG
pen_tilt.txt http://ix.io/3xfH
pen_pressure.txt http://ix.io/3xfI
pen_buttons.txt http://ix.io/3xfJ
frame_dials.txt http://ix.io/3xfK (only scroll strip, has no 'frame dial')
frame_buttons.txt http://ix.io/3xfL (also scrolled on the scroll strip in the middle)

kurikaesu, 2021-08-29:
WOW, the raw data you are recording is very unique.
It appears that your tablet is producing two outputs for each action. Let's take the frame buttons for example.
It is sending the frame button message on the `02` interface as well as the `03` interface.

`02` is a 10 byte interface and from the looks of the messages, apart from the touch/scroll strip has an identical structure as my 22R pro. I could probably quickly write something to support your tablet apart from the touch strip as the touch strip appears to only provide two unique messages: `01 2D` and `01 2E`. I'd have to know which of these messages correspond to a swipe up or down and if it doesn't matter where the swipes start from.

nic3-14159, 2021-08-29:
@kurikaesu The tablet sending reports on both interfaces is indicative of the wrong key packet being sent. I have found that tablets send the 02b10... response if the key packet was at least somewhat correct.

@zenny Could you try the script from the link below and paste the results? Make sure you hold the pen to the tablet surface while running the program. If you see a line that starts with "Key is 0x" could you also collect diagnostics afterwards and post the results?

https://github.com/nic3-14159/uclogic-new-init/blob/base/uclogic_new_init.py

zenny, 2021-08-29:
> @kurikaesu The tablet sending reports on both interfaces is indicative of the wrong key packet being sent. I have found that tablets send the 02b10... response even if the key packet is incorrect. It seems like as long as the key starts with 02b0 it sends a response, but the value of the next byte is the actual key.
> 
> @zenny Could you try the script from the link below and paste the results? Make sure you hold the pen to the tablet surface while running the program. If you see a line that starts with "Key is 0x" could you also collect diagnostics afterwards and post the results?
> 
> https://github.com/nic3-14159/uclogic-new-init/blob/base/uclogic_new_init.py

@nic3-14159 Thanks.

Here is the output:

```
$ sudo python ./uclogic_new_init.py 28bd:080a
Please hold the pen on the tablet surface while running this program.
Press Enter to continue
Trying init packet 0x02B00200000000000000:
Received 02B10200000000000000
Key is 0x02B00200000000000000, interface 2, endpoint 0x03
```

zenny, 2021-08-29:
> WOW, the raw data you are recording is very unique.
> It appears that your tablet is producing two outputs for each action. Let's take the frame buttons for example.
> It is sending the frame button [...]

@kurikaesu. Thanks. Please find the debug outputs by scrolling DOWN and UP on the touchstrip on the tablet:

frame_buttons-TOUCHSTRIP-DOWN.txt http://ix.io/3xhE
frame_buttons-TOUCHSTRIP-UP.txt http://ix.io/3xhF

Will that help?

nic3-14159, 2021-08-29:
@zenny libwacom and libinput not detecting the tablet anymore after running my script is expected, it does that with a different XP-Pen tablet I have. Usbhid-dump should still work though.

Could you also collect and post the raw data samples after running my script? Sorry about all the dumps we're asking you to do, I hope you don't mind.

zenny, 2021-08-29:
> @zenny libwacom and libinput not detecting the tablet anymore after running my script is expected, it does that with a different XP-Pen tablet I have. Usbhid-dump should still work though.
> 
> Could you also collect and post the raw data samples after running my script?

The usbhid-dump above in https://github.com/DIGImend/digimend-kernel-drivers/issues/556#issuecomment-907739964 is after running your script.

> Sorry about all the dumps we're asking you to do, I hope you don't mind.

Never mind. Do you suggest me to redo all the diagnostics again after running your script?

zenny, 2021-08-29:
> > @zenny libwacom and libinput not detecting the tablet anymore after running my script is expected, it does that with a different XP-Pen tablet I have. Usbhid-dump should still work though.
> > Could you also collect and post the raw data samples after running my script?
> 
> The usbhid-dump above in [#556 (comment)](https://github.com/DIGImend/digimend-kernel-drivers/issues/556#issuecomment-907739964) is after running your script.
> 
> > Sorry about all the dumps we're asking you to do, I hope you don't mind.
> 
> Never mind. Do you suggest me to redo all the diagnostics again after running your script?

Prior to running your script:

```
$ grep "Using input driver 'wacom'" /var/log/Xorg.0.log
[    76.676] (II) Using input driver 'wacom' for 'UGTABLET 11.6 inch PenDisplay Mouse'
[    76.812] (II) Using input driver 'wacom' for 'UGTABLET 11.6 inch PenDisplay'
[    76.836] (II) Using input driver 'wacom' for 'UGTABLET 11.6 inch PenDisplay'
[    76.860] (II) Using input driver 'wacom' for 'UGTABLET 11.6 inch PenDisplay Keyboard'
[    76.958] (II) Using input driver 'wacom' for 'UGTABLET 11.6 inch PenDisplay eraser'
```

The keyboard is not shown either by xinput:

```
$ xinput list | grep UG
⎜   ↳ UGTABLET 11.6 inch PenDisplay Mouse touch id=8    [slave  pointer  (2)]
⎜   ↳ UGTABLET 11.6 inch PenDisplay stylus      id=9    [slave  pointer  (2)]
⎜   ↳ UGTABLET 11.6 inch PenDisplay pad         id=10   [slave  pointer  (2)]
⎜   ↳ UGTABLET 11.6 inch PenDisplay eraser      id=16   [slave  pointer  (2)]
```

nor by `xsetwacom`:

```
$ xsetwacom list devices
UGTABLET 11.6 inch PenDisplay Mouse touch       id: 8   type: TOUCH
UGTABLET 11.6 inch PenDisplay stylus    id: 9   type: STYLUS
UGTABLET 11.6 inch PenDisplay pad       id: 10  type: PAD
UGTABLET 11.6 inch PenDisplay eraser    id: 16  type: ERASER
```

Then I ran your script from https://github.com/nic3-14159/uclogic-new-init/raw/base/uclogic_new_init.py:

```
$ T=28bd:080a
$ sudo python3 ./uclogic_new_init.py $T
Password:
Please hold the pen on the tablet surface while running this program.
Press Enter to continue
Trying init packet 0x02B00200000000000000:
Received 02B10200000000000000
Key is 0x02B00200000000000000, interface 2, endpoint 0x03
```

Here comes the raw diagnostics capture after running your script with -nic suffix:

frame_wheel_srolling-nic.txt http://ix.io/3xme
pen_coords-nic.txt http://ix.io/3xmh
pen_tilt-nic.txt http://ix.io/3xmj
pen_pressure-nic.txt http://ix.io/3xml
pen_buttons-nic.txt http://ix.io/3xmm
frame_dials-UP-nic.txt http://ix.io/3xmn (scrolling from down to upward in touch strip)
frame_dials-DOWN-nic.txt http://ix.io/3xmo (scrolling from up to downward in touch strip)
frame_buttons-nic.txt http://ix.io/3xmq

Is the above something usable?

nic3-14159, 2021-08-29:
@zenny Yes, those are very useful. Events are no longer being sent on multiple interfaces, and the touch strip reports also appear to be a lot more reasonable. Thanks for posting!

@kurikaesu looks like the touch strip now reports bitmap dial events instead of the keyboard events, so it should be pretty easy to get it working.

kurikaesu, 2021-08-30:
Yeah the new reports look practically identical to the 22R other than that there is no support for tilt. The product site also confirms the absence of tilt: https://www.xp-pen.com/product/64.html

I've added support to the tablet in my PR here: https://github.com/DIGImend/digimend-kernel-drivers/pull/557

Please let me know if it does or doesn't work.

zenny, 2021-08-30:
> Yeah the new reports look practically identical to the 22R other than that there is no support for tilt. The product site also confirms the absence of tilt: https://www.xp-pen.com/product/64.html
> 
> I've added support to the tablet in my PR here: #557
> 
> Please let me know if it does or doesn't work.

@kurikaesu it did not work (even stylus pen failed to work), despite I appended the USBID (28bd:080a) into line 26 `xorg.org` in your source (`git clone -b xppen-artist22r-pro https://github.com/kurikaesu/digimend-kernel-drivers.git`) to look like before compiling:

```
	MatchUSBID "28bd:007[1458]|28bd:0094|28bd:0042|28bd:080a|5543:004[57]|5543:0081|5543:0004|5543:3031"
```

But this time it failed not only the buttons, but even stylus stopped working at all :(

```
$ uname -r
5.4.142_1
$ grep "Using input driver 'wacom'" /var/log/Xorg.0.log
[    49.559] (II) Using input driver 'wacom' for 'UGTABLET 11.6 inch PenDisplay Pen'
[    49.581] (II) Using input driver 'wacom' for 'UGTABLET 11.6 inch PenDisplay Pad'
$ lsmod| grep hid_
hid_uclogic            36864  0
hid_generic            16384  0
usbhid                 53248  1 hid_uclogic
hid                   131072  4 usbhid,hid_uclogic,hid_generic,uhid
usbcore               266240  10 xhci_hcd,ohci_hcd,ehci_pci,usbhid,hid_uclogic,usbkbd,ehci_hcd,usbmouse,xhci_pci,ohci_pci
$ libinput record
Available devices:
/dev/input/event2:      UGTABLET 11.6 inch PenDisplay Pen
/dev/input/event3:      UGTABLET 11.6 inch PenDisplay Pad
Select the device event number: ^C
$ xinput list | grep UG
⎜   ↳ UGTABLET 11.6 inch PenDisplay Pen stylus  id=8    [slave  pointer  (2)]
⎜   ↳ UGTABLET 11.6 inch PenDisplay Pad pad     id=9    [slave  pointer  (2)]
```

kurikaesu, 2021-08-30:
I'll need to figure out what is going on with the stylus but for the pad buttons themselves, could you run libinput record and pick the device and press buttons?

I want to know if the buttons themselves are at least registering events with libinput.

zenny, 2021-08-31:
> I'll need to figure out what is going on with the stylus but for the pad buttons themselves, could you run libinput record and pick the device and press buttons?
> 
> I want to know if the buttons themselves are at least registering events with libinput.

Actually, pressing the buttons on the pad were unresponsive to `libinput record`. However, choosing the `pad` gave the following ouptut:

```
$ libinput record
Available devices:
/dev/input/event2:      UGTABLET 11.6 inch PenDisplay Pen
/dev/input/event3:      UGTABLET 11.6 inch PenDisplay Pad
Select the device event number: 3
Recording to 'stdout'.
# libinput record
version: 1
ndevices: 1
libinput:
  version: "1.18.1"
  git: "unknown"
system:
kernel: "5.4.142_1"
dmi: "dmi:bvnAwardSoftwareInternational,Inc.:bvrF8:bd03/30/2012:svnGigabyteTechnologyCo.,Ltd.:pnGA-970A-D3:pvr:rvnGigabyteTechnologyCo.,Ltd.:rnGA-970A-D3:rvrx.x:cvnGigabyteTechnologyCo.,Ltd.:ct3:cvr:"
devices:
- node: /dev/input/event3
  evdev:
    # Name: UGTABLET 11.6 inch PenDisplay Pad
    # ID: bus 0x3 vendor 0x28bd product 0x80a version 0x100
    # Supported Events:
    # Event type 0 (EV_SYN)
    # Event type 1 (EV_KEY)
    #   Event code 256 (BTN_0)
    #   Event code 257 (BTN_1)
    #   Event code 258 (BTN_2)
    #   Event code 259 (BTN_3)
    #   Event code 260 (BTN_4)
    #   Event code 261 (BTN_5)
    #   Event code 262 (BTN_6)
    #   Event code 263 (BTN_7)
    # Event type 2 (EV_REL)
    #   Event code 8 (REL_WHEEL)
    #   Event code 11 (REL_WHEEL_HI_RES)
    # Event type 4 (EV_MSC)
    #   Event code 4 (MSC_SCAN)
    # Properties:
    name: "UGTABLET 11.6 inch PenDisplay Pad"
    id: [3, 10429, 2058, 256]
    codes:
      0: [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15] # EV_SYN
      1: [256, 257, 258, 259, 260, 261, 262, 263] # EV_KEY
      2: [8, 11] # EV_REL
      4: [4] # EV_MSC
    properties: []
  hid: [
    0x05, 0x0d, 0x09, 0x02, 0xa1, 0x01, 0x85, 0x02, 0x09, 0x20, 0xa0, 0x14, 0x25, 0x01, 0x09, 0x42,
    0x09, 0x44, 0x09, 0x46, 0x75, 0x01, 0x95, 0x03, 0x81, 0x02, 0x95, 0x02, 0x81, 0x01, 0x09, 0x32,
    0x95, 0x01, 0x81, 0x02, 0x95, 0x02, 0x81, 0x01, 0x75, 0x10, 0x95, 0x01, 0xa4, 0x05, 0x01, 0x55,
    0xfd, 0x65, 0x13, 0x34, 0x09, 0x30, 0x27, 0x20, 0x64, 0x00, 0x00, 0x47, 0x6b, 0x27, 0x00, 0x00,
    0x81, 0x02, 0x09, 0x31, 0x27, 0x52, 0x38, 0x00, 0x00, 0x47, 0x2c, 0x16, 0x00, 0x00, 0x81, 0x02,
    0xb4, 0x09, 0x30, 0x27, 0xff, 0x1f, 0x00, 0x00, 0x81, 0x02, 0xa4, 0x54, 0x65, 0x14, 0x35, 0xc3,
    0x45, 0x3c, 0x15, 0xc3, 0x25, 0x3c, 0x75, 0x08, 0x95, 0x02, 0x09, 0x3d, 0x09, 0x3e, 0x81, 0x02,
    0xb4, 0xc0, 0xc0, 0x05, 0x01, 0x09, 0x07, 0xa1, 0x01, 0x85, 0xf7, 0x05, 0x0d, 0x09, 0x39, 0xa0,
    0x14, 0x25, 0x01, 0x75, 0x01, 0x95, 0x08, 0x81, 0x01, 0x05, 0x09, 0x19, 0x01, 0x29, 0x08, 0x95,
    0x08, 0x81, 0x02, 0x95, 0x20, 0x81, 0x01, 0x05, 0x01, 0x09, 0x38, 0x75, 0x08, 0x95, 0x01, 0x15,
    0xff, 0x25, 0x08, 0x81, 0x06, 0x75, 0x01, 0x95, 0x16, 0x81, 0x01, 0xc0, 0xc0
  ]
  udev:
    properties:
    - ID_INPUT=1
    - ID_INPUT.tags=seat0
    - ID_INPUT_KEY=1
    - LIBINPUT_DEVICE_GROUP=3/28bd/80a:usb-0000:02:00.0-1
  quirks:
  events:
  # Current time is 08:32:32
^C%
```