DIGImend/digimend-kernel-drivers #558: Huion H1161 stops working out of the box in Ubuntu 21.04 after installing v10 digimend kernel

LIyuby, 2021-08-21:


edoiks, 2021-11-28:
Hello
I have the same problem:

dmesg:

>[  271.135123] usb 1-13: new full-speed USB device number 7 using xhci_hcd
[  271.285269] usb 1-13: New USB device found, idVendor=256c, idProduct=006d, bcdDevice= 1.00
[  271.285272] usb 1-13: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  271.285273] usb 1-13: Product: Huion Tablet
[  271.285274] usb 1-13: Manufacturer: HUION
[  271.287556] input: HUION Huion Tablet as /devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.0/0003:256C:006D.0009/input/input42
[  271.287789] input: HUION Huion Tablet Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.0/0003:256C:006D.0009/input/input43
[  271.287996] input: HUION Huion Tablet Touch Strip as /devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.0/0003:256C:006D.0009/input/input44
[  271.288165] input: HUION Huion Tablet Dial as /devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.0/0003:256C:006D.0009/input/input45
[  271.288420] uclogic 0003:256C:006D.0009: input,hiddev2,hidraw4: USB HID v1.11 Keypad [HUION Huion Tablet] on usb-0000:00:14.0-13/input0
[  271.342560] uclogic 0003:256C:006D.000A: No inputs registered, leaving
[  271.342667] uclogic 0003:256C:006D.000A: hidraw5: USB HID v1.11 Device [HUION Huion Tablet] on usb-0000:00:14.0-13/input1

lsusb:

>...
Bus 001 Device 007: ID 256c:006d HUION Huion Tablet
...

xsetwacom list:

>HUION Huion Tablet Touch Strip pad	id: 11	type: PAD   
HUION Huion Tablet Pad pad      	id: 12	type: PAD   
HUION Huion Tablet stylus       	id: 13	type: STYLUS

additionally it the gnome settings in tablet settings after pluging huion tablet I see 3 Huion H950P devices
[image]

Br,
Lukasz

edoiks, 2021-11-28:
After uninstaling digimend-kernel-driver, the Huion H1161 is stil recognized as H950P
lsmod (after uninstaling digimend-kernel-drivers):

>...
usbhid                 61440  2 hid_uclogic,hid_logitech_dj
hid                   135168  8 **usbhid**,**hid_uclogic**,**hid_generic**,hid_logitech_dj,hid_logitech_hidpp
nvidia_drm             61440  5
...

it seams that there is some problem with default HID driver