DIGImend/digimend-kernel-drivers #559: H640P Calibrate won't work

pixelprato, 2021-08-21:
Hi all, I'm on latest Mint Cinnamon, with a H640P (Huion). I did install with the .deb and also another time manual dkms.

Each time the mapping on the tablet is squeezed to the left, like taking only the half of the tablet and trying the GUI provided from Super key / Graphics Tablet / Calibrate, show me dark screen with some target to click on, but I don't have my cursor displayed so it's not working.

Next step for me, linux newbe, will be to try digging manually into the conf file maybe...
But some proposed path started with /etc/X11 while there's no config file already created, maybe I should try to create ones from the /usr/share/X11 one, as few tutos about it could be found onto internet.

pixelprato, 2021-08-22:
I did try to change the Area from xsetwacom for its default 0 0 32767 32767, but it didn't work. lower or higher x2 value don't change the tablet area but the "scale" to the screen.
[image]

kurikaesu, 2021-08-22:
Usually the "pad" are the shortcut/express keys there on the left.
The actual digitizer/pen area is named Pen or Stylus.