DIGImend/digimend-kernel-drivers #563: Does not even work on Multilaser MX002 on Ubuntu 20.04

cssobral2013, 2021-09-02:


nic3-14159, 2021-09-02:
Hello! I don't think DIGImend currently has support for this tablet. If you would like support, could you collect and post information as described on this page? https://digimend.github.io/support/howto/trbl/diagnostics/

That would be very useful for anyone working on supporting this tablet. Thanks!

cssobral2013, 2021-09-02:
@nic3-14159 - lsusb returns the following:
Bus 001 Device 010: ID 08f2:6811 Gotop Information Inc. [T501] Driver Inside Tablet

cssobral2013, 2021-09-02:
@nic3-14159 - The USB HID descriptor is [here](https://github.com/DIGImend/digimend-kernel-drivers/files/7095193/hid_report_descriptors.txt) and lsusb descriptor is 
 [here](https://github.com/DIGImend/digimend-kernel-drivers/files/7095195/descriptors.txt).