DIGImend/digimend-kernel-drivers #576: s56/huion_t156_160726 pen not working in debug/usb/usbmon/1u but streams in usbhid-dump

Veek, 2021-10-30:
**_lsusb_**
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0781:5575 SanDisk Corp. Cruzer Glide
Bus 001 Device 003: ID 256c:006e  Tablet Monitor
Bus 001 Device 005: ID 04f2:0939 Chicony Electronics Co., Ltd Amazon Basics mouse
Bus 001 Device 002: ID 04d9:1203 Holtek Semiconductor, Inc. Keyboard
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
root@debian:~# _**modprobe hid_uclogic**_
root@debian:~# dmesg
[ 1765.667982] input: Tablet Monitor as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:256C:006E.0003/input/input42
[ 1765.668379] input: Tablet Monitor Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:256C:006E.0003/input/input43
[ 1765.668551] input: Tablet Monitor Touch Strip as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:256C:006E.0003/input/input44
[ 1765.668738] input: Tablet Monitor Dial as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:256C:006E.0003/input/input45
[ 1765.669001] uclogic 0003:256C:006E.0003: input,hiddev1,hidraw2: USB HID v1.11 Keypad [Tablet Monitor] on usb-0000:00:14.0-2/input0
[ 1765.780596] input: Tablet Monitor Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:256C:006E.0004/input/input46
[ 1765.781963] uclogic 0003:256C:006E.0004: input,hiddev0,hidraw3: USB HID v1.11 Mouse [Tablet Monitor] on usb-0000:00:14.0-2/input1
[ 1765.995740] uclogic 0003:256C:006E.0005: interface is invalid, ignoring

```

evtest
```
root@debian:~# **_evtest_**
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      HID 04d9:1203
/dev/input/event1:      HID 04d9:1203 System Control
/dev/input/event2:      HID 04d9:1203 Consumer Control
/dev/input/event3:      Tablet Monitor
/dev/input/event4:      Tablet Monitor Pad
/dev/input/event5:      Tablet Monitor Mouse
/dev/input/event23:     HDA Intel PCH HDMI/DP,pcm=10
/dev/input/event26:     Tablet Monitor Touch Strip
/dev/input/event27:     Tablet Monitor Dial
```

[pad.txt usbmon output after unplugplug](https://github.com/DIGImend/digimend-kernel-drivers/files/7447016/pad.txt)
**_lsusb_** shows it on _Bus 001 Device 007: ID 256c:006e  Tablet Monitor_
The pen doesn't register any activity on usbmon even though it streams data and is working via **_usbhid-dump -es -m 256c:006e_**

Pad was working but stopped .. I am on Debian 11.1 debian 5.10.0-9-amd64

kurikaesu, 2021-10-31:
This device is not on the supported device list.
Could you follow the instructions here on how to collect the necessary diagnostics in order to have a developer add support for it? https://digimend.github.io/support/howto/trbl/diagnostics/

Veek, 2021-11-03:
I enabled debugging and this is what I see in dmesg just before a data-structure dump

`root@debian:~# dmesg|grep -i ignor
[  164.819470] uclogic 0003:256C:006E.000A: interface is invalid, ignoring
[  717.945439] uclogic 0003:256C:006E.000D: interface is invalid, ignoring
[  718.003310] uclogic 0003:256C:006E.000D: interface is invalid, ignoring
[  736.265583] uclogic 0003:256C:006E.0010: interface is invalid, ignoring
[  736.354551] uclogic 0003:256C:006E.0010: interface is invalid, ignoring
[  756.999865] Generic FE-GE Realtek PHY r8169-0-200:00: attached PHY driver [Generic FE-GE Realtek PHY] (mii_bus:phy_addr=r8169-0-200:00, irq=IGNORE)
[  886.057513] uclogic 0003:256C:006E.0013: interface is invalid, ignoring
[  886.082904] uclogic 0003:256C:006E.0013: interface is invalid, ignoring
[  907.033953] uclogic 0003:256C:006E.0016: interface is invalid, ignoring
[  907.059292] uclogic 0003:256C:006E.0016: interface is invalid, ignoring
[ 1077.185413] uclogic 0003:256C:006E.0019: interface is invalid, ignoring
[ 1077.213553] uclogic 0003:256C:006E.0019: interface is invalid, ignoring
`

```
.frame_list = {
[  164.819417] hid_uclogic:uclogic_params_hid_dbg:125: uclogic 0003:256C:006E.000A
:     {
[  164.819418] hid_uclogic:uclogic_params_frame_hid_dbg:91: uclogic 0003:256C:006E
.000A:                .desc_ptr = 0000000000000000
[  164.819420] hid_uclogic:uclogic_params_frame_hid_dbg:92: uclogic 0003:256C:006E
.000A:                .desc_size = 0
[  164.819421] hid_uclogic:uclogic_params_frame_hid_dbg:93: uclogic 0003:256C:006E
.000A:                .id = 0
[  164.819423] hid_uclogic:uclogic_params_frame_hid_dbg:94: uclogic 0003:256C:006E
.000A:                .suffix = (null)
[  164.819424] hid_uclogic:uclogic_params_frame_hid_dbg:95: uclogic 0003:256C:006E
.000A:                .re_lsb = 0

My pen_probe starts with but I am not able to generate the stream every time
001:003:000:STREAM             1635932820.215477
 07 00 00 00 6A 20 9B 04

001:003:000:STREAM             1635932820.217191
 07 80 E8 32 A8 0C 00 00

001:003:000:STREAM             1635932820.221230
 07 80 E8 32 A8 0C 00 00

```

Veek, 2021-11-03:
[Uploading dmesg-tablet.txt…]()
So I've uploaded my dmesg logs for the error

kurikaesu, 2021-11-03:
The messages about invalid interfaces are not necessarily errors. Some of the interfaces that the devices have that we probe just aren't useful for running the standard tablet functionality and are ignored by the driver by marking them as invalid.

The pen streams you pasted show looks like digitizer/pen messages. You were moving the pen around on the tablet surface right?
I think with just a little bit of probing I may be able to get that tablet working on my userspace drivers.

Veek, 2021-11-03:
yep, okay lemme upload the full pen_probe.txt as per video BUT NOTE THAT most-times the pen is dead.. and I cannot generate any data.. for whatever mysterious reason.. usbhid-dump -es -m 256c:006e does nothing currently..

[pen_probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7468002/pen_probe.txt)

Veek, 2021-11-03:
crud it seems to be a problem with the pen making proper connections with the tiny connector - I used a wire to hook it up and it works.. weird how the heck was it generating stream data but not working and displaying a pointer..

kurikaesu, 2021-11-03:
Do you mean the cable you are using was broken? Does it work correctly now that the cable has been replaced?

Veek, 2021-11-03:
nope - not the cable - yep it's working (pointer's back)
1. the pen for sure - there's a spring at the bottom of the pen barrel that connects the copper barrel along the body of the pen to the battery-ve terminal. That was off somehow so I fixed that by shoving some aluminium foil to make a better connection between -vebattery-spring-copperBarrelThatRunsTheLengthOfPen
2. then at the top of the pen copper-barrel  it has to slot or connect to a tiny tab like thing that takes the -ve to the copper coil in the pen tip - so I slotted some more aluminium foil to make a better connection
3. now the frigging thing works (after screwing it all in)

curses! I tested this before using wire and it was not working so I concluded that my debian kernel change had mucked things up. Thanks man! (for bothering to reply)

kurikaesu, 2021-11-03:
hahaha good to know that it works and its actually a hardware issue!
If you don't have any actual software issues its best to close this ticket

closed 2021-11-03 by Veek