DIGImend/digimend-kernel-drivers #579: Medion/Tevion MD41217

beaka, 2021-11-09:
I am sending required info on a Medion/Tevion MD41217 graphics tablet. It is an older model but still widely requested for drivers.
I am using Fedora 34 Gnome X.org and all the drivers/methods I can find but still no luck.
On USB trace I can see the packets coming from the device when pen down and moving so I guess the tablet works.
I can see the Aiptek driver as an input on xinput list.
Following your diagnostics page I cannot see the device id's in the HID folder /sys/kernel/debug/hid

Hope you can help.

/home/bruce/Documents/medion/descriptors.txt
/home/bruce/Documents/medion/frame_wheel_srolling.txt
/home/bruce/Documents/medion/hid_report_descriptors.txt
/home/bruce/Documents/medion/md41217.pdf
/home/bruce/Documents/medion/medion41217.jpg
/home/bruce/Documents/medion/pen_buttons.txt
/home/bruce/Documents/medion/pen_coords.txt
/home/bruce/Documents/medion/pen_pressure.txt
/home/bruce/Documents/medion/probe.txt
/home/bruce/Documents/medion/tablet.txt

nic3-14159, 2021-11-09:
Thanks for posting diagnostics. I took a look at them and some things seem off. It seems to be reporting X and Y coordinates correctly, but nothing seems to correspond to the pressure and pen buttons. Based on the original hid-descriptors, I would expect that the 3rd last byte would change when pressing the tablet buttons, and the last 2 bytes would change as you increased the pen pressure. Perhaps the Windows driver does something to initialize the tablet and enable the pen buttons and pressure?

If you have access to a Windows machine, you could try installing the driver and then using Wireshark to sniff the USB traffic while plugging in the tablet and performing similar motions as when you used usbhid-dump, and then post the pcap dump. [Supposedly this tablet is a rebranded Waltop 12000U](https://www.deviantart.com/darklordsdaughter/journal/Driver-for-Tevion-Medion-MD-41217-288259780), and drivers are available from Waltop (link at bottom of page, under Joy Graphic Tablet):
http://www.waltop.com/english/03_support/01_download.php?xpid=1&MID=1

I took a look at the files in the driver download and there is an ini file that does contain the USB ids for this tablet, so it seems like that claim is correct.

beaka, 2021-11-09:
Hi nic, thanks for the info, it helped a lot.
I tried the driver on win8 and krita and got movement but no pen down to paint. I suspect it might be the pen as the original was broken so I got  huion P68 which is supposed to be compatible.
I am attempting to fix the original pen but my smd skills are a bit rusty so will see how it goes.
The pen is supposed to work at 165Khz so I am trying to find info about how it works, just to test.
Probably easier to buy new tablet, but where's the fun in that :)
Will let yo u know my progress.

xchg-dot-ca, 2023-03-16:
Hey guys, I've dug-up such tablet from storage and was wondering if I can make it working under modern Ubuntu
Apparently it was working with older version here is some info, may be it will help:
https://help.ubuntu.com/community/AiptekTablet

beaka, 2023-03-16:
I am using Fedora 37 and it works, it turned out my pen was faulty and with
a new pen it just worked with no special drivers needed but a while since I
worked on it so I will dig it out and see what was required.

BobScribbler, 2023-07-11:
Way out of date and far too late to help anyone, but. Problems with Windows (10 etc) and getting the great **Medion (Aldi) USB tablets and wireless pens** (and mouse) to work. Typically 'not recognised', 'pen' curser jumping all over the screen uncontrollably etc, 'unsupported' drivers out of date, re-installing the s/ware (drivers etc) changes nothing? Just uninstall the Medion s/ware all together. Windows will identify there is a USB tablet (and pen) and gets it 'wrong'. Windows 'generic' USB tablet and 'wireless' pen drivers will now work pretty well. Far better than sending more of these Medion tablets to landfill - Happy Scribbling . . . . .