DIGImend/digimend-kernel-drivers #580: XP-Pen Artist 24 Pro kernel unable to enumerate device.

acklavidian, 2021-11-13:
I have somehow fudged the config for my XP-Pen Artist24 Pro. I had another XP-Pen Artist22E and for some reason the pen stopped working but the display was fine. I ended trading up to a Artist24 Pro with thinking it was a hardware fault. Now I have the same problem with the new one. The only output is I can find is from dmesg:
```
[Nov13 11:50] usb 3-1.1: new full-speed USB device number 11 using xhci_hcd
[  +0.092111] usb 3-1.1: device descriptor read/64, error -32
[  +0.199997] usb 3-1.1: device descriptor read/64, error -32
[  +0.199887] usb 3-1.1: new full-speed USB device number 12 using xhci_hcd
[  +0.092116] usb 3-1.1: device descriptor read/64, error -32
[  +0.199941] usb 3-1.1: device descriptor read/64, error -32
[  +0.108372] usb 3-1-port1: attempt power cycle
[  +0.631575] usb 3-1.1: new full-speed USB device number 13 using xhci_hcd
[  +1.015544] usb 3-1.1: Device not responding to setup address.
[  +1.219994] usb 3-1.1: Device not responding to setup address.
[  +0.204458] usb 3-1.1: device not accepting address 13, error -71
[  +0.092003] usb 3-1.1: new full-speed USB device number 14 using xhci_hcd
[  +1.015552] usb 3-1.1: Device not responding to setup address.
[  +1.219982] usb 3-1.1: Device not responding to setup address.
[  +0.204461] usb 3-1.1: device not accepting address 14, error -71
[  +0.000521] usb 3-1-port1: unable to enumerate USB device
```

```Linux pop-os 5.13.0-7620-generic #20~1634827117~21.04~874b071-Ubuntu SMP Fri Oct 29 15:06:55 UTC  x86_64 x86_64 x86_64 GNU/Linux```

kurikaesu, 2021-11-14:
That message usually happens to me when the device is plugged into a hub in the USB device tree that has run out of bandwidth. Most often happens when I plug two webcams into the same device tree.

acklavidian, 2021-11-16:
Well I figured it out. I was managing my PopOS dkms through `apt` and `flatpak` CLI tools. At some point was messing with this thread: https://askubuntu.com/questions/1098845/xp-pen-pentablet-driver-installing-issues-on-ubuntu-16-04 After removing the `.rules` files and user/permissions I purged and reinstalled the dkms&digimend `.rules` and rebooted thinking that was enough to trigger udev to load the digimend rules. Apparently I was wrong. After running `udevadm control --reload-rules && udevadm trigger` everything started working normally again. I thought that after deleting the rules and `apt`/reinstall/reboot would have triggered udev to load the new config and override the old rules.  Anyhow, I am just glad to have it working again. Thanks for the awesome work on these drivers. My new Artist24 Pro is a blast on linux! [image]

P.S. if anyone has a `gamemode` script for switching this thing off I'd be super grateful :smile:

closed 2021-11-16 by acklavidian