DIGImend/digimend-kernel-drivers #581: Left Handed Orientation doesn't work with HUION H610 Pro V2 on Ubuntu 20.04.3

osmium64, 2021-11-16:
Hi. This is my first time using github to solve an issue with my Huion graphics tablet.
I would like to know if there is any way to make left handed orientation work on my laptop with Ubuntu 20.04.3 and my Huion H610 Pro V2.

Other than need to use it in left handed mode, the tablet works fine (pressure and tilt works; can assign shortcuts on the default Ubuntu's Wacom tablet gui...)

Thanks in advance.

osmium64, 2021-11-16:
> Assuming you are using the wacom X11 drivers for your device you can use xsetwacom to rotate the orientation:
> 
> ```
>  xsetwacom set "device_name" Rotate none|half|cw|ccw
> ```
> 
> https://www.mankier.com/1/xsetwacom

I didn't install the digimend driver first because it removes the wacom panel gui to set shortcuts (which looks way more convenient to me).
Anyways, I ended up installing it and setting my Huion to rotate 180 with the following command:

`xsetwacom set "HUION Huion Tablet stylus" Rotate half`

Thanks for the xsetwacom link. Hope I don't mess with the tablet keys to set shortcuts.

kurikaesu, 2021-11-16:
If your tablet was configurable through the wacom panel gui then it is likely that your device was already being supported by the wacom drivers and you didn't need to install digimend in order to use the xsetwacom command above.

osmium64, 2021-11-16:
Ok, I thought so. But switching to Left Handed Orientation wasn't working. Still isn't. That is why I started this topic. The only way I can turn my tablet orientation was by instaling digimend drivers and the command line above...