DIGImend/digimend-kernel-drivers #584: gaomon s620 two cursor

serhio1212, 2021-11-26:
Hi
I install  digimend-dkms V10 for deboan 10.
I have Gaomon s620 tablet. 
the tablet is working, but bad.
I see two pointers at the same time. Sometimes the pen cursor disappears. Especially often on the page google.com.
I turned off hardware acceleration in my browser settings, but it didn't help.
Also, the cursor sometimes changes from pointer to cross.
Can I recommend something?

kurikaesu, 2021-11-27:
Debian 10 uses wayland by default as the compositor. There is currently a bug in wayland that makes 2 mouse pointers appear when using a tablet. I see the same thing when I run wayland on Ubuntu 21+

serhio1212, 2021-11-27:
Ubuntu 18.04 worked normally. Although there is also wayland.

kurikaesu, 2021-11-27:
Ubuntu 18.04 does not use wayland by default. It uses X11/xorg.

serhio1212, 2021-11-27:
Sorry, you're right. Planned with wayland but eventually released with X11/xorg.  Wait new version wayland (updates) or digimend-dkms.