DIGImend/digimend-kernel-drivers #585: H1161 Mistaken as H950P - Two buttons missing

persephius, 2021-12-10:
I've been messing around with DIGImend and it's been such a blessing, works out of the box (almost) perfectly and I'd like to thank the devs immensely for that, but there's a small issue: My H1161 gets mistaken as a H950P in the KDE configs.

Without the drivers, the tablet isn't even recognized, so I'm assuming somehow H1161 and H950P share the same product ID (006D), that wouldn't be a problem if the H1161 didn't have two extra buttons, and KDE is showing as if there were only 8, rendering two of the tablet's buttons unusable (or at least kind of a pain to configure since I would probably have to find a way to configure them through xsetwacom scripts). I've seen people reporting theirs worked fine, and it works perfectly fine apart from this small issue, is there any way I could solve this?

[image]

Using lsusb shows that my tablet is indeed under the 006d ID.
`Bus 003 Device 002: ID 256c:006d HUION Huion Tablet`

**System**: Kubuntu 21.10 x86_64
**Kernel**: 5.13.0-22-generic
**DE**: KDE
**DIGImend version**: 10

I've read about using GUI to configure tablets not being the best way, but so far KDE has the best configuration tool I've seen, I'm really not into messing with scripts to set buttons here and there, seems very gimmicky, so any help is appreciated. I'm dualbooting with Windows 10 and I can confirm the tablet works properly there.

albfan, 2022-01-26:
Probably nothing to do with kernel, but with library exposing it: https://github.com/linuxwacom/libwacom/pull/288#issuecomment-1022420947

see all or many huion tablets expose same usb vendor and product.

For me disabling H950p was the only option.

JoseExposito, 2024-05-03:
Hi!

[DIGImend v13](https://github.com/DIGImend/digimend-kernel-drivers/releases/tag/v13) exposes a new field to be able to recognize your tablet. You'll need to wait a little bit until the next release of libwacom is released and it is packaged by your distro, but it'll work again in the future.

Closing the issue as the DIGImend part of the work is done.

closed 2024-05-03 by JoseExposito