DIGImend/digimend-kernel-drivers #586: Support XP-Pen Star 03 v1

MaurizioCasciano, 2021-12-11:
I just uploaded the diagnostics of the XP-Pen Star 03 v1
at https://github.com/DIGImend/tablets/pull/49
orat
I would be happy to code and test it if there is some documentation you can point me to.

nic3-14159, 2021-12-12:
Hello!
DIGImend appears to have code already for that tablet's ids. It seems like the Star 03 is a rebrand of the Ugee M708. Could you install DIGImend as is and report if it works?

MaurizioCasciano, 2021-12-12:
Hi, I installed the v10 deb package and the pen is working.

```
$ lsusb
Bus 001 Device 005: ID 5543:0081 UC-Logic Technology Corp. TABLET 1060N
```

The tablet is correctly handled by the Wacom driver.
```
$ xsetwacom list
UC-Logic TABLET 1060N Pen stylus	id: 18	type: STYLUS    
UC-Logic TABLET 1060N Pad pad   	id: 19	type: PAD   
```

I was also able to [configure the buttons](https://github.com/DIGImend/digimend-kernel-drivers#configuring-wacom-xorg-driver) of the tablet:

```
$ xsetwacom set "UC-Logic TABLET 1060N Pad pad" button 1 key A
$ xsetwacom set "UC-Logic TABLET 1060N Pad pad" button 2 key B
$ xsetwacom set "UC-Logic TABLET 1060N Pad pad" button 3 key C
$ xsetwacom set "UC-Logic TABLET 1060N Pad pad" button 8 key D
$ xsetwacom set "UC-Logic TABLET 1060N Pad pad" button 9 key E
$ xsetwacom set "UC-Logic TABLET 1060N Pad pad" button 10 key F
$ xsetwacom set "UC-Logic TABLET 1060N Pad pad" button 11 key G
$ xsetwacom set "UC-Logic TABLET 1060N Pad pad" button 12 key H
```

ehtishamullah, 2022-08-08:
I also installed v10 deb package but it is not working.
lsusb gives:
Bus 001 Device 020: ID 5543:0081 UC-Logic Technology Corp. TABLET 1060
but "xsetwacom list" gives nothing.
the tablet is connected but settings>wacom doesnt detect anything

nic3-14159, 2022-08-08:
> I also installed v10 deb package but it is not working. lsusb gives: Bus 001 Device 020: ID 5543:0081 UC-Logic Technology Corp. TABLET 1060 but "xsetwacom list" gives nothing. the tablet is connected but settings>wacom doesnt detect anything

@ehtishamullah Is this after restarting the system? As far as I know some of the new Xorg settings that are installed with the package in order for xsetwacom to work don't take effect until the X11 server is restarted, such as after a reboot.

ehtishamullah, 2022-08-08:
my bad. just restarted pc and now it is working. I am new to linux (2nd day) and was told that you dont need to restart pc for anything when using ubuntu. Thank you so much for your quick reply