DIGImend/digimend-kernel-drivers #588: How can I map "Alt + Middle Mouse Button" to a key on my tablet?

amanda-santini, 2021-12-19:
Hi there!

I'm running your amazing driver and I'm loving that I can use my tablet (Huion H320M) on Linux. 
However, there's a little problem: on WIndows I used to map, say, the last button on the bottom, to "Alt + Middle Mouse Button". This helps me a lot since I have tendinitis and I don't need to press many keys while I work.

How could I do the same with your driver? I tried something like the command bellow, but I'm not able to set that key as I want it to behave.
xsetwacom set "HUION Huion Tablet_H320M Pad pad" button 1 key Alt button 3

nic3-14159, 2021-12-19:
Hello! I think that xsetwacom command should work. However, the middle mouse button is button 2, not button 3 (which is right click)

amanda-santini, 2021-12-19:
> 

I kinda got it working by the following command:
[code]  xsetwacom set "HUION Huion Tablet_H320M Pad pad" button 1 "key +Alt +button 1"[/code]

However, it's as if I quickly press and release the keys, even though I have the button pressed all the time.
Still investigating it.

amanda-santini, 2021-12-19:
Thanks guys. If this is of any help, I was able to do it via configuration file. 
xsetwacom set "HUION Huion Tablet_H320M Pad pad" button 15 "key +Alt button +3"

However, it doesn't work with the stylus tip (hovering over the tablet) for some reason, it only works if I press the tablet key and if I move my actual mouse. Super weird.

nic3-14159, 2021-12-19:
What software are you trying to use with this mapping? That would be useful for trying to reproduce this behavior. It is strange that your mapping isn't working with the stylus and only with the actual mouse though. My XP-Pen tablet seems to work with a similar mapping and the window manager shortcuts assigned to alt+mouse buttons

amanda-santini, 2021-12-19:
Maya, Substance Painter, and Blender.
I've now tested all 3 of them. Blender and Substance Painter work really well!
It's only Maya that's giving me a headache. 

See this video: https://www.youtube.com/watch?v=10pA4R093S0

It's like Maya only shows the first and end parts of my movement, so I don't a fluid movement.

skytale75, 2022-01-04:
Hi, great drivers . . . updated my tablet lately to a huion q620m, and even though it is not listed on your compatibility list, it still works for setting buttons . . . 

However, I cant make the dial work . . . and more importantly . . . the tilt function on the pen. The dial doesn't bother me much, but the titl function on the pen would be huge.

Anyways, thanks for putting the driver together, so I can use this at all. If I knew C I woul offer to help.