DIGImend/digimend-kernel-drivers #597: Acepen 1060N Diagnostics

myosotiss, 2022-02-04:
lsusb shows:
Bus 003 Device 002: ID 5543:0081 UC-Logic Technology Corp. TABLET 1060N
The usb device id is found in xorg.conf under the "Ugee/XP-Pen" section, but it does not work at all with the current drivers.
I'm getting continue IRQ Status -75 kernel messages when attempting to use it.

[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/8006177/frame_buttons.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/8006178/pen_buttons.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/8006179/pen_pressure.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/8006180/pen_coords.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/8006181/probe.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/8006182/hid_report_descriptors.txt)
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/8006183/descriptors.txt)

nic3-14159, 2022-02-06:
Hello! Thanks for posting these. It seems like this tablet uses 10 byte reports, whereas the existing code for tabets with the same USB IDs as the Acepen expects 8 byte reports. If I recall correctly, the -75 IRQ status is one of the side effects of the reports and the descriptor not matching in length.

So some code will need to be written to properly reflect the 10 byte reports, as well as distinguish the Acepen 1060N from other tablets with the same IDs. Reused USB IDs makes telling different tablets apart more difficult, but I think there is enough unique information in probe.txt to tell this one apart.

myosotiss, 2022-02-06:
Thanks for those details!
I was able to get the driver working for me with the following patch on master.
```diff
diff --git a/hid-uclogic-rdesc.c b/hid-uclogic-rdesc.c
index 6381911..3a292e4 100644
--- a/hid-uclogic-rdesc.c
+++ b/hid-uclogic-rdesc.c
@@ -703 +703 @@ const __u8 uclogic_rdesc_v1_frame_arr[] = {
-       UCLOGIC_RDESC_FRAME_BUTTONS_BYTES(UCLOGIC_RDESC_V1_FRAME_ID, 8)
+       UCLOGIC_RDESC_FRAME_BUTTONS_BYTES(UCLOGIC_RDESC_V1_FRAME_ID, 10)
```

Obviously this doesn't do "device detection" but since I don't have any other tablets, this works for me for now. Thanks for your help!