DIGImend/digimend-kernel-drivers #599: Gaomon 1060 Pro - not correct work

bo-skor, 2022-02-06:
Hello! For first I want to say great thanks for GREAT work!

Unfortunately, I have a trouble with Gaomon 1060 Pro
kubuntu 21.10
Operating System: Kubuntu 21.10

> KDE Plasma Version: 5.22.5
> KDE Frameworks Version: 5.86.0
> Qt Version: 5.15.2
> Kernel Version: 5.13.0-21-generic (64-bit)
> Graphics Platform: X11
> Processors: 8 × AMD Ryzen 7 PRO 2700U w/ Radeon Vega Mobile Gfx
> Memory: 6.7 ГиБ of RAM
> Graphics Processor: AMD Radeon™ Vega 10 Graphics

lsusb: `Bus 001 Device 002: ID 256c:0064 HUION Huion Tablet`

after installing driver (deb version) no reactions from device, but when i try to use wacom method with editing `/etc/X11/xorg.conf.d/50-tablet.conf` device begin recive actions. But not correct. The active area was strongly elongate horizontally.

I will be very grateful for any help!

nic3-14159, 2022-02-06:
Hello! Based on the ID in the lsusb output, DIGImend does not yet support this tablet. Could you collect and post diagnostics as described by [this page](https://digimend.github.io/support/howto/trbl/diagnostics/)? That would be very helpful for adding support for this tablet to DIGImend. Thanks!

JoseExposito, 2024-05-03:
Hi!

Your issue should be fixed by [DIGImend v13](https://github.com/DIGImend/digimend-kernel-drivers/releases/tag/v13). Please, if you have a chance, give it a try and report any problems found.

Thanks!

closed 2024-05-03 by JoseExposito