DIGImend/digimend-kernel-drivers #614: enquiry about Deco L

linuxgirl22, 2022-04-15:
Hello :-) I can see it listed in the vid:pid thread, but may I ask if the Deco L is supported? Thanks.

nic3-14159, 2022-04-18:
Hello! The Deco L does not appear to be supported at the moment. If you have the tablet and would like support for it, could you collect and post diagnostics as described on [this page](https://digimend.github.io/support/howto/trbl/diagnostics/)?
Thanks!

linuxgirl22, 2022-04-18:
Thanks! It's in the post, I think arriving some time this week. When it arrives, I'll make sure to follow the instructions on that page and post back. :-)

JoseExposito, 2022-06-20:
Hi!

I added support for the Deco L in the mainline (official) kernel, it'll be released in 5.20:

https://git.kernel.org/pub/scm/linux/kernel/git/hid/hid.git/commit/?h=for-5.20/uclogic&id=0cb1fc0988e32bda84c2b7218e0c761af1430baf

In case waiting until 5.20 is released is not an option, I built a copy of DIGImend adding support for your tablet, here is the source code:

[digimend-kernel-drivers.tar.gz](https://github.com/DIGImend/digimend-kernel-drivers/files/8938140/digimend-kernel-drivers.tar.gz)
 
Follow the instructions available in the [README](https://github.com/DIGImend/digimend-kernel-drivers#digimend-kernel-drivers) to install it.

@nic3-14159 I see you are active in the issue tracker, if you have merge permissions I can open a PR with the code present upstream.

linuxgirl22, 2022-06-20:
Thank you very much indeed @JoseExposito Really appreciate! Am running 5.10 kernel on MX Linux, where can also do USB backup snapshot of system ... will do that first, just incase I mess something up, lol, then take a better look at the readme information. Thanks for the zip file and link. :-)

JoseExposito, 2022-06-20:
Cool, thanks for your feedback. The problems with Wine are unrelated to the kernel driver.

Honestly, I have no idea what's required to get a tablet working on Wine.
Hopefully, someone around here will be able to help you.

linuxgirl22, 2022-06-21:
Thanks again for your help with this. :-)

nic3-14159, 2022-06-26:
> @nic3-14159 I see you are active in the issue tracker, if you have merge permissions I can open a PR with the code present upstream.

Hi! Yeah, I've pretty much been the only person somewhat active here on DIGImend lately, but unfortunately I don't have merge permissions. Nothing has been merged for nearly a year, and it's been around 2 years since any actual code changes to the driver have been merged. I'll see if I can find a way to revive the project; I'd hate to see it die out.