DIGImend/digimend-kernel-drivers #620: Huion H430P - Relative mode behaving like Absolute

Boxieharp, 2022-06-23:
Hey guys, recently I bought this tablet. My intended purpose was to use it for videogames.

After installing [the drivers described here](https://wiki.archlinux.org/title/Wacom_tablet) (xf86-input-wacom for access to xsetwacom, and digimend-kernel-drivers-dkms-git for my tablet to be recognized) the tablet worked was detected by my system and I could finally move the cursor. It works well when moving the cursor on the desktop and in painting applications. But I've run into a few issues, the biggest of them being the fact that if I try setting relative mode by using:

> xsetwacom set *ID* mode relative

The only appreciable change is that the sensitivity becomes higher, the tablet itself still behaves as if I was in Absolute mode, jumping the pen around makes the cursor jump around as if it was mapped to a section of the screen instead of behaving as a mouse.

Every tutorial, video, blog, and information I've found seems to describe that it should change the behavior but I'm stumped.

The other big issue would be that, in games. If the game has a cursor I can move it perfectly fine, it's flawless. But as soon as the game swaps over to moving the camera with the mouse, I'm unable to move it. Sometimes I'm able to swap over to Relative mode and press a couple buttons on the tablet, and suddenly I can start moving the camera with the pen. No idea what's going on there either.

I've tried with [the official Huion drivers](https://huion.com/index.phpm=content&c=index&a=lists&catid=16&myform=1&down_title=h430p) and while these allow me to use relative mode without issues, when it comes to games, the camera just goes crazy and I'm unable to control it as the character just spins in place endlessly regardless if I'm holding the pen close to the surface or not.

I'm using Arch Linux with the Zen Kernel version 5.18.6, however I've tried with the default Kernel and I see the same behavior.

My window manager is herbstluftwm, however I have tried with i3, Openbox, and a full fledged DE like Gnome, and I see the same behavior. Even swapping over the settings from "Pen" to "Mouse" mode in the Gnome tablet settings was showing the same behavior.

At this point I'm wondering if it might be a config error with the usable area or something, but no idea how I could go around fixing it.

Hopefully you guys can help me. If this helps, the Vendor and Product ID of the tablet in question is  256c:**006d**.

Thanks beforehand!

MisterNoNameLP, 2022-06-23:
Where do I get the driver logs to post it here?

Boxieharp, 2022-06-23:
> I have the same issue with the H640P since the last update (today). I dont know since when this issue appears, have not updated the driver for some time. but it was working a while ago.
> 
> As work around you can use the 'opentabletdriver' from the aur repo. But you have to disable this drive before using by executing "sudo rmmod hid-uclogic".

Hey! Thanks a lot, this is a pretty nice workaround for now. I'll stick with it and might try with these drivers if there's a solution.
If there might be any information I can provide to help with it, please let me know.