DIGImend/digimend-kernel-drivers #621: HOw do i start adding a unsupported tablet

101Garfield101, 2022-06-24:
Hi,  i have XP-Pen star 06 tablet and works fine under windows.
How do i get it to work under linux. I did some tests a while ago and got some data (although i forgot what i did)
So data is available and can retrieved from the device... and since other tablets probably work the same, it must be possible to get it working. 
Since i am not a person to look and seek individual puzzle pieces and glue-ing them together,  And since other tablets have been added before:  Is there a procedure on how to get is working?  Is the a manual / step-by-step procedure? 

please help i want to understand!

nic3-14159, 2022-06-26:
Hello! Is this the wireless version of the Star 06? (The output of `lsusb` should show an id of `28bd:0078` for this tablet)? If so DIGImend already has basic support for the pen (no buttons), but I believe is currently broken due to some changes and requires #553 to fix.
However, I also have this tablet and have support for it on my fork: https://github.com/nic3-14159/digimend-kernel-drivers/tree/Star06_new_init

101Garfield101, 2022-06-26:
Thanks Nicholas, it can be used both wired or wireless. I will try when I
am home.

Is there any documentation on how to incorporate a new tablet in digimed?

Kind regards, Ruud

nic3-14159, 2022-06-26:
> Thanks Nicholas, it can be used both wired or wireless. I will try when I am home. Is there any documentation on how to incorporate a new tablet in digimed? Kind regards, Ruud

Unfortunately there isn't really any documentation on incorporating a new tablet into digimend. However it is something that I would like to work on at some point.

JoseExposito, 2022-07-11:
Hi @101Garfield101,

I just added support for the XP-PEN Deco Pro Small tablet to the mainline kernel:
https://lore.kernel.org/linux-input/20220710175043.192901-1-jose.exposito89@gmail.com/T/

I don't own the tablet. Instead, I built a modified version of DIGImend and the user who reported the issue tested it for me:
https://gitlab.freedesktop.org/libinput/libinput/-/issues/738#note_1453926

Your tablet should be very similar. If you could test the code, I'll add support for it in the kernel as well.

Could you test these drivers?

First, download the source code:  [digimend-kernel-drivers.tar.gz](https://github.com/DIGImend/digimend-kernel-drivers/files/9081429/digimend-kernel-drivers.tar.gz)

The source code contains the code from DIGImend plus the changes required to add support for your tablet.

In order to install it, you'll need to install the kernel headers:

```
$ sudo apt-get install -y "linux-headers-$(uname -r)"
```

After extracting the code, cd into the source code and:

```
$ make
$ sudo make install
```

You can uninstall it running:

```
$ sudo make uninstall
```

More detailed instructions are available in the [DIGImend README file](https://github.com/DIGImend/digimend-kernel-drivers#digimend-kernel-drivers).

I hope it works, if not I'll ask you for more logs :)

101Garfield101, 2022-07-11:
Thanks, I'll check it out. It will take some while for met to test is.but I
will. Tnx