DIGImend/digimend-kernel-drivers #622: Support for XP-Pen Deco Pro MW Tablet

andigandi, 2022-07-12:
The XP-Pen Deco Pro MW supports both USB and Bluetooth Connection: https://www.xp-pen.com/product/915.html

As mentioned here https://github.com/DIGImend/digimend-kernel-drivers/issues/366#issuecomment-1180654739 @JoseExposito what more information do you need? What can I do to test the driver with the tablet?

ID when connected via USB. The computer does not have bt, so I can't test that.

`
    # ID: bus 0x3 vendor 0x28bd product 0x934 version 0x100
`

JoseExposito, 2022-07-12:
Hi and thanks a lot for the vendor and product IDs.

As I mentioned in the other issue, I own a XP-Pen Deco Pro SW and I'm working on supporting it. I expect the SW model to be very similar to the MW.

Once I have the code for my tablet ready I'll add yours and hopefully it'll work. It'll take me a while, but I'll ping you once it's ready.

JoseExposito, 2022-10-03:
Hi @andigandi ,

Sorry for the delay, but kernel development is slow. Linux 6.0 was just released and I had already changes queued to be included in 6.1, so I needed to wait until now to add support for the Deco Pro SW and MW.

Hopefully, it should be included in 6.2 if you could help me test it in your tablet.

Could you test these drivers?

First, download the source code: [digimend-kernel-drivers.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/9696619/digimend-kernel-drivers.zip)
The source code contains the code from DIGImend plus the changes required to add support for your tablet.

In order to install it, you'll need to install the kernel headers:

```
$ sudo apt-get install -y "linux-headers-$(uname -r)"
```

After extracting the code, cd into the source code and:

```
$ make
$ sudo make install
```

You can uninstall it running:

```
$ sudo make uninstall
```

More detailed instructions are available in the [DIGImend README file](https://github.com/DIGImend/digimend-kernel-drivers#digimend-kernel-drivers).

-----------------

The important things to test are:

 - Verify that the tablet works when connected by USB cable or USB dongle (pen, frame buttons, mouse and dial)
 - Its battery % should be displayed in both cases. In GNOME it can be checked [here](https://jeroenverhoeckx.com/media/posts/6/GNOME-Power-settings-desktop-2.png)
 - Testing on my SW I noticed that the dial did not work out of the box, so I had to add a quirk for my device. If using the dial doesn't generate scroll on your tablet I'll add it for yours as well

Thanks a lot!!

andigandi, 2022-10-14:
Hi @JoseExposito ! Today I managed to try out the driver, but sadly it does not work. Tested under Arch Linux with kernel 6.0.1-arch2-1. The only thing the device does is lighting the LEDs around the active area, if I come near with the pencil. Any ideas?

This is the kernel log after plugging in via USB:
```
usb 1-10: new full-speed USB device number 14 using xhci_hcd
usb 1-10: New USB device found, idVendor=28bd, idProduct=0934, bcdDevice= 0.00
usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-10: Product: Deco Pro MW
usb 1-10: Manufacturer: UGTABLET
usb 1-10: SerialNumber: 000000
input: UGTABLET Deco Pro MW Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:28BD:0934.0013/input/input44
uclogic 0003:28BD:0934.0013: input,hidraw5: USB HID v1.00 Mouse [UGTABLET Deco Pro MW] on usb-0000:00:14.0-10/input0
uclogic 0003:28BD:0934.0014: interface is invalid, ignoring
input: UGTABLET Deco Pro MW Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.2/0003:28BD:0934.0015/input/input45
input: UGTABLET Deco Pro MW Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.2/0003:28BD:0934.0015/input/input46
input: UGTABLET Deco Pro MW Battery as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.2/0003:28BD:0934.0015/input/input47
uclogic 0003:28BD:0934.0015: input,hiddev97,hidraw6: USB HID v1.00 Keypad [UGTABLET Deco Pro MW] on usb-0000:00:14.0-10/input2
uclogic 0003:28BD:0934.0014: interface is invalid, ignoring
```

JoseExposito, 2022-10-15:
Hi @andigandi , thanks a lot for testing the patches :D  

It is unfortunate that they don't work. Could you install [hid-tools](https://archlinux.org/packages/community/any/hid-tools/) and share recordings of the 4 interfaces available, please?

You can do it by running `sudo hid-recorder` and selecting the interfaces one after the other.

Once the recording starts:
- On the mouse interface, move the pointer around a bit
- On the pen interface, move the pointer around a bit
- On the pad interface, press all buttons and move the dial
- On the battery interface, wait a bit until some information shows up

Let's see what's different between the SW and the MW. Thanks a lot!!

andigandi, 2022-10-15:
Of course! For me, the touchpad has a separate device from the other inputs, is this intentional? And I was not quite sure if I managed to capture battery events, so I started hid-recorder again and let it settle some time.

See this file for touchpad:
[hid-record-touchpad.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9792483/hid-record-touchpad.txt)

See this file for pen, scrollwheel and buttons (and perhaps battery):
[hid-record-pen-scrollwheel-button.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9792477/hid-record-pen-scrollwheel-button.txt)

See this file for battery: 
[hid-record-battery.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9792478/hid-record-battery.txt)

Hope this helps :)

JoseExposito, 2022-10-15:
Hi @andigandi, thanks a lot for the recordings.

Did you recorded them with the driver installed? I should have mentioned that having the driver installed was important, sorry.

> For me, the touchpad has a separate device from the other inputs, is this intentional?

With the driver installed, you should see the 4 interfaces listed in your dmesg output: UGTABLET Deco Pro MW Mouse/Pen/Pad/Battery. And yes, it is intentional, libinput and other user-space apps expect multiple interfaces.

andigandi, 2022-10-15:
Hi @JoseExposito sorry for the mess, I'm totally new to the kernel stuff. Now that you asked, it might be, that some kernel update slipped through since I last tried the driver. Today I recompiled and reinstalled it and got 3 devices now. This is with kernel 6.0.1-arch2-1. Are those files better? Do I have to reboot? Really sorry, not a kernel developer here.

First the kernel log after plugging in:
```
usb 1-10: new full-speed USB device number 12 using xhci_hcd
usb 1-10: New USB device found, idVendor=28bd, idProduct=0934, bcdDevice= 0.00
usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-10: Product: Deco Pro MW
usb 1-10: Manufacturer: UGTABLET
usb 1-10: SerialNumber: 000000
input: UGTABLET Deco Pro MW Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:28BD:0934.0012/input/input37
input: UGTABLET Deco Pro MW Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:28BD:0934.0012/input/input38
hid-generic 0003:28BD:0934.0012: input,hidraw6: USB HID v1.00 Mouse [UGTABLET Deco Pro MW] on usb-0000:00:14.0-10/input0
input: UGTABLET Deco Pro MW as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.1/0003:28BD:0934.0013/input/input39
hid-generic 0003:28BD:0934.0013: input,hidraw7: USB HID v1.00 Device [UGTABLET Deco Pro MW] on usb-0000:00:14.0-10/input1
hid-generic 0003:28BD:0934.0014: hiddev97,hidraw8: USB HID v1.00 Device [UGTABLET Deco Pro MW] on usb-0000:00:14.0-10/input2
```

[hid-recorder-pen.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9793222/hid-recorder-pen.txt)

[hid-recorder-touchpad-scrollwheel-buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9793223/hid-recorder-touchpad-scrollwheel-buttons.txt)

[hid-recorder-perhaps-battery.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9793225/hid-recorder-perhaps-battery.txt)

andigandi, 2022-10-15:
After reboot:

[hid-recorder-pen.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9793241/hid-recorder-pen.txt)

[hid-recorder-touchpad-scrollwheel-buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9793242/hid-recorder-touchpad-scrollwheel-buttons.txt)

[hid-recorder-perhaps-battery.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9793244/hid-recorder-perhaps-battery.txt)

JoseExposito, 2022-10-16:
Hi @andigandi ,

Thanks for the recordings, and don't worry, these thinks are really complicated.

It looks like the kernel module didn't load. In your dmesg output, you can see that the tablet is being handled by the hid-generic driver in this line(s):

```
input: UGTABLET Deco Pro MW Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:28BD:0934.0012/input/input38
hid-generic 0003:28BD:0934.0012: input,hidraw6: USB HID v1.00 Mouse [UGTABLET Deco Pro MW] on usb-0000:00:14.0-10/input0
^ hid-generic is the name of the driver used
```

When the module is successfully loaded, it should read like you first dmesg log:

```
input: UGTABLET Deco Pro MW Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:28BD:0934.0013/input/input44
uclogic 0003:28BD:0934.0013: input,hidraw5: USB HID v1.00 Mouse [UGTABLET Deco Pro MW] on usb-0000:00:14.0-10/input0
^ Here is prints uclogic, the name of the DIGImend driver
```

So probably the best idea is to uninstall and install the module:

```
$ cd <the source code directory>
$ sudo rmmod hid_uclogic # This will unload the driver, ignore errors as it migh not be loaded yet
$ sudo make uninstall
$ make clean
$ make
$ sudo make install
```

Now, you can run in a terminal `dmesg --follow` to see the kernel log and, without rebooting your system, connect your tablet. At this point, you should hopefully see `uclogic` in the output.

Once the driver is loaded, `sudo hid-recorder` should show 2 nodes. Here is an example for my tablet:

```
$ sudo hid-recorder
[sudo] password for jose:
Available devices:
/dev/hidraw0:	HP HP Wireless Mouse 220 
/dev/hidraw1:	Logitech USB Receiver
/dev/hidraw2:	Logitech M705
/dev/hidraw3:	José Expósito’s Keyboard
/dev/hidraw4:	MSI MYSTIC LIGHT 
/dev/hidraw5:	Hanvon Ugee Deco Pro SW <-- This is the mouse
/dev/hidraw6:	Hanvon Ugee Deco Pro SW <-- This is the pen, pad and battery
```

andigandi, 2022-10-16:
Not sure what had gone wrong, but after uninstalling and building again, I am finally getting uclogic output in the kernel log:

```
usb 1-10: new full-speed USB device number 17 using xhci_hcd
usb 1-10: New USB device found, idVendor=28bd, idProduct=0934, bcdDevice= 0.00
usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-10: Product: Deco Pro MW
usb 1-10: Manufacturer: UGTABLET
usb 1-10: SerialNumber: 000000
input: UGTABLET Deco Pro MW Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:28BD:0934.0021/input/input52
input: UGTABLET Deco Pro MW Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:28BD:0934.0021/input/input53
hid-generic 0003:28BD:0934.0021: input,hidraw6: USB HID v1.00 Mouse [UGTABLET Deco Pro MW] on usb-0000:00:14.0-10/input0
input: UGTABLET Deco Pro MW as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.1/0003:28BD:0934.0022/input/input54
hid-generic 0003:28BD:0934.0022: input,hidraw7: USB HID v1.00 Device [UGTABLET Deco Pro MW] on usb-0000:00:14.0-10/input1
hid-generic 0003:28BD:0934.0023: hiddev97,hidraw8: USB HID v1.00 Device [UGTABLET Deco Pro MW] on usb-0000:00:14.0-10/input2
input: UGTABLET Deco Pro MW Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:28BD:0934.0021/input/input55
uclogic 0003:28BD:0934.0021: input,hidraw6: USB HID v1.00 Mouse [UGTABLET Deco Pro MW] on usb-0000:00:14.0-10/input0
uclogic 0003:28BD:0934.0022: interface is invalid, ignoring
input: UGTABLET Deco Pro MW Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.2/0003:28BD:0934.0023/input/input56
input: UGTABLET Deco Pro MW Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.2/0003:28BD:0934.0023/input/input57
input: UGTABLET Deco Pro MW Battery as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.2/0003:28BD:0934.0023/input/input58
uclogic 0003:28BD:0934.0023: input,hiddev97,hidraw7: USB HID v1.00 Keypad [UGTABLET Deco Pro MW] on usb-0000:00:14.0-10/input2
uclogic 0003:28BD:0934.0022: interface is invalid, ignoring
```

No idea on what went wrong, but here it might finally be:

[hid-recorder-pen-buttons-scrollwheel-perhaps-battery.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9794051/hid-recorder-pen-buttons-scrollwheel-perhaps-battery.txt)

[hid-recorder-touchpad.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9794052/hid-recorder-touchpad.txt)

JoseExposito, 2022-10-16:
Thanks for the recordings, I think that this time it worked as expected :smile: 

I'm a little bit surprised that your device doesn't work though, your output seems to be correct.

### Mouse events:

In your [hid-recorder-touchpad.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9794052/hid-recorder-touchpad.txt) I see events like:

```
# ReportID: 1 / Button: 0  0 | # | X:    -28 | Y:     36 | # 
E: 000000.000000 8 01 00 e4 ff 24 00 00 00
```

The first line is the result of parsing the binary data on the second line. As you can see, X and Y coordinates are recognized and, if you tap on the touchpad while running `sudo hid-recorder`, you should see a log like:

```
# ReportID: 1 / Button: 1  0 | # | X:      0 | Y:      0 | # 
E: 000000.816952 8 01 01 00 00 00 00 00 00
```

As you can see "Button" is "1 0". Recorded from my own tablet. The log looks identical... And in my case that moves the mouse around.

### Battery events:

In your [hid-recorder-pen-buttons-scrollwheel-perhaps-battery.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9794051/hid-recorder-pen-buttons-scrollwheel-perhaps-battery.txt) recording I see battery events:

```
# ReportID: 186 / # / 0x850065:  100 | 0x850044: 1 | # | # 
                      ^                ^
                      100%             1 means charging, connected to USB
E: 000000.000000 12 ba f2 01 64 01 00 00 00 00 00 00 00
```

In my case the battery percentage is displayed:

[image]

### Buttons:

```
# ReportID: 247 / # / Button: 0  1  0  0  0  0  0  0 , 0 , 0 | # | # | Wheel:    0 | #
                                 ^ Second button clicked
E: 000016.220153 12 f7 f0 02 00 00 00 00 00 00 00 00 00
```

### Dial:

```
# ReportID: 247 / # / Button: 0  0  0  0  0  0  0  0 , 0 , 0 | # | # | Wheel:   -1 | # 
E: 000020.018003 12 f7 f0 00 00 00 00 00 ff 00 00 00 00
```

The "Wheel:   -1"  means counter clockwise and it should scroll down.

### Pen:

Last but not least, pen events seem to be OK:

```
# ReportID: 2 / Tip Switch: 1 | Barrel Switch: 1 | Tablet Pick: 0 | # | In Range: 1 | # | X:  17597 | Y:  15949 | Tip Pressure:  5423 | # | X Tilt:    0 | Y Tilt:    0 
E: 000031.117961 12 02 a3 bd 44 4d 3e 2f 15 00 00 00 00
```

As you can see, X, Y, pressure and in range information is reported. In my case a recording like that moves the pen around

----------------------------------------------

So, from the information I can see in the recordings, it looks like everything is working as expected :facepalm: 

Does unloading and loading the module makes the tablet work? (`sudo rmmod hid_uclogic && sudo modprobe hid_uclogic`)

PS - No, there is no need to @ me, I'm subscribed to this issue.

andigandi, 2022-10-16:
That's what I thought, too, from seeing the records... Is there anything else that I might have forgotten? I am using xf86-input-libinput for handling input devices, according to the digimend documentation this should be sufficient? I tested the tablet with Krita and Xournal++, do you know anything perhaps more low-level to test this?

Unloading and loading hid_uclogic made no difference...

Just to get this right, battery shall be in the system info area, where a laptop battery also would be?

Using the touchpad shall actually move the mouse cursor?

Using the wheel shall scroll in any window?

Using the pen shall do some drawing in a drawing application? I used to have a tablet PC with an integrated digitizer that moved the mouse cursor on the desktop while using the pen, is this supposed to happen here, too?

JoseExposito, 2022-10-16:
> Just to get this right, battery shall be in the system info area, where a laptop battery also would be?

Yes, it depends a bit on your desktop environment. On GNOME it is displayed in settings:
https://jeroenverhoeckx.com/media/posts/6/GNOME-Power-settings-desktop-2.png

> Using the touchpad shall actually move the mouse cursor?

Yes, it should move the cursor.

> Using the wheel shall scroll in any window?

Yes, it should work on any window.

> Using the pen shall do some drawing in a drawing application? I used to have a tablet PC with an integrated digitizer that moved the mouse cursor on the desktop while using the pen, is this supposed to happen here, too?

Yes, moving the pen close to the tablet should move the cursor and touching the tablet should "click".

----------------------------

Anyway, we can debug the next layer of the stack and see if something is going wrong there. The next layer is libinput. libinput includes two handy tools (`libinput record` and `libinput debug-events`) that will show the output of the kernel and what is send to the Wayland compositor/X11 driver. The high level schema is something like this:

```
Kernel -> libinput record -> libinput -> libinput debug-events -> Wayland/X11 -> Apps
```

`sudo libinput record` is similar to `hid-recorder`. For reference, I attach a bit of the output of my tablet so you can see how it works:

- Mouse (moving the cursor and left click): [mouse.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9795281/mouse.txt)
- Pen interface (moving the pen): [pen.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9795274/pen.txt)
- Pad (pressing 1 button and using the dial): [pad.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9795275/pad.txt)
- Battery: There is no output, so there is no need to record it

Now that we recorded the output of the kernel, we can see what libinput sends to the next layer by running `sudo libinput debug-events --verbose`. For reference, here is the output of my tablet pressing a button and using the pen:  [debug-events.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9795292/debug-events.txt)

Let me know if you can run similar command and attach the output so we can see if something looks different. Thanks!!

andigandi, 2022-10-16:
Thanks for explaining all this! It's always cool to get some knowledge debugging your stuff :)

The results of `libinput recored` look okay to me: [libinput-record-mouse.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9795540/libinput-record-mouse.txt) [libinput-record-pad.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9795541/libinput-record-pad.txt) [libinput-record-pen.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9795542/libinput-record-pen.txt)

But unfortulately, `libinput debug-events` does not log any events, except for when I started to move my mouse: [libinput-debug-events.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9795543/libinput-debug-events.txt)

These lines look suspicious to me, the device files are listed as UGTABLET Deco Pro MW Mouse/Pen/Pad when calling libinput record, but libinput debug-events logs those errors:
```
  ... event29: device is ignored
event29 - failed to create input device '/dev/input/event29'
  ... event30: device is ignored
event30 - failed to create input device '/dev/input/event30'
  ... event31: device is ignored
event31 - failed to create input device '/dev/input/event31'
```

JoseExposito, 2022-10-17:
Aha! I think we found the issue :D All you recordings contain the `LIBINPUT_IGNORE_DEVICE` udev property:

```
  udev:
    properties:
    - ID_INPUT=1
    - ID_INPUT_MOUSE=1
    - LIBINPUT_DEVICE_GROUP=3/28bd/934:usb-0000:00:14.0-10
    - LIBINPUT_IGNORE_DEVICE=1
```

You can see the documentation for that property here: https://wayland.freedesktop.org/libinput/doc/latest/device-configuration-via-udev.html#ignoring-devices

I don't have that property in my system. Is it possible that you included it? Grepping might help you finding the config file:

```
$ grep -rnw '/etc/udev/' -e 'LIBINPUT_IGNORE_DEVICE'
```

andigandi, 2022-10-17:
OMG thank you, that was exactly the problem. That must have been a remainder from when I tested with opentabletdriver... Sorry for the mess!

I can now confirm the tablet working when connected via USB cable.

Unfortunately, It does only work partially when using the USB dongle. I thought the devices shall behave identical?

Here some kernel log:
```
usb 1-10: new full-speed USB device number 13 using xhci_hcd
usb 1-10: New USB device found, idVendor=28bd, idProduct=0934, bcdDevice= 0.00
usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-10: Product: Deco Pro MW
usb 1-10: Manufacturer: Hanvon Ugee
usb 1-10: SerialNumber: 000000
input: Hanvon Ugee Deco Pro MW Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:28BD:0934.0015/input/input46
uclogic 0003:28BD:0934.0015: input,hidraw11: USB HID v1.11 Mouse [Hanvon Ugee Deco Pro MW] on usb-0000:00:14.0-10/input0
uclogic 0003:28BD:0934.0016: interface is invalid, ignoring
input: Hanvon Ugee Deco Pro MW Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.2/0003:28BD:0934.0017/input/input47
input: Hanvon Ugee Deco Pro MW Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.2/0003:28BD:0934.0017/input/input48
input: Hanvon Ugee Deco Pro MW Battery as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.2/0003:28BD:0934.0017/input/input49
uclogic 0003:28BD:0934.0017: input,hiddev99,hidraw12: USB HID v1.11 Keypad [Hanvon Ugee Deco Pro MW] on usb-0000:00:14.0-10/input2
uclogic 0003:28BD:0934.0016: interface is invalid, ignoring
```

hid-record only shows two devices, one seems to be for battery, the other one reacty to the touchpad and the scrollwheel. I get no reaction for the buttons and the pen:
[hid-recorder-battery.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9803524/hid-recorder-battery.txt) 
[hid-recorder-touchpad-scrollwheel.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9803527/hid-recorder-touchpad-scrollwheel.txt)

In the meantime, I found a bluetooth USB stick and tested that, too. It seems like the driver is not picking the tablet up, but I am able to use it... But the quick select buttons have really weird keybindings using bluetooth. Is that the digimend driver or is that something else?
```
input: Deco Pro MW Mouse as /devices/virtual/misc/uhid/0005:28BD:0934.0018/input/input50
input: Deco Pro MW Keyboard as /devices/virtual/misc/uhid/0005:28BD:0934.0018/input/input51
input: Deco Pro MW Stylus as /devices/virtual/misc/uhid/0005:28BD:0934.0018/input/input52
hid-generic 0005:28BD:0934.0018: input,hidraw11: BLUETOOTH HID v0.01 Mouse [Deco Pro MW] on xx:xx:xx:xx:xx:xx
```
[hid-recorder-bt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9803701/hid-recorder-bt.txt)

JoseExposito, 2022-10-18:
Cool! I'm glad we finally found the issue :partying_face: 

> I can now confirm the tablet working when connected via USB cable.
>
> Unfortunately, It does only work partially when using the USB dongle.

Yes, that's something we noticed testing this other tablet:
https://github.com/DIGImend/digimend-kernel-drivers/issues/635#issuecomment-1276482316

We observed that switching on the tablet and then connecting the dongle works. However, switching the tablet is off with the dongle is connected and then switching it on doesn't work.

I'm working on this bit at the moment. Now that we now that your device works I'll send you a new version of the driver once it is ready.

> In the meantime, I found a bluetooth USB stick and tested that, too. It seems like the driver is not picking the tablet up, but I am able to use it... But the quick select buttons have really weird keybindings using bluetooth. Is that the digimend driver or is that something else?

Interesting, I didn't managed to connect my tablet via USB. There was a long list of Bluetooth devices listed (from my neighbours) and I wasn't sure if my tablet was one of them because it didn't displayed its name. Did your tablet exposed its name?

I might need to upgrade my distro, I think I'm using an old bluetooth stack.

In theory, supporting bluetooth devices is pretty straightforward, I'll look into it.

-----------------------

By the way, since you are helping so much testing the patches, would you like me to add your name and email in a Tested-by tag in the patches?
https://www.kernel.org/doc/html/latest/process/submitting-patches.html#using-reported-by-tested-by-reviewed-by-suggested-by-and-fixes

I'd need your name and an email address.

andigandi, 2022-10-21:
> Cool! I'm glad we finally found the issue partying_face
> 
> > I can now confirm the tablet working when connected via USB cable.
> > Unfortunately, It does only work partially when using the USB dongle.
> 
> Yes, that's something we noticed testing this other tablet: [#635 (comment)](https://github.com/DIGImend/digimend-kernel-drivers/issues/635#issuecomment-1276482316)
> 
> We observed that switching on the tablet and then connecting the dongle works. However, switching the tablet is off with the dongle is connected and then switching it on doesn't work.
> 
> I'm working on this bit at the moment. Now that we now that your device works I'll send you a new version of the driver once it is ready.
I noticed the switching on oddity, too. If I can provide you with any data, just tell! :) 

> Interesting, I didn't managed to connect my tablet via USB. There was a long list of Bluetooth devices listed (from my neighbours) and I wasn't sure if my tablet was one of them because it didn't [...]

The Bluetooth device showed as this and was (as far as I tested it) fully functional. Is it using the digimend driver via bluetooth? Because I did not see it in the kernel messages.
[image]

> By the way, since you are helping so much testing the patches, would you like me to add your name and email in a Tested-by tag in the patches? [...]
If that's the way to go: Andreas Grosse <andig.mail@t-online.de>
Thanks for mentioning :)

JoseExposito, 2022-12-11:
Hi again @andigandi,

I've been busy, but I finally found some time to finish the support for your tablet. I plan to send this patches to the upstream kernel on January 2023, when the development cycle of Linux 6.2 starts.

Here are the drivers:  [digimend-kernel-drivers.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/10202006/digimend-kernel-drivers.zip)

Please uninstall the previous version and install this one.

This new version handles reconnections when the USB dongle use used. Meaning that, if you turn off and on you tablet using the switch placed under the frame, or connect the USB dongle with the tablet switched on, it should reconnect automatically without issues.

Also, I added Bluetooth support. I updated my PC to Fedora 37 and I'm still not able to connect my tablet by Bluetooth... But since you were able to do it, your tablet should be fully supported in this mode.

So, can I ask you to...

- Share the output of `sudo libinput record` (and select one of your tablet's interfaces) when connected by Bluetooth, please? I'd like to make sure that the vendor/device ID does not change
- Run `dmesg` right after connecting your device. You should see a log like `###### DIGIMend test: Frame type = ?`. I'd need to know that number. My tablet reports an incorrect frame type (it claims to have no mouse) I wonder if yours has the same issue.

Thanks!!

JoseExposito, 2022-12-26:
For reference, this is fixed upstream: https://lore.kernel.org/linux-input/20221226125454.16106-1-jose.exposito89@gmail.com/T/

andigandi, 2023-01-04:
Sorry for the late reply! Being fixed upstream does mean, you no longer need the outputs requested in https://github.com/DIGImend/digimend-kernel-drivers/issues/622#issuecomment-1345526180 ?

JoseExposito, 2023-01-05:
Yes, they are no longer necessary. I added an "Edit" note explaining why. Unfortunately Bluetooth support would require quite a lot of work, so we will have to use USB for the moment.

Thanks a lot for helping me out with testing!!

RA-Kooi, 2023-12-06:
I've been following along with this discussion hoping to find an answer to my problems with this tablet, but so far I haven't been able to get it to work in Xorg. The tablet is recognized and all the functions work through `evtest`.  
However when I bind it to Xorg with the following snippet:
```xorg.conf
Section "InputClass"
	Identifier "XP-Pen Deco Pro MW"
	Driver "wacom"
	MatchDevicePath "/dev/input/event*"
	MatchUSBID "28bd:0934"
EndSection
```
It allows it to show up in `xsetwacom`, for which it shows the `Mouse`, `stylus`, and `eraser`.  
However the "Mouse" is ignored, which I assume to be the `Hanvon Ugee Deco Pro MW Mouse` as detected by the kernel. Using `evtest` on that device detects both the dial and the virtual wheel, the dial is detected as scroll and the wheel is detected as mouse movement. Trying to use the virtual wheel spams errors in `Xorg.0.log` with something along the lines of `wacom: rel event recv'd (0)!`. I can get the actual errors if need be later if you need them.  
The stylus and eraser function work as expected and can be configured through `xsetwacom`.  
Missing here is the keyboard which is also ignored by Xorg, meaning the buttons don't do anything and I can't rebind them. Same here, `evtest` on the input device detects all input.

`libinput list-devices` detects the following devices:
* `Hanvon Ugee Deco Pro MW Keyboard`
* `Hanvon Ugee Deco Pro MW`

While also throwing an error for the "Mouse" component: `event24 - Hanvon Ugee Deco Pro MW Mouse: libinput bug: missing tablet capabilities: pen btn-stylus resolution. Ignoring this device.`. Though unlike https://github.com/DIGImend/digimend-kernel-drivers/issues/275#issuecomment-778644421 I don't recall seeing these anywhere in my `Xorg.0.log`. Again, I can get actual logs for this if you need me to.

Somewhere here in the issues I found a reference to the [Userspace Tablet Driver Daemon](https://github.com/kurikaesu/userspace-tablet-driver-daemon) repo, which is what I'm using right now. This seems to support all the tablet features offered by the proprietary driver except for battery level reporting.

As for my kernel, I am running kernel 6.5.9 on Arch Linux. If there's anything else you need to know, let me know.

> Adding Bluetooth support is going to be really complicated, so we will have to use the USB dongle of cable for the moment :(

This has been fixed, as I tried it both over cable and bluetooth and there's no difference in my testing experience.

ChaseStorm, 2025-04-21:
It doesn't work :(