DIGImend/digimend-kernel-drivers #625: Support for Gaomon 1060pro digital boards

shengliawa, 2022-07-24:
It just like Huion New 1060 Plus digital board.Here is some picture to compare.
[image]
[image]
They have different USBID,  **Huion New 1060 Plus digital board** is _256c:006e_,
**Gaomon 1060pro digital boards** is _256c:0064_.
Starting from the upper left corner, four corners are pointed clockwise
command `sudo usbhid-dump -es -m 256c:0064 | tee frame_wheel_srolling.txt`
output is 
[log.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/9175507/log.txt)

shengliawa, 2022-07-24:
Use gitee repo https://gitee.com/parkerace/digimend-kernel-drivers .
But there is a problem with scaling. My screen is within this range:
[image]

pyfile, 2022-10-27:
I have a problem of my Gaomon 1060pro with digimend. On my Opensuse Tumbleweed with kernel 6.0.2-1-default, Gaomon 1060pro digital board is 256c:006e but not 0064 you said. After 'sudo make install' without err output and reboot, my board still can't work.