DIGImend/digimend-kernel-drivers #628: Kamvas Pro 16 doesn't get detected on Plasma

NyaomiDEV, 2022-08-05:
Dmesg log:
```
[16033.557922] usb 5-3: new full-speed USB device number 18 using xhci_hcd
[16033.748825] usb 5-3: New USB device found, idVendor=256c, idProduct=006d, bcdDevice= 0.00
[16033.748829] usb 5-3: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[16033.772900] input: HID 256c:006d as /devices/pci0000:00/0000:00:08.1/0000:10:00.3/usb5/5-3/5-3:1.0/0003:256C:006D.0020/input/input72
[16033.772967] input: HID 256c:006d Pad as /devices/pci0000:00/0000:00:08.1/0000:10:00.3/usb5/5-3/5-3:1.0/0003:256C:006D.0020/input/input73
[16033.773011] input: HID 256c:006d Touch Strip as /devices/pci0000:00/0000:00:08.1/0000:10:00.3/usb5/5-3/5-3:1.0/0003:256C:006D.0020/input/input74
[16033.773053] input: HID 256c:006d Dial as /devices/pci0000:00/0000:00:08.1/0000:10:00.3/usb5/5-3/5-3:1.0/0003:256C:006D.0020/input/input75
[16033.773126] uclogic 0003:256C:006D.0020: input,hiddev98,hidraw7: USB HID v1.11 Keypad [HID 256c:006d] on usb-0000:10:00.3-3/input0
[16033.779837] uclogic 0003:256C:006D.0021: No inputs registered, leaving
[16033.779864] uclogic 0003:256C:006D.0021: hidraw8: USB HID v1.11 Device [HID 256c:006d] on usb-0000:10:00.3-3/input1
```

I am using Plasma on Wayland and Artix Linux. Compiled DIGImend from the `master` branch.

Deevad, 2023-01-23:
Hey @NyaomiDEV ,
You are the only user of the Kamvas Pro 16 on GNU/Linux I could find. Can you confirm it is this model: https://www.huion.com/pen_display/KamvasPro/kamvas-pro-16-4k.html ? 
Have you found a way to get it working with Digimend? 

 (I have my eyes on it, but unfortunately I don't want to risk buying hardware with only a proprietary driver).

NyaomiDEV, 2023-01-23:
@Deevad 
Nope, mine's the 1080p version.

I stopped using Linux completely just because there was nothing I can do to produce a decent experience out of it. DIGImend, OTD and Huion's proprietary drivers all produce a really subpar experience after extensive tinkering, with touch strip and buttons not working or doing strange stuff; in some cases you can't even calibrate the display, and calibration seems to be required at every PC restart if you use DIGImend.

If you want to draw on Linux, the best bet is Wacom. Not even Huion can produce decent drivers, anyway.

Deevad, 2023-01-23:
Hey, thanks for the feedback, your words of caution, and I totally understand your situation and feedback. 

Yes, the Linux FOSS driver like this one of Digimend is dormant for non-Wacom, (and X11 only, the Wayland situation is even worst) and the proprietary drivers made in-house by all the brand of tablets (XP-Pen, Huion, etc) looks like made by the same third party sub-company; often the same structure in the package and often are bad quality; [I tested it on a XPPen video review](https://www.davidrevoy.com/article842/review-of-the-xp-pen-artist-24-pro-on-linux). 

I'll probably keep all non Wacom brands far from my usage if I want peace of mind on GNU/Linux.

Thanks again for your quick reply, and sorry for waking your thread for a non-solution.

NyaomiDEV, 2023-01-23:
@Deevad

By the way, the best place to contribute to is OpenTabletDriver. If you happen to have unsupported tablets and features, that is. They also already have a "Linux Artist Mode" ready for use. I think it's worth a shot even if you already have working Wacom stuff.

Also, if you're on GNOME Wayland, their experience is guaranteed to work.

Deevad, 2023-01-23:
Thanks! I'll study it. I paste here the link to the repo to ease redirect future readers of this discussion. https://github.com/OpenTabletDriver/OpenTabletDriver