DIGImend/digimend-kernel-drivers #630: Problem with tablet huion h640P + Fedora

BGPavelAng, 2022-08-10:
Hi.

The pen doesn't work if you don't touch the tablet with the pen all time.

when you have a moment, can you try to fix it?, please... I saw some posts talking about that problem on Fedora 36 because on Kali and maybe Ubuntu works fine.

Thank you for the driver.

Murilovsky, 2022-08-16:
Having the same issue. Fedora 36 Wayland(Xorg didn't work as well)
kernel 5.18.17

The pen just works the first time it comes close to the tablet. If it gets away once, the only way to make it work again is unpluging the usb.

I did run `sudo usbhid-dump -es -m 256c:006e` and it seems to be recognized,but still doesn't work
[image]

WhiteCat6142, 2022-09-02:
I changed /usr/share/X11/xorg.conf.d/50-digimend.conf

```
Section "InputClass"
        Identifier "Huion tablets with Wacom driver"
        MatchUSBID "5543:006e|256c:006e|256c:006d"
        MatchDevicePath "/dev/input/event*"
        MatchIsTablet "True"
        Driver "libinput"
        Option "Suppress" "0"
EndSection
```
and run the following on tty3

```sh
sudo udevadm hwdb --update  
```
that works.
I'm not sure why.

WhiteCat6142, 2022-09-03:
I find why I needed to run on tty3.
I touch the pad, and then execute

```
sudo udevadm hwdb --update
```

and switch tty and back.
That was fixed.

WhiteCat6142, 2022-09-03:
Surprisely, I found the latest kernel of Fedora recognized my tablet without installing digimend drivers.
There are "DIGImend patches" on kernel mainline.

https://patchwork.kernel.org/project/linux-input/list/?series=632252&state=*&archive=both

https://patchwork.kernel.org/project/linux-input/list/?series=613192&state=*

zastrixarundell, 2024-06-22:
Weirdly currently on fedora 40 the tablet doesn't seem to be recognized.

tankersss, 2024-07-10:
> Weirdly currently on fedora 40 the tablet doesn't seem to be recognized.

I'm running 40 kde, no problem here, but it really depends on HW ID version you have from what I read on other issues.
HID 256c:006d is the one that is working for me (2020 model).