DIGImend/digimend-kernel-drivers #632: Huion H640s Physical usable area is not correct, HUION H640p

agoogleuser, 2022-08-14:
I've installed the drivers on my Fedora KDE 36, but when I'm using the tablet I can only use the left square portion, while a huge area is unusable or detectable as as the edge of the tablet, I've tried fiddling with the command
```bash
xsetwacom --set 15 area 0 0 32767 32767
```
The above value is the default, I've tried changing the X-axis but it didn't work well, I also tried to play with KDE's settings, It recognized it as a HUION H950p, and fiddling with the settings doesn't do much, and have no idea how to fix this.

More info:
```bash
$ xsetwacom list
HUION Huion Tablet_H640P Stylus stylus  id: 15  type: STYLUS    
HUION Huion Tablet_H640P Stylus eraser  id: 22  type: ERASER 
```
The tablet ID from `lsusb` command is shown as 
```
Bus 001 Device 009: ID 256c:006d HUION Huion Tablet_H640P
```

The pen I use doesn't have an Eraser.
I'm on kernel 5.18.13-200 and on X11

phispi, 2025-12-26:
There are some interesting comments regarding the same problem in #308. It mentions that the tablet supports two modes: The PC mode where the full area is used and the Android mode where only part of the area is used. So one explanation is that the tablet might be in Android mode while it should be in PC mode. The most recent manual says that the tablet detects automatically which system it is connected to. Pressing button 1 and 5 simultaneously for 5 seconds like proposed in #308 doesn't change anything on my system.

The output of `xsetwacom --get 18 Area` is `0 0 32767 32767` on my system while - according to #308 - it should be `H950P 0 0 44200 27600`.

phispi, 2025-12-26:
Just to ensure that the firmware of the tablet is not causing the problem I updated the firmware with the Huion Firmware Tool from version 200718 to 240919. And: voila. The full area is used :-)
Still interesting why it used to work previously with the old firmware as well.

If this fixes the problem for @agoogleuser as well, this ticket can be closed.