DIGImend/digimend-kernel-drivers #634: dkms_uninstall from 10 nor 11 not working (??)

msmafra, 2022-09-02:
Hello there. 
I was trying to remove and reinstall Digimend 10 but was not able to remove nor reinstall it. Thinking it could be a problem with my version saw that a new version (11) was available. Trying to uninstall and install with 11 also not successful.
I'm on Fedora 36 Workstation. I have both normal and fsync kernels installed.

```
❯ doas make dkms_install
DKMS has DIGImend modules added already, aborting.
Run "make dkms_uninstall" first
make: *** [Makefile:99: dkms_modules_install] Error 1
❯ doas make dkms_uninstall
rm -vf /usr/sbin/digimend-debug
rm -vf /usr/share/X11/xorg.conf.d/50-digimend.conf
rm -vf /lib/udev/rules.d/90-digimend.rules /lib/udev/hid-rebind
udevadm control --reload
set -e -x; \
if test -e /usr/lib/dracut/dracut.conf.d/90-digimend.conf; then \
    rm -v /usr/lib/dracut/dracut.conf.d/90-digimend.conf; \
    dracut --force; \
fi
+ test -e /usr/lib/dracut/dracut.conf.d/90-digimend.conf
rm -vf /etc/depmod.d/digimend.conf
depmod -a
set -e -x; \
dkms status digimend | \
    while IFS=':' read -r modules status; do \
        echo "$modules" | { \
            IFS=',/ ' read -r modules_name modules_version \
                              kernel_version kernel_arch ignore; \
            if [ -z "$kernel_version" ]; then \
                dkms remove \
                            "$modules_name/$modules_version" \
                            --all; \
            else \
                dkms remove \
                            "$modules_name/$modules_version" \
                            -k "$kernel_version/$kernel_arch"; \
            fi; \
        } \
    done
+ dkms status digimend
+ IFS=:
+ read -r modules status
+ echo 'digimend/10, 5.19.4-201.fsync.fc36.x86_64, x86_64'
+ IFS=',/ '
+ read -r modules_name modules_version kernel_version kernel_arch ignore
+ '[' -z 5.19.4-201.fsync.fc36.x86_64 ']'
+ dkms remove digimend/10 -k 5.19.4-201.fsync.fc36.x86_64/x86_64
Module digimend-10 for kernel 5.19.4-201.fsync.fc36.x86_64 (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

hid-kye.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/5.19.4-201.fsync.fc36.x86_64/extra/
 - Original module
   - Archived original module found in the DKMS tree
   - Moving it to: /lib/modules/5.19.4-201.fsync.fc36.x86_64/extra/

hid-uclogic.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/5.19.4-201.fsync.fc36.x86_64/extra/
 - Original module
   - Archived original module found in the DKMS tree
   - Moving it to: /lib/modules/5.19.4-201.fsync.fc36.x86_64/extra/

hid-polostar.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/5.19.4-201.fsync.fc36.x86_64/extra/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

hid-viewsonic.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/5.19.4-201.fsync.fc36.x86_64/extra/
 - Original module
   - Archived original module found in the DKMS tree
   - Moving it to: /lib/modules/5.19.4-201.fsync.fc36.x86_64/extra/
depmod.....

Removing original_module from DKMS tree for kernel 5.19.4-201.fsync.fc36.x86_64 (x86_64)
+ IFS=:
+ read -r modules status
❯ doas make dkms_install
DKMS has DIGImend modules added already, aborting.
Run "make dkms_uninstall" first
make: *** [Makefile:99: dkms_modules_install] Error 1

```

msmafra, 2022-09-02:
Manually running  doas dkms remove digimend/10 --all and doas rm -rf /var/lib/dkms/digimend made the installation work for me.
But running doas dracut --force --regenerate-all --verbose I get this message with all kernels:
```
dracut-install: Failed to find module 'hid-polostar'
dracut: FAILED:  /usr/lib/dracut/dracut-install -D /var/tmp/dracut.eKJI3k/initramfs -N nvidia|nvidia_drm|nvidia-modeset|nvidia-uvm --kerneldir /lib/modules/5.19.6-200.fc36.x86_64/ -m hid_kye hid-polostar hid-uclogic hid-viewsonic
dracut: *** Installing kernel module dependencies ***
```

WhiteCat6142, 2022-09-03:
I broke my boot files. But that worked on another machine.
```
sudo dkms uninstall digimend/10
sudo make dkms_uninstall
git checkout v11
sudo make dkms_install
```

WhiteCat6142, 2022-09-20:
I didn't broke boot files, but broke services.
I temporarily remove dracut-* and dkms services, that was fixed from systemd emergency mode.