DIGImend/digimend-kernel-drivers #635: XP-Pen Deco LW module sometimes sends bogus pointer position

kanashimia, 2022-09-20:
Sometimes module sends bogus events for XP-Pen Deco LW, more precisely it forces cursor to be in the middle top part of the screen.
When you draw sometimes cursor jumps borking everything, the cursor jumps every time after lifting a pen.

This doesn't happen without digimend kernel modules on linux 5.19.8
This happens without digimend kernel modules on linux 6.0-rc5, should be fixed before the release if it is possible.

Excerpt from `evtest`:
```
Event: time 1663705880.465158, type 3 (EV_ABS), code 0 (ABS_X), value 25601
Event: time 1663705880.465158, type 3 (EV_ABS), code 1 (ABS_Y), value 1
```

I think this is caused by this packet, excerpt from `usbhid-dump --entity=stream --model=28bd`: 
```
001:003:002:STREAM             1663707385.352695
 02 F2 01 64 01 00 00 00 00 00
```

Deco L/M/MW probably all have same problem.

CC: @JoseExposito

kanashimia, 2022-09-22:
Iterestingly windows driver seems to send a similar packet too:

Ok, so while using a BT dongle it first sends packets that look like this:
`02 F2 01 63 00 00 00 00 00 00 00 00`

While plugged in by USB it sends this:
`02 F2 01 63 01 00 00 00 00 00`

So that fifth byte indicates whether the tablet is plugged in or not.

This fourth one is probably battery status or something stupid like that.
While using tablet wireless for some time it now sends this:
`02 F2 01 50 00 00 00 00 00 00 00 00`
So it is probably battery status indeed, needs more checking.

With wireshark i was able to see that this packet is being sent constanty, at 5 second intervals.

I can't seem to sniff traffic that xp-pen proprietary driver for linux does, sadly, usbhid-dump says that device is unavailable, and i don't have a hardware debugger.

JoseExposito, 2022-09-22:
Hi!

Indeed, that package is the battery status of the tablet. 

As you mentioned, the 4rd byte is the batter percentage and the 5th byte is the battery status (charging/discharging).

I'm adding support for the Deco SW and I plan to have it finished for 6.1.

What it is not expected is your tablet to be handled in 6.0-rc5 though. Do you mind to share the output of `lsusb`, please?

I'd need to know you tablet's vendor and product IDs. 6.0-rc5 should include support for the Deco L and the Deco S, but not for the wireless versions of the tablets... Unless they reused the same IDs (sigh).

kanashimia, 2022-09-22:
> Unless they reused the same IDs (sigh).

Yeah... They have reused same IDs, probably should have noted from the beginning.

<details>
<summary>SPOILER: Output of the lsusb</summary>

```
!lsusb -vd 28bd:0935          

Bus 001 Device 002: ID 28bd:0935 XP-Pen Deco LW
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x28bd XP-Pen
  idProduct          0x0935 
  bcdDevice            0.00
  iManufacturer           1 Hanvon Ugee
  iProduct                2 Deco LW
  iSerial                 3 000000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x005b
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          4 ACD02-C_V1.1.2
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     177
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     109
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      36
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0001
  Self Powered
```

</details>

JoseExposito, 2022-09-22:
Damn, that the same ID of the Deco L! That's going to make it really difficult to differentiate between both tablets.

Since you were using WireShark, could you share a recording of the tablet connection, please? There is no need to press any button or use the pen, just connecting the tablet to a usb port.

The driver gets a string descriptor from the tablet containing useful information like the number of buttons, resolution, etc

I didn't manage to find a byte indicating whether the tablet has battery or not, but let's see if we can find it...

kanashimia, 2022-09-23:
Ok here are the `pcapng` files and the exported packet data.
I provided files for both usb cable and bt usb dongle. 

[bt dump](https://redpilled.dev/dump/bt_dump_1.pcapng)
[usb dump](https://redpilled.dev/dump/usb_dump_1.pcapng)

[bt dump plaintext data export](https://redpilled.dev/dump/export_plain_bt_1.txt)
[usb dump plaintext data export](https://redpilled.dev/dump/export_plain_usb_1.txt)

JoseExposito, 2022-09-27:
Thanks for the recordings @kanashimia , they are really useful.

The packet that I use to get the tablet features seems to be identical in the Deco L and Deco LW... So it won't help us.

I'll need to find a different way to detect whether the tablet supports battery reporting or not. Thankfully, I'm working on support for the Deco SW and MW:
https://github.com/DIGImend/digimend-kernel-drivers/issues/622

So I'll try to find a method that works for the 3 tablets... I'll ping you when I made some progress.

JoseExposito, 2022-10-03:
Hi again @kanashimia,

I've working on this and I have the initial version of the patches almost ready. However, I wonder if you could share the output of `libinput record` to check the device version.

On the Deco L the output is:

```
$ sudo libinput record
[...]
/dev/input/event21:	Hanvon Ugee Technology Co.,Ltd Deco L Mouse
/dev/input/event22:	Hanvon Ugee Technology Co.,Ltd Deco L Pen
/dev/input/event23:	Hanvon Ugee Technology Co.,Ltd Deco L Pad
Select the device event number: 23
# libinput record
[...]
devices:
- node: /dev/input/event23
  evdev:
    # Name: Hanvon Ugee Technology Co.,Ltd Deco L Pad
    # ID: bus 0x3 vendor 0x28bd product 0x935 version 0x111
[...]
```

The relevant information is the "version" field. I'm trying to see if it is different between the Deco L and the Deco LW to add or not battery reporting.

Thanks!

kanashimia, 2022-10-03:
Hello, it seems version ID is the same.

<details>
<summary>output of libinput record</summary>

```
!libinput record        
Available devices:
/dev/input/event0:	AT Translated Set 2 keyboard
/dev/input/event1:	joric Jorne Keyboard
/dev/input/event2:	joric Jorne Keyboard System Control
/dev/input/event3:	joric Jorne Keyboard Consumer Control
/dev/input/event4:	joric Jorne Keyboard
/dev/input/event5:	Lid Switch
/dev/input/event6:	Power Button
/dev/input/event7:	Power Button
/dev/input/event8:	Wireless hotkeys
/dev/input/event9:	HP Wide Vision HD Camera: HP Wi
/dev/input/event10:	ST LIS3LV02DL Accelerometer
/dev/input/event11:	Hanvon Ugee Technology Co.,Ltd Deco LW
/dev/input/event12:	Hanvon Ugee Technology Co.,Ltd Deco LW Pad
/dev/input/event13:	HDA NVidia HDMI/DP,pcm=3
/dev/input/event14:	HDA NVidia HDMI/DP,pcm=7
/dev/input/event15:	HDA NVidia HDMI/DP,pcm=8
/dev/input/event16:	HDA NVidia HDMI/DP,pcm=9
/dev/input/event17:	ETPS/2 Elantech Touchpad
/dev/input/event18:	HP WMI hotkeys
/dev/input/event19:	Video Bus
/dev/input/event20:	Video Bus
/dev/input/event21:	HDA Intel PCH Mic
/dev/input/event22:	HDA Intel PCH Headphone
/dev/input/event23:	HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event24:	HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event25:	HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event26:	HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event27:	HDA Intel PCH HDMI/DP,pcm=10
Select the device event number: 12
Recording to 'stdout'.
# libinput record
version: 1
ndevices: 1
libinput:
  version: "1.21.0"
  git: "unknown"
system:
  os: "nixos:22.11"
  kernel: "5.19.12"
  dmi: "dmi:bvnInsyde:bvrF.26:bd08/10/2021:br15.26:efr70.30:svnHP:pnHP15-cx0041ur:pvrType1ProductConfigId:rvnHP:rn8478:rvr70.30:cvnHP:ct10:cvrChassisVersion:sku4PP88EA#ACB:"
devices:
- node: /dev/input/event12
  evdev:
    # Name: Hanvon Ugee Technology Co.,Ltd Deco LW Pad
    # ID: bus 0x3 vendor 0x28bd product 0x935 version 0x111
    # Supported Events:
    # Event type 0 (EV_SYN)
    # Event type 1 (EV_KEY)
    #   Event code 256 (BTN_0)
    #   Event code 257 (BTN_1)
    #   Event code 258 (BTN_2)
    #   Event code 259 (BTN_3)
    #   Event code 260 (BTN_4)
    #   Event code 261 (BTN_5)
    #   Event code 262 (BTN_6)
    #   Event code 263 (BTN_7)
    # Event type 4 (EV_MSC)
    #   Event code 4 (MSC_SCAN)
    # Properties:
    name: "Hanvon Ugee Technology Co.,Ltd Deco LW Pad"
    id: [3, 10429, 2357, 273]
    codes:
[... 135 lines cut ...]
  hid: [
    0x05, 0x0d, 0x09, 0x01, 0xa1, 0x01, 0x85, 0x02, 0x09, 0x20, 0xa1, 0x00, 0x09, 0x42, 0x09, 0x44, 
    0x09, 0x46, 0x75, 0x01, 0x95, 0x03, 0x14, 0x25, 0x01, 0x81, 0x02, 0x95, 0x02, 0x81, 0x03, 0x09, 
    0x32, 0x95, 0x01, 0x81, 0x02, 0x95, 0x02, 0x81, 0x03, 0x75, 0x10, 0x95, 0x01, 0x35, 0x00, 0xa4, 
    0x05, 0x01, 0x09, 0x30, 0x65, 0x13, 0x55, 0x0d, 0x27, 0x70, 0xc6, 0x00, 0x00, 0x47, 0x10, 0x27, 
    0x00, 0x00, 0x81, 0x02, 0x09, 0x31, 0x27, 0x10, 0x77, 0x00, 0x00, 0x47, 0x70, 0x17, 0x00, 0x00, 
    0x81, 0x02, 0xb4, 0x09, 0x30, 0x45, 0x00, 0x27, 0xff, 0x1f, 0x00, 0x00, 0x75, 0x0d, 0x95, 0x01, 
    0x81, 0x02, 0x75, 0x01, 0x95, 0x03, 0x81, 0x01, 0x09, 0x3d, 0x35, 0xc3, 0x45, 0x3c, 0x15, 0xc3, 
    0x25, 0x3c, 0x75, 0x08, 0x95, 0x01, 0x81, 0x02, 0x09, 0x3e, 0x35, 0xc3, 0x45, 0x3c, 0x15, 0xc3, 
    0x25, 0x3c, 0x81, 0x02, 0xc0, 0xc0, 0x05, 0x01, 0x09, 0x07, 0xa1, 0x01, 0x85, 0xf7, 0x05, 0x0d, 
    0x09, 0x39, 0xa0, 0x75, 0x01, 0x95, 0x08, 0x81, 0x01, 0x05, 0x09, 0x19, 0x01, 0x2a, 0x08, 0x00, 
    0x95, 0x0a, 0x14, 0x25, 0x01, 0x81, 0x02, 0x95, 0x46, 0x81, 0x01, 0xc0, 0xc0
  ]
  udev:
    properties:
    - ID_INPUT=1
    - ID_INPUT_HEIGHT_MM=152
    - ID_INPUT_TABLET=1
    - ID_INPUT_WIDTH_MM=254
  quirks:
  events:
  # Current time is 22:00:29
  ```
  
  </details>

JoseExposito, 2022-10-09:
Hi @kanashimia 

Thanks for the information!

I generated a bundle with the required patches to support the Deco LW: [digimend-kernel-drivers.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/9741386/digimend-kernel-drivers.zip)

The source code contains the changes described in [this comment](https://github.com/DIGImend/digimend-kernel-drivers/issues/622#issuecomment-1265234212) plus a change on top to support the Deco LW battery reporting. The zip file contains the git history, just in case you want to see what's going on.

Hopefully, on tablet connection, you should be able to see its battery level and the annoying pointer jumps should be gone.

Check [the comment](https://github.com/DIGImend/digimend-kernel-drivers/issues/622#issuecomment-1265234212) for install/uninstall instructions.

> Also i found two packets when tablet is powered off and on while dongle is connected:
> It seems those indicate that connection is lost/found.

Thanks for the heads up, I didn't test this case on my Deco Pro SW, but I'll test it and see if I can fix it.

I'll post here as soon as I make some progress on device reconnection.

kanashimia, 2022-10-09:
Hello @JoseExposito. 

So i've tested your changes and now cursor doesen't jump and pad reports correctly.

Battery reporting still needs work.
`acpi` constantly shows this:
```
›acpi
Battery 0: Discharging, 0%, rate information unavailable
```
You can see than supply capacity is actually set correctly, but status isn't.
Something else should also be needed for `acpi` to report percentage correctly.  
```
›cat /sys/class/power_supply/hid-28bd-935-battery/uevent                              
POWER_SUPPLY_NAME=hid-28bd-935-battery
POWER_SUPPLY_TYPE=Battery
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_ONLINE=1
POWER_SUPPLY_CAPACITY=100
POWER_SUPPLY_MODEL_NAME=Hanvon Ugee Technology Co.,Ltd Deco LW
POWER_SUPPLY_STATUS=Discharging
POWER_SUPPLY_SCOPE=Device
```

---

On the other hand i've found that upon connecting a tablet `dmesg` shows this:
```
[  137.090296] usb 1-1: new full-speed USB device number 5 using xhci_hcd
[  137.220423] usb 1-1: New USB device found, idVendor=28bd, idProduct=0935, bcdDevice= 0.00
[  137.220426] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  137.220428] usb 1-1: Product: Deco LW
[  137.220429] usb 1-1: Manufacturer: Hanvon Ugee Technology Co.,Ltd
[  137.220430] usb 1-1: SerialNumber: 0000000000
[  137.224659] input: Hanvon Ugee Technology Co.,Ltd Deco LW Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:28BD:0935.0004/input/input30
[  137.276506] uclogic 0003:28BD:0935.0004: input,hidraw0: USB HID v1.11 Mouse [Hanvon Ugee Technology Co.,Ltd Deco LW] on usb-0000:00:14.0-1/input0
[  137.277438] uclogic 0003:28BD:0935.0005: interface is invalid, ignoring
[  137.278873] uclogic 0003:28BD:0935.0006: Deco LW
[  137.279035] input: Hanvon Ugee Technology Co.,Ltd Deco LW Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:28BD:0935.0006/input/input31
[  137.279231] input: Hanvon Ugee Technology Co.,Ltd Deco LW Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:28BD:0935.0006/input/input32
[  137.279322] input: Hanvon Ugee Technology Co.,Ltd Deco LW Battery as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:28BD:0935.0006/input/input33
[  137.279533] uclogic 0003:28BD:0935.0006: input,hiddev96,hidraw1: USB HID v1.11 Keypad [Hanvon Ugee Technology Co.,Ltd Deco LW] on usb-0000:00:14.0-1/input2
```
Notice the `Product: Deco LW` there, with this info i was able to slightly improve your tablet matching:

```diff
diff --git a/hid-uclogic-params.c b/hid-uclogic-params.c
index 49eb134..c655929 100644
--- a/hid-uclogic-params.c
+++ b/hid-uclogic-params.c
@@ -1245,6 +1245,7 @@ static int uclogic_params_ugee_v2_init_frame_mouse(struct uclogic_params *p)
 static bool uclogic_params_ugee_v2_has_battery(struct hid_device *hdev)
 {
 	struct uclogic_drvdata *drvdata = hid_get_drvdata(hdev);
+	struct usb_device *udev = hid_to_usb_dev(hdev);
 
 	/* The XP-PEN Deco LW vendor, product and version are identical to the
 	 * Deco L. The only difference reported by their firmware is the device
@@ -1253,11 +1254,9 @@ static bool uclogic_params_ugee_v2_has_battery(struct hid_device *hdev)
 	 */
 	if (hdev->vendor == USB_VENDOR_ID_UGEE &&
 	    hdev->product == USB_DEVICE_ID_UGEE_XPPEN_TABLET_DECO_L) {
-		const char *dongle_name = "Hanvon Ugee Deco LW";
-		const char *usb_name = "Hanvon Ugee Technology Co.,Ltd Deco LW";
+		const char *product_name = "Deco LW";
 
-		if (strncmp(hdev->name, dongle_name, strlen(dongle_name)) == 0 ||
-		    strncmp(hdev->name, usb_name, strlen(usb_name)) == 0)
+		if (strncmp(udev->product, product_name, strlen(product_name)) == 0)
 			return true;
 	}
 
 ```
 Also notice `interface is invalid, ignoring` there, do you have any idea why is that?

JoseExposito, 2022-10-10:
Hi!

Thanks a lot for testing it, I'm glad the battery percentage reporting is working.

> You can see than supply capacity is actually set correctly, but status isn't.

Yes, my understanding is that charging status is not supported by hid-input yet.

The code that should update the battery status is here:
https://github.com/torvalds/linux/blob/master/drivers/hid/hid-input.c#L477

As you can see, it is set to either unknown or discharging:

```
		if (dev->battery_status == HID_BATTERY_UNKNOWN)
			val->intval = POWER_SUPPLY_STATUS_UNKNOWN;
		else
			val->intval = POWER_SUPPLY_STATUS_DISCHARGING;
```

The HID descriptor I provided for the Deco LW provides information about the battery status, but the pieces to handle such information are not there just yet:

```
$ sudo hid-recorder
[...]
# ReportID: 186 / # / 0x850065:   99 | 0x850044: 1 | # | # 
                      ^                ^
                      Battery%         Charging/Discharging
E: 000000.000000 12 ba f2 01 63 01 00 00 00 00 00 00 00
```

I could try to implement it in a different change set, but it's always difficult to change code in the core... You can break a lot of devices that you can't even test.

> Notice the Product: Deco LW there, with this info i was able to slightly improve your tablet matching

Thanks! I'll update my patch to use your code.

> Also notice interface is invalid, ignoring there, do you have any idea why is that?

That's expected. By default, the tablet exposes many interfaces, but most of them need to be disabled. The driver prints that message, but it doesn't imply that something failed.

> Another question left is how to make this work with bluez.
> Regarding this, i've tried working on it, and it seems hid_uclogic is very usb specific, and would require quite alot of changes,
> above mentioned hid_to_usb_dev doesen't work with HID_BLUETOOTH_DEVICE as it seems.

Yes, I think that Bluetooth would require a fair amount of work. The device doesn't even show up in my Bluetooth settings (probably a user space problem?). You'd need help from someone with more experience dealing with Bluetooth than me.

JoseExposito, 2022-10-12:
Hi again @kanashimia

I have some time to look into this:

> Also i found two packets when tablet is powered off and on while dongle is connected:
> It seems those indicate that connection is lost/found.
> 
> 001:015:002:STREAM             1664825491.708831
>  02 F8 04 01 00 00 00 00 00 00 00 00
> 
> 001:015:002:STREAM             1664825495.761839
>  02 F8 02 01 00 00 00 00 00 00 00 00
> 
> After doing such power cycle tablet doesn't work.

My tablet does something similar.

When the dongle is connected and the tablet's switch is set to off, I see the `02 F8 04 01 00 00 00 00 00 00 00 00` packet.

Once the tablet's switch is set to on, I see the `02 F8 02 01 00 00 00 00 00 00 00 00` packet.

However, once it is on, it works immediately, without any issue.

Is yours not working after installing the drivers I sent you?

The only issue that I noticed is that those packets send the mouse pointer to the 0,0 coordinate. But since I already added a mechanism to filter unwanted events, I could add them to the list.

------------------------

Oh! And an important question. Since you are helping so much testing the patches, would you like me to add your name and email in a Tested-by tag in the patches?
https://www.kernel.org/doc/html/latest/process/submitting-patches.html#using-reported-by-tested-by-reviewed-by-suggested-by-and-fixes

I'd need your name and an email address :smile:

kanashimia, 2022-10-12:
> Since you are helping so much testing the patches, would you like me to add your name and email in a Tested-by tag in the patches?
> I'd need your name and an email address

Yes, here: `Mia Kanashi <chad@redpilled.dev>`

JoseExposito, 2022-10-16:
Thanks a lot, I added your email to the patches :smile: 

The changes to handle reconnection are a bit more complicated than I'd like, so I think I'm going to submit them in a separate series because they'd be easier to review and because, even though it is a bug, I think is better to reconnect the USB dongle from time to time than having your pointer jumping around.

So, for the first batch of patches, I'll summit this:
[digimend-kernel-drivers.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/9794311/digimend-kernel-drivers.zip)

Just in case you want/can test them. I'll test them for a few more days and submit them once the merge window is closed.

In my ToDo list now I have the reconnection issues and the battery charging/discharging issue ;)

JoseExposito, 2022-10-27:
Hi! I just emailed the patch to add support for the Deco LW. I cc'ed you so you know the status of the code, follow or participate in the conversation and also compile/use it.

JoseExposito, 2022-12-11:
Hi again @kanashimia,

I just finished the patches to support reconnection. I plan to send them to the upstream kernel on January 2023, when the development cycle of Linux 6.2 starts.

Here is the new driver:  [digimend-kernel-drivers.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/10202022/digimend-kernel-drivers.zip)

Please uninstall the previous version and install this one.

This new version handles reconnections when the USB dongle use used. Meaning that, if you turn off and on you tablet using the switch placed under the frame, or connect the USB dongle with the tablet switched on, it should reconnect automatically without issues.

I don't know if your tablet can be connected by Bluetooth. If so let me know and I'll add support for Bluetooth as well.

Let me know if you find any issues with the tablet not reconnecting, please.

Thanks!!

kanashimia, 2022-12-11:
At once hello @JoseExposito 

I tested you code and it seems to work, but there is something weird in the dmesg

<details>

<summary>[SPOILER] dmesg output</summary>

```
[  421.735337] usb 1-1: new full-speed USB device number 6 using xhci_hcd
[  421.867769] usb 1-1: New USB device found, idVendor=28bd, idProduct=0935, bcdDevice= 0.00
[  421.867785] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  421.867792] usb 1-1: Product: Deco LW
[  421.867797] usb 1-1: Manufacturer: Hanvon Ugee
[  421.867803] usb 1-1: SerialNumber: 000000000000000
[  421.873758] input: Hanvon Ugee Deco LW Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:28BD:0935.0007/input/input34
[  421.873936] uclogic 0003:28BD:0935.0007: input,hidraw2: USB HID v1.11 Mouse [Hanvon Ugee Deco LW] on usb-0000:00:14.0-1/input0
[  421.874649] uclogic 0003:28BD:0935.0008: interface is invalid, ignoring
[  421.876629] ###### DIGIMend test: Frame type = 0
[  421.876720] sysfs: cannot create duplicate filename '/class/power_supply/hid-28bd-935-battery'
[  421.876721] CPU: 4 PID: 195 Comm: kworker/4:2 Tainted: P           O       6.0.12 #1-NixOS
[  421.876723] Hardware name: HP HP 15-cx0041ur/8478, BIOS F.26 08/10/2021
[  421.876725] Workqueue: usb_hub_wq hub_event [usbcore]
[  421.876738] Call Trace:
[  421.876741]  <TASK>
[  421.876742]  dump_stack_lvl+0x45/0x5e
[  421.876747]  sysfs_warn_dup.cold+0x17/0x24
[  421.876750]  sysfs_do_create_link_sd+0xce/0xe0
[  421.876753]  device_add+0x2a9/0x960
[  421.876756]  ? dev_set_name+0x53/0x70
[  421.876760]  __power_supply_register+0x273/0x480
[  421.876764]  hidinput_setup_battery.isra.0+0x1d2/0x2c0 [hid]
[  421.876770]  hidinput_connect+0x120b/0x4a80 [hid]
[  421.876777]  hid_connect+0x599/0x600 [hid]
[  421.876783]  hid_hw_start+0x41/0x70 [hid]
[  421.876787]  uclogic_probe+0xdf/0x120 [hid_uclogic]
[  421.876791]  hid_device_probe+0x107/0x170 [hid]
[  421.876814]  really_probe+0xde/0x390
[  421.876816]  ? pm_runtime_barrier+0x50/0x90
[  421.876818]  __driver_probe_device+0x78/0x180
[  421.876820]  driver_probe_device+0x1e/0x90
[  421.876823]  __device_attach_driver+0x85/0x110
[  421.876825]  ? driver_allows_async_probing+0x60/0x60
[  421.876827]  ? driver_allows_async_probing+0x60/0x60
[  421.876828]  bus_for_each_drv+0x7b/0xd0
[  421.876831]  __device_attach+0xbc/0x210
[  421.876833]  bus_probe_device+0x8e/0xb0
[  421.876835]  device_add+0x42f/0x960
[  421.876837]  ? __debugfs_create_file+0x105/0x1c0
[  421.876840]  hid_add_device+0xed/0x240 [hid]
[  421.876844]  ? kmalloc_order+0x6e/0xa0
[  421.876861]  usbhid_probe+0x383/0x480 [usbhid]
[  421.876865]  usb_probe_interface+0xdf/0x2a0 [usbcore]
[  421.876877]  really_probe+0xde/0x390
[  421.876879]  ? pm_runtime_barrier+0x50/0x90
[  421.876881]  __driver_probe_device+0x78/0x180
[  421.876883]  driver_probe_device+0x1e/0x90
[  421.876885]  __device_attach_driver+0x85/0x110
[  421.876887]  ? driver_allows_async_probing+0x60/0x60
[  421.876889]  ? driver_allows_async_probing+0x60/0x60
[  421.876890]  bus_for_each_drv+0x7b/0xd0
[  421.876892]  __device_attach+0xbc/0x210
[  421.876895]  bus_probe_device+0x8e/0xb0
[  421.876897]  device_add+0x42f/0x960
[  421.876898]  ? usb_create_ep_devs+0x77/0xd0 [usbcore]
[  421.876910]  usb_set_configuration+0x46e/0x880 [usbcore]
[  421.876922]  usb_generic_driver_probe+0x50/0x70 [usbcore]
[  421.876934]  usb_probe_device+0x33/0x100 [usbcore]
[  421.876945]  really_probe+0xde/0x390
[  421.876946]  ? pm_runtime_barrier+0x50/0x90
[... 35 lines cut ...]
[  519.168725] ###### DIGIMend test: Frame type = 0
[  519.169540] input: Hanvon Ugee Deco LW Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:28BD:0935.000C/input/input39
[  519.170084] input: Hanvon Ugee Deco LW Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:28BD:0935.000C/input/input40
[  519.170750] uclogic 0003:28BD:0935.000C: input,hiddev96,hidraw1: USB HID v1.11 Keypad [Hanvon Ugee Deco LW] on usb-0000:00:14.0-1/input2
```

</details>

`###### DIGIMend test: Frame type = 0`
what is this, it shows as an error (in red)

`cannot create duplicate filename '/class/power_supply/hid-28bd-935-battery'` 
this one with the call trace is weird too

`uclogic 0003:28BD:0935.0009: can't register power supply: -17`
?

I'm currently running linux 6.0.12 with like 3 another patches, that one that adds this tablets support to a mainline module, battery reporting fix, and my own fix for damn broken laptop ACPI/EC firmware, it is probably not related, but just so you know.   

Anyways, it actually works, just weird log output.

Also you are right, this tablet works through bluetooth, but needs correct logic that is currently only implemented for USB.
I see that you have managed to implement support through `HID_BLUETOOTH_DEVICE` for deco pro tablets, that is awesome, how have you managed to do that?
I remember that you can't get usb_device from bluetooth devices, it should be broken if you just add that.
Or is this just something you are working on, and doesn't really work?

JoseExposito, 2022-12-12:
> ###### DIGIMend test: Frame type = 0
> what is this, it shows as an error (in red) 

Oh yes, I forgot to mention it in this issue. 

While working on the reconnection issues, I'm working with another user to add support for their tablet:
https://github.com/DIGImend/digimend-kernel-drivers/issues/622#issuecomment-1345526180

My tablet reports an incorrect frame type, I added that log just to check with if their tablet has the same problem or not. I'll be removed from the final patch of course.

More info in the commit message:

```
The report descriptor used to get information about UGEE v2 devices is
incorrect in the XP-PEN Deco Pro SW. It indicates that the device frame
is of type UCLOGIC_PARAMS_FRAME_BUTTONS but the device has a frame of
type UCLOGIC_PARAMS_FRAME_MOUSE.

Here is the original report descriptor:

  0x0e 0x03 0xc8 0xb3 0x34 0x65 0x08 0x00 0xff 0x1f 0xd8 0x13 0x00 0x00
                                     ^ This byte should be 2
```

> cannot create duplicate filename '/class/power_supply/hid-28bd-935-battery'
this one with the call trace is weird too
>
> uclogic 0003:28BD:0935.0009: can't register power supply: -17
?

Mmmm that message should not be there. It looks like when the tablet is trying to add the battery related files they already exists and fail.

Could you verify that after disconnecting the tablet the  `/class/power_supply/hid-28bd-935-battery` is removed, please?

Does this happen every time you connect your tablet? Or does it happen randomly?

And, did it happen before running this version of the driver or is it something introduced by the latest version?

> Also you are right, this tablet works through bluetooth, but needs correct logic that is currently only implemented for USB.
> I see that you have managed to implement support through [...]

About Bluetooth,  for some reason I'm not able to connect my tablet by Bluetooth, even using a vanilla kernel. I thought that I was because I was using a very old Bluetooth stack, but I updated to Fedora 37 and the issue remains.

The user in the other issue is able to connect their tablet, so this was a work in progress change as I'm unable to test that code path.

However, as you mentioned, it is going to crash when it tries to acces the USB device 🤦 Thanks for catching that up, I'll have to remove that code.

kanashimia, 2022-12-12:
> Mmmm that message should not be there. It looks like when the tablet is trying to add the battery related files they already exists and fail.
> Could you verify that after disconnecting the tablet the [...]

I see, it happends when you try to plug the BT reciever while tablet is also connected by the USB, fun.

 > About Bluetooth, for some reason I'm not able to connect my tablet by Bluetooth, even using a vanilla kernel. I thought that I was because I was using a very old Bluetooth stack, but I updated to Fedora 37 and the issue remains.

For me turning tablet itself on/off helps, but it is unreliable, it seems tablet wants similar magic reconnection handshake.

> However, as you mentioned, it is going to crash when it tries to acces the USB device facepalm Thanks for catching that up, I'll have to remove that code.

Well actually it shouldn't crash, this check should just cause the driver to immediately abort:  
```
!hid_is_using_ll_driver(hdev, &usb_hid_driver)
```
  :sweat_smile: 
It will crash if you remove it tho. 
Overall i think this is something to ask about a person who knows how this bluetooth stuff actually works. I searched through the kernel code and most of bluetooth code is pretty different from the one ulogic driver has. Things to figure out is how to remove `bInterfaceNumber` hacks and all the occurances of the `usb_device`, then perhaps it would just work (wishful thinking). Otherwise adding a whole different code path for the bluetooth devices is pretty bad.

JoseExposito, 2022-12-12:
> I see, it happends when you try to plug the USB BT dongle while tablet is also connected by the USB cable, fun.

Lol, I didn't even try that. I don't think there is a way to differentiate between the dongle and the cable, but I'll have a quick look and see if I can fix it in a different patch series.

It is a bit of a niche use case though, so it doesn't feel that bad to leave as it is.

> Overall i think this is something to ask about a person who knows how this bluetooth stuff actually works. I searched through the kernel code and most of the bluetooth code is pretty different from the one uclogic driver has. Things to figure out is how to remove bInterfaceNumber hacks and all the occurances of the usb_device, like there is a different way to send raw packets for the BT devices, then perhaps it would just work (wishful thinking). Otherwise adding a whole different code path for the bluetooth devices is pretty bad.
I'm pretty dissapointed by the lack of the kernel documentation on this topic.

Yes, Bluetooth is way out of my area of knowledge... Well, I'm a full stack web dev, so the kernel in general is out of my area of knowledge xD

Adding Bluetooth will require quite a lot of changes and it'll affect so many  devices I won't be able to test that it is too risky.

Ignoring Bluetooth for the moment...

kanashimia, 2022-12-29:
Upgraded to the linux 6.2-rc1, and now tablet works without any patches!
Correct battery status reporting doesn't seem to be merged yet tho.

JoseExposito, 2022-12-29:
Cool, thanks for testing! Battery status is queued for 6.3, we will have to wait a bit more:
https://git.kernel.org/pub/scm/linux/kernel/git/hid/hid.git/log/?h=for-6.3/hid-core

mal-w, 2023-12-20:
Are the patches you mention here part of a specific Digimend release, or only available in the linked zip? Just wondering because I have linux kernel `5.4.0` and I tried the digimend v11 release and I still had this issue...

> Here is the new driver: [digimend-kernel-drivers.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/10202022/digimend-kernel-drivers.zip)
> 
> Please uninstall the previous version and install this one.

From [this comment](https://github.com/DIGImend/digimend-kernel-drivers/issues/635#issuecomment-1345526788)

kanashimia, 2023-12-20:
@mal-w looking at the commits I don't think those patches were merged back here, I suggest using latest stable 6.6 kernel if you can, and not installing digimend, although 6.1 maybe will work too, digimend drivers are looking dead at this point and all of the progress is happening upstream thanking to @JoseExposito.

So if you install digimend on 6.6 you will get a worse experience, that's pretty bad, some people who not know about this would report same issues again. @JoseExposito can you merge all of the upstream changes here?

There is no automatic workflow (bot) here that pulls patches from the upstream, and the repo isn't as maintained as it was before.

mal-w, 2023-12-20:
Hey thanks for the quick reply.

Makes sense to me... I think if there are not plans to pull in the upstream changes then for sure there should be some notice that this project's goals are instead being implemented upstream and for more casual users what this practically means (i.e. tell people to upgrade their kernel which for most users mean upgrade your distro right?)... some places that make sense to me would be

1. At the top of the README on master of this repo
2. The install section of the Readme would be a good spot as well
3. the digimend project homepage (digimend.github.io)
4. The tablets page of that same website 

Not sure if that makes sense 🤔 but hopefully it does..

JoseExposito, 2023-12-23:
> So if you install digimend on 6.6 you will get a worse experience, that's pretty bad, some people who not know about this would report same issues again. @JoseExposito can you merge all of the upstream changes here?

I keep my `upstream` branch up to date with the work that happens in the kernel, so it was a matter of clicking a button: https://github.com/DIGImend/digimend-kernel-drivers/pull/675

Let's see if the changes get merged and a new release is tagged :smile: 

> (i.e. tell people to upgrade their kernel which for most users mean upgrade your distro right?)

Yes, for most user this mean upgrading their distro to a more recent version or in the case of rolling release distros like Arch, just upgrading.

mal-w, 2023-12-27:
Awesome.. I upgraded my kernel and the tablet works fantastically! I also haven't seen any of the specific tracking issues mentioned in this issue.

Thanks :)