DIGImend/digimend-kernel-drivers #636: UGEE 1910b support?

Vetpetmon, 2022-09-23:
Driver download link, it's hard to find: https://www.ugee.com/download/470

One of the reasons why I've been holding off on transitioning to Linux for so long is the fact that I can't find any Linux support for my tablet model, only UG-2150. I have no idea if 1910b will work right out of the box, so I want to be sure before I jump from Windows to Linux, as I've never seen artists using Linux for digital art (for this very reason).

nic3-14159, 2022-09-23:
I can't say for sure without knowing the USB ID of this tablet, but it very likely isn't supported by Digimend/upstream Linux yet and likely would not work out of the box

Vetpetmon, 2022-09-25:
[image]
Windows is giving me `HID\VID_5543&PID_0047&REV_0000&MI_00&Col01` for hardware (USB) ID

I've filled a change order at my office for a 2nd SSD to put Linux onto, so I might be able to test this next month. Some 1910b users are reporting the 1910b works with 2150 drivers, and some third party drivers (Such as XP-PEN's Artist 22) on Windows and Mac.

nic3-14159, 2022-09-25:
Hmm, there is code for USB-IDs 5543:0047 so I guess it is supported. There appear to be some generic Ugee USB IDs that are used in a number of devices

Vetpetmon, 2022-11-12:
I can confirm that the UGEE 1910b tablet has full working support (pressure sensitivity and buttons) with DIGImend outside of some WINE instances, it might be wise to update the website's list with this information.

closed 2022-11-12 by Vetpetmon